esphome/issues #6595: power_save_mode NONE is incompatible with esp32_ble, but I don't have esp32_ble in my config?

k1lln1n3, 2024-12-22:
### The problem

So i'm working on setting up a bunch of bluetooth proxies. I've noticed that setting power_save_mode to none has fixed a number of issues with my esp32s so I've changed my config to remove esp32_ble from all of my devices. However, despite it not exiting in my instance anymore, I still get this error. 

I can no longer add any new esp32 and set the power save mode to none. This clearly seems to be a bug of some kind, but im also hoping to get a workaround while I wait. Any ideas?  Let me know if there is more info I can provide.

power_save_mode NONE is incompatible with esp32_ble. Please remove the power save mode. See also https://github.com/esphome/issues/issues/2141#issuecomment-865688582.

### Which version of ESPHome has the issue?

2024.12.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.12.3

### What platform are you using?

ESP32

### Board

esp32 S3 but other various esp32s and esp8266s

### Component causing the issue

esp32_ble

### YAML Config

```yaml
esphome:
  name: esp32-bedroompresence
  friendly_name: ESP32-BedroomPresence

esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "redacted="

ota:
  - platform: esphome
    password: "redacted"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  power_save_mode: NONE

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "redacted"
    password: "redacted"

captive_portal:
    
bluetooth_proxy:
```

### Anything in the logs that might be useful for us?

```txt
Failed config

wifi: [source /config/esphome/esp32-bedroompresence.yaml:22]
  
  power_save_mode NONE is incompatible with esp32_ble. Please remove the power save mode. See also https://github.com/esphome/issues/issues/2141#issuecomment-865688582.
```

ssieb, 2024-12-22:
You have the `bluetooth_proxy` still in there.

closed 2024-12-22 by ssieb

k1lln1n3, 2024-12-22:
> You have the `bluetooth_proxy` still in there.

The specific lines that started it were esp32_ble:. I was able to deploy with bluetooth_proxy and power_Save_mode NONE before I startedto use esp32 in other projects. Are you saying that:

a) bluetooth_proxy is a component of esp32_ble and
b) I can't have a bluetooth proxy with power_save_mode none?

ssieb, 2024-12-22:
`esp32_ble` is the base component that provides BLE support, so the proxy of course has to use that.
So b is correct.  How were you able to do it before?

k1lln1n3, 2024-12-22:
@ssieb I was able to deploy it before ever implementing the esp32_ble component in my esp32s. I'll try and replicate it in a new environment, but based on what you said, THAT might be a bug?

ssieb, 2024-12-22:
Yes, that sounds like it would be a bug.  But I don't see how that would change, since the yaml is validated separately each time.  It doesn't keep information from previous versions.  But I think the validation was added recently.  Previously, there would just be an error in the logs.

k1lln1n3, 2024-12-22:
Makes much more sense now. Thanks