esphome/issues #6597: Не отображает экран

dwt15, 2024-12-23:
### The problem

в версии ESPHome 2024.12.0  после обновления перестал отображать экран в версии 2024,11 работало

### Which version of ESPHome has the issue?

ESPHome 2024.12.0 

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.12.4

### What platform are you using?

ESP32-IDF

### Board

Sunton ESP32-S3 

### YAML Config

```yaml
esphome:
  name: display2748
  friendly_name: display2748
  platformio_options:
     board_build.flash_mode: dio
 
esp32:
  board: esp32-s3-devkitc-1  
  flash_size: 16MB
  partitions: "default_16MB.csv"
  framework:
    type:  esp-idf 
    version: recommended
    sdkconfig_options:
      CONFIG_ESP32S3_DEFAULT_CPU_FREQ_240: y
      CONFIG_ESP32S3_DATA_CACHE_64KB: y
      CONFIG_SPIRAM_FETCH_INSTRUCTIONS: y
      CONFIG_SPIRAM_RODATA: y 
      CONFIG_ESPTOOLPY_FLASHSIZE_16MB: y

psram:
  mode: octal
  speed: 80MHz
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2024.12.0
INFO Reading configuration /config/esphome/display43bx25.yaml...
INFO Generating C++ source...
INFO Updating https://github.com/espressif/esp-protocols.git@mdns-v1.3.2
INFO Compiling app...
Processing display2748 (board: esp32-s3-devkitc-1; framework: espidf; platform: https://github.com/pioarduino/platform-espressif32.git#51.03.07)
--------------------------------------------------------------------------------
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
 - framework-espidf @ 3.50105.0 (5.1.5) 
 - tool-cmake @ 3.21.3 
 - tool-esptoolpy @ 4.8.1 
 - tool-mklittlefs @ 3.2.0 
 - tool-ninja @ 1.7.1 
 - tool-riscv32-esp-elf-gdb @ 12.1.0+20221002 
 - tool-xtensa-esp-elf-gdb @ 12.1.0+20221002 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-riscv32-esp @ 12.2.0+20230208 
 - toolchain-xtensa-esp32s3 @ 12.2.0+20230208
Reading CMake configuration...
Dependency Graph
|-- noise-c @ 0.1.6
|-- lvgl @ 8.4.0
Compiling .pioenvs/display2748/src/main.cpp.o
Linking .pioenvs/display2748/firmware.elf
RAM:   [=         ]  11.8% (used 38528 bytes from 327680 bytes)
Flash: [==        ]  18.3% (used 1199421 bytes from 6553600 bytes)
Building .pioenvs/display2748/firmware.bin
Creating esp32s3 image...
Successfully created esp32s3 image.
esp32_create_combined_bin([".pioenvs/display2748/firmware.bin"], [".pioenvs/display2748/firmware.elf"])
Wrote 0x134eb0 bytes to file /data/build/display2748/.pioenvs/display2748/firmware.factory.bin, ready to flash to offset 0x0
esp32_copy_ota_bin([".pioenvs/display2748/firmware.bin"], [".pioenvs/display2748/firmware.elf"])
========================= [SUCCESS] Took 22.91 seconds =========================
INFO Successfully compiled program.
INFO Connecting to 192.168.0.138 port 3232...
INFO Connected to 192.168.0.138
INFO Uploading /data/build/display2748/.pioenvs/display2748/firmware.bin (1199792 bytes)
Uploading: [============================================================] 100% Done...

INFO Upload took 21.88 seconds, waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from 192.168.0.138 using esphome API
INFO Successfully connected to display2748 @ 192.168.0.138 in 4.014s
INFO Successful handshake with display2748 @ 192.168.0.138 in 0.149s
[10:02:56][I][app:100]: ESPHome version 2024.12.0 compiled on Dec 19 2024, 10:02:06
[10:02:56][C][wifi:600]: WiFi:
[10:02:56][C][wifi:428]:   Local MAC: DC:DA:0C:57:0D:C4
[10:02:56][C][wifi:433]:   SSID: 'Archer'[redacted]
[10:02:56][C][wifi:436]:   IP Address: 192.168.0.138
[10:02:56][C][wifi:439]:   BSSID: 28:EE:52:42:13:7F[redacted]
[10:02:56][C][wifi:441]:   Hostname: 'display2748'
[10:02:56][C][wifi:443]:   Signal strength: -67 dB ▂▄▆█
[10:02:56][C][wifi:447]:   Channel: 1
[10:02:56][C][wifi:448]:   Subnet: 255.255.255.0
[10:02:56][C][wifi:449]:   Gateway: 192.168.0.1
[10:02:56][C][wifi:450]:   DNS1: 192.168.0.1
[10:02:56][C][wifi:451]:   DNS2: 192.168.0.1
[10:02:56][C][logger:185]: Logger:
[10:02:56][C][logger:186]:   Level: DEBUG
[10:02:56][C][logger:188]:   Log Baud Rate: 115200
[10:02:56][C][logger:189]:   Hardware UART: USB_SERIAL_JTAG
[10:02:56][C][ledc.output:180]: LEDC Output:
[10:02:56][C][ledc.output:181]:   Pin GPIO2
[10:02:56][C][ledc.output:182]:   LEDC Channel: 0
[10:02:56][C][ledc.output:183]:   PWM Frequency: 1000.0 Hz
[10:02:56][C][ledc.output:184]:   Phase angle: 0.0°
[10:02:56][C][ledc.output:185]:   Bit depth: 14
[10:02:56][C][:135]: RPI_DPI_RGB LCD
[10:02:56][C][rpi_dpi_rgb:136]:   Height: 272
[10:02:56][C][rpi_dpi_rgb:137]:   Width: 480
[10:02:56][C][rpi_dpi_rgb:138]:   DE Pin: GPIO40
[10:02:56][C][rpi_dpi_rgb:143]:   Data pin 0: GPIO15
[10:02:56][C][rpi_dpi_rgb:143]:   Data pin 1: GPIO16
[10:02:56][C][rpi_dpi_rgb:143]:   Data pin 2: GPIO4
[10:02:56][C][rpi_dpi_rgb:143]:   Data pin 3: GPIO45
[10:02:56][C][rpi_dpi_rgb:143]:   Data pin 4: GPIO48
[10:02:56][C][psram:020]: PSRAM:
[10:02:56][C][psram:021]:   Available: YES
[10:02:56][C][psram:024]:   Size: 7040 KB
[10:02:56][C][gt911.touchscreen:127]: GT911 Touchscreen:
[10:02:56][C][gt911.touchscreen:128]:   Address: 0x5D
[10:02:56][C][lvgl:397]: LVGL:
[10:02:56][C][captive_portal:089]: Captive Portal:
[10:02:56][C][sntp:042]: SNTP Time:
[10:02:56][C][sntp:045]:   Server 0: '0.pool.ntp.org'
[10:02:56][C][sntp:045]:   Server 1: '1.pool.ntp.org'
[10:02:56][C][sntp:045]:   Server 2: '2.pool.ntp.org'
[10:02:56][C][mdns:116]: mDNS:
[10:02:56][C][mdns:117]:   Hostname: display2748
[10:02:56][C][esphome.ota:073]: Over-The-Air updates:
[10:02:56][C][esphome.ota:074]:   Address: display2748.local:3232
[10:02:56][C][esphome.ota:075]:   Version: 2
[10:02:56][C][esphome.ota:078]:   Password configured
[10:02:56][C][safe_mode:018]: Safe Mode:
[10:02:56][C][safe_mode:019]:   Boot considered successful after 60 seconds
[10:02:56][C][safe_mode:021]:   Invoke after 10 boot attempts
[10:02:56][C][safe_mode:022]:   Remain in safe mode for 300 seconds
[10:02:56][C][api:140]: API Server:
[10:02:56][C][api:141]:   Address: display2748.local:6053
[10:02:56][C][api:143]:   Using noise encryption: YES
[10:02:56][C][homeassistant.binary_sensor:039]: Homeassistant Binary Sensor 'rz1_s'
[10:02:56][C][homeassistant.binary_sensor:040]:   Entity ID: 'switch.z2_rz01'
 
[10:02:57][D][esp-idf:000]: E (7030) lcd_panel: esp_lcd_panel_draw_bitmap(33): invalid panel handle

[10:02:57][E][rpi_dpi_rgb:084]: lcd_lcd_panel_draw_bitmap failed: ESP_ERR_INVALID_ARG
[10:02:57][D][esp-idf:000]: E (7036) lcd_panel: esp_lcd_panel_draw_bitmap(33): invalid panel handle
```

### Additional information

делаю откат на 11 версию - работает

dwt15, 2024-12-23:
работает если указать
  framework:
    type:  esp-idf  
    version: 4.4.5  
    platform_version: 5.4.0

clydebarrow, 2025-01-09:
Please use English - I appreciate that you may not be proficient in English, but you can use an online translator just as easily as anyone else and you will get more attention if more people can read your post.

You also did not include the display configuration in the YAML and since I'm guessing it's a display issue you have, that is essential.

Also collect serial logs, not OTA logs - errors early in the boot process are not captured by OTA logs.