esphome/issues #6599: Voice assistant: changing "volume_multiplier" has no effect

Pippisbruffi, 2024-12-23:
### The problem

I built a voice assistant using a ICS43434 microphone and a MAX98357A amplifier connected to an ESP32 S3 N16R8 Devkit C-1 board. 
It works, but, as per title, changing the "volume_multiplier" in the YAML config has no effect. I'm able to listen to the recordings and the volume of my voice stays the same no matter if I set "volume_multiplier" to 1 or to 1000.
Some months ago I was using the INMP441 microphone and I remember that that setting was actually working. 
I checked the changelogs of the last months and found nothing about volume multiplier.
Is it the microphone's fault?
Or maybe there is something wrong in my config?
Or is it a bug?

### Which version of ESPHome has the issue?

2024.12.2

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Board

ESP32 S3 Devkit C1 (N16R8)

### YAML Config

```yaml
esphome:
  name: assist13
  friendly_name: Assist13
  on_boot:
     - priority: -100
       then:
         - wait_until: api.connected
         - if:
             condition:
               - switch.is_on: use_wake_word
             then:
               - media_player.volume_set: 70%
               - delay: 1s
               - voice_assistant.start_continuous:

esp32:
  board: esp32s3box
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "..."

ota:
  - platform: esphome
    password: "..."

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Assist13 Fallback Hotspot"
    password: "..."

captive_portal:
    

i2s_audio:
  - id: i2s_in
    i2s_lrclk_pin: GPIO5 ##mic WS
    i2s_bclk_pin: GPIO4 ##mic SCK
  - id: i2s_out
    i2s_lrclk_pin: GPIO10 ##ampli LRC
    i2s_bclk_pin: GPIO11 ##ampli BCLK

microphone:
  - platform: i2s_audio
    i2s_audio_id: i2s_in
    id: mic
    adc_type: external
    i2s_din_pin: GPIO6 ##mic SD
    channel: left
    bits_per_channel: 24bit

media_player:
  - platform: i2s_audio
    i2s_audio_id: i2s_out
    id: big_speaker
    dac_type: external
    name: stogazio
    i2s_dout_pin: 
      number: GPIO12 ##ampli DIN
      allow_other_uses: true
    ##i2s_mode: primary

voice_assistant:
  microphone: mic
  use_wake_word: true
  noise_suppression_level: 1
  volume_multiplier: 1
  media_player: big_speaker
  id: assist

  on_wake_word_detected:
    then:
      ##- voice_assistant.stop
      ##- lambda: id(assist).set_use_wake_word(false);
      ##- delay: 800ms
      ##- media_player.play_media: http://192.168.1.143:8123/local/350hz.mp3
      ##- wait_until:
          ##not:
            ##media_player.is_playing: big_speaker
      ##- delay: 800ms
      ##- voice_assistant.start_continuous
      - light.turn_on:
          id: onboardled
          brightness: 100%
          red: 0%
          green: 0%
          blue: 100%
  on_client_connected: 
    then:
      - light.turn_on:
          id: onboardled
          brightness: 100%
          red: 0%
          green: 100%
          blue: 0%
  on_client_disconnected: 
    then:
      - light.turn_on:
          id: onboardled
          brightness: 100%
          red: 100%
          green: 0%
          blue: 0%
  on_idle:
      - light.turn_on:
          id: onboardled
          brightness: 100%
          red: 0%
          green: 100%
          blue: 0%
  on_stt_end: 
    then:
    - delay: 1000ms
    - voice_assistant.start_continuous
    - lambda: id(assist).set_use_wake_word(true);

interval:
  - interval: 30s
    then:
      - if:
          condition:
            microphone.is_capturing:
          then:
            - switch.turn_on: microfono2
          else:
            - switch.turn_off: microfono2
  - interval: 31s
    then:
      - if: 
          condition: 
            - switch.is_off: microfono2
          then:
            - voice_assistant.stop: 
            - delay: 10s     
            - voice_assistant.start_continuous: 

switch:
  - platform: template
    name: Use wake word
    id: use_wake_word
    optimistic: true
    restore_mode: RESTORE_DEFAULT_ON
    entity_category: config
    on_turn_on:
      - media_player.stop: 
      - delay: 1s
      - lambda: id(assist).set_use_wake_word(true);
      - voice_assistant.start_continuous
    on_turn_off:
      - voice_assistant.stop
      - delay: 1s
      - lambda: id(assist).set_use_wake_word(false);
  - platform: template
    name: microfono
    id: microfono2
    optimistic: true

output:
  - platform: gpio
    pin:
        number: GPIO12
        allow_other_uses: true
    id: set_low_speaker
light:
  - platform: esp32_rmt_led_strip
    id: onboardled 
    rgb_order: GRB
    pin: GPIO48
    num_leds: 1
    rmt_channel: 0
    chipset: ws2812
    name: "My Light"

button:
  - platform: restart
    name: "Reboot assistente"
  - platform: template
    name: pippo
    on_press: 
      then:
          - media_player.play_media: http://192.168.1.143:8123/local/beep.mp3
```

zestysoft, 2025-01-01:
I've discovered the same thing.

ESP32-S3-DevKitC-1-N16R8-WROOM-1
HA 2024.12.5
Mic: SPH0645LM4H

I'm making this determination based on the wav files being saved via the assist_pipeline.debug_recording_dir setting.

To hear the sound from these files, I have to crank the pc's volume to max and then it's still pretty subtle.

pkkrusty, 2025-01-01:
I'm having the same issue with INMP441 mic on my setup. Volume multiplier has no effect for me. 

Do you happen to have the YAML you used back when you were jusing INMP441 and multiplier was working?

@Pippisbruffi Have you found the SPH0645 to be significantly better than the INMP441? It's double the price on AliExpress, so I started with the cheap option...

Pippisbruffi, 2025-01-01:
> I'm having the same issue with INMP441 mic on my setup. Volume multiplier has no effect for me.
> 
> Do you happen to have the YAML you used back when you were jusing INMP441 and multiplier was working?
> 
> @Pippisbruffi Have you found the SPH0645 to be significantly better than the INMP441? It's double the price on AliExpress, so I started with the cheap option...
> 
> I have a issue opened on the HA github also... [home-assistant/home-assistant.io#35402](https://github.com/home-assistant/home-assistant.io/issues/35402)

The yaml is practically identical, except for "bits_per_channel" which I did not specify when I was using INMP441. I can confirm that noise suppression and volume multiplier were working 2-3 months ago. There is an Home Assistant addon called Esphome legacy that lets you install older version of esphome. If you have time, try to install the 2024.8 version (for example) and see if volume multiplier works with your INMP441. I have some spare INMP441 and I will try as well with other esp32 boards.

Anyway, I'm not using the SPH0645, I'm using the ICS43434 soldered on a custom pcb that I developed on kicad and ordered online. There is definitely less background noise compared to INMP441, but nothing dramatic.

@zestysoft me too. I can listen to the recordings saved in assist_pipeline.debug_recording_dir and I have to turn the volume of my pc all the way up to hear something.

Pippisbruffi, 2025-01-01:
If the devs are interested, I tried changing the board to "esp32-s3-devkitc-1" and the framework to "esp-idf" but the issue was still present

zestysoft, 2025-01-10:
Reverting to 2024.8.3 shows no difference for me.

pkkrusty, 2025-01-16:
Probably something in the Wyoming code then that ignores the volume multiplier instruction? That seems more likely since reverting to old Esphome had no effect, but we know it worked a few months ago.

gb0101010101, 2025-02-15:
I also see no effects when changing `auto_gain` or `volume_multiplier` when using:

- esp32-s3-devkitc (ESP32-S3-WROOM-1-N16R8) with INMP441 via I2S Audio
- esp-wrover-kit (ESP32-Korvo v1.1 ESP32-WROVER-E) with ES8311 via ESP ADF

Currently on ESPHome 2024.12.4.

zestysoft, 2025-02-19:
Hey all, I got some info from the folks via their Discord channel. micro_wake_word requires a dedicated mic. You'll need a second mic for the stuff after.

When I debugged the code, what happens is the pipeline that is normally responsible for enhancing the audio is instantiated by the micro wake word instead -- that doesn't pass in stuff like the volume multiplier.  My understanding is that by using a second mic for voice_assistance, it creates a second pipeline with the parameters we were expecting to get applied.

pkkrusty, 2025-02-19:
Amazing.  Thanks for continuing to work on this. I'll try with a second mic. [This ](https://gist.github.com/EverythingSmartHome/055fbdde31a607ef9d695d5cac780e94#gistcomment-5380865)discussion has a yaml setup for that.

AtrejaAlwinnen, 2025-02-27:
Hi, I also have the problem with the INMP441 and the low volume. Is there a fix planned for this, or do we really have to use a second mic now? 
This doesn't improve the wake word recognition, only the speech recognition. 
That's not a good solution.

rmeissn, 2025-03-03:
I basically got the same problem. Using MWW and VA, and wanted to tune in the microphone. Unfortunately, changing `volume_multiplier` and `auto_gain` seems to have absolutely no effect on the recorded audio.
Additionally, the documentation is quite unclear on the topic.
```
auto_gain (Optional, dBFS): Auto gain level to apply to the assist pipeline. Between 0dBFS and 31dBFS inclusive. Defaults to 0 (disabled).
volume_multiplier (Optional, float): Volume multiplier to apply to the assist pipeline. Must be larger than 0. Defaults to 1 (disabled).
```

What does "assist pipeline" exactly mean? Microphone only or only audio input to VA (so not the microphone, as e.g. MWW uses that too) or does assist pipeline also include the response audio (e.g. multiplier at two: response is twice as lound)?

DragonQ, 2025-04-20:
I don't think this is (completely) related to piper/whisper/wyoming. After getting my voice satellites working with ESPHome 2024.4.2 and the stock voice assistant package last May/June (github://esphome/firmware/wake-word-voice-assistant/esp32-s3-box-3.yaml@main), I never updated them and they continued to work well, save for the odd freeze requiring a reset.

Last week I took the plunge and updated ESPHome to 2025.3.x and the latest version of the voice assistant package, and I'm now experiencing the problem described in this thread: I have to say the wake word really loudly for it to hear me. This was not the case before. I don't really want to downgrade ESPHome because it'd mean downgrading all of my other devices too but it seems to be the only option. I also noticed that the `vad_threshold` parameter has been removed, not sure what its replacement is? Is there any actual documentation for the voice assistant package?

It feels like once voice was added to HA, it was subsequently just abandoned and left to rot. There's no clear documentation anywhere to tweak basic settings like volume and the whole functionality seems to actually be broken with current code. If anyone bought an ESP32-S3-BOX-3 and used current versions of all components, they'd probably be confused as to how anyone is using them successfully.

surajgoel5, 2025-08-19:
Is there any progress with this issue? It seems to be stale but I am facing the same problem.

ErikDou, 2025-08-24:
> Is there any progress with this issue? It seems to be stale but I am facing the same problem.

I'm also still having the same Issue.

The only thing that works for me is adding gain for both VA and MWW

```
voice_assistant:
  microphone:
    microphone: mic
    gain_factor: 20
```

```
micro_wake_word:
  microphone:
    microphone: mic
    gain_factor: 10
```

rmeissn, 2026-02-12:
Update for ESPHome 2026.1:
I got two mics on one I2S bus. When settings the microphone to `channels: stereo`, for some weird reason I get a signal-noise-ratio (SNR) of about 1-2db (should be >20db). When using `channels: right` (or left), I get a SNR of at lest 16db (much better). But  then, I can only use one of the microphones (for mww and va), not separate ones for each of these.

Using gain_factor (I use 40) amplifies the recorded audio so I can play it back on my laptop and hear my voice. Unfortunately, noise is also amplified by this setting, leaving me with a SNR of about 16db from 50cm away (>20 would be better).

I did also experiment with other options from va, like `noise_supression_level`, `volume_multiplier`, and `auto_gain`, but these settings won't change anything in the produced sound files. Maybe this is related to 
1) my "one mic for mww and va" setup
2) these settings are not working
3) I only get the recorded audio before these settings are applied

rmeissn, 2026-02-13:
FYI: Someone working on a waveshare-s2 implemented an additional component (named `esp_afe`) that provides:

- Acoustic Echo Cancellation (AEC)
- Blind Source Separation (BSS)
- Noise Suppression (NS)
- MISO (Multi Input Single Output)
- AGC (Automatic Gain Control)

directly on the ESP side.
See https://docs.espressif.com/projects/esp-sr/en/latest/esp32/audio_front_end/README.html#overview for an explanation on those topics.

See https://github.com/craigmillard86/waveshare-s2-audio_esphome_voice/blob/91c998092c60985a7944a89b937d1b1fbc3337b8/packages/audio_hardware.yaml#L40 for the component definition and https://github.com/craigmillard86/waveshare-s2-audio_esphome_voice/tree/rework-branch/components/esp_afe for the component itself.

I have not tried to use or integrate this yet.