esphome/issues #6600: Platform not found: 'sensor.mpl115a2'

CobusTaljaard, 2024-12-24:
### The problem

Hi
According to https://github.com/esphome/esphome/pull/6584, the MPL115A2 Barometric Pressure/Temperature Sensor was added. When trying to compile my updated configuration file, I keep getting the message, " Platform not found: 'sensor.mpl115a2'".

I searched breaking changes, documentation, forums etc. The sensor is also not listed here: https://github.com/esphome/esphome-docs/tree/current/components/sensor

I think this might be a new issue?

### Which version of ESPHome has the issue?

2024.12.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

n/a (I'm using ESPHome cli on Windows)

### What platform are you using?

ESP32

### Board

Generic, but memory best matches az-delivery-devkit-v4

### Component causing the issue

i2c MPL115A2

### YAML Config

```yaml
esphome:
  project:
    name: "cobus.oslo-stoep-sensors-esp32"
    version: "2024.12.23"
  name: oslo-stoep-sensors-esp32
  friendly_name: oslo-stoep-sensors-esp32

esp32:
  board: az-delivery-devkit-v4
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "*deleted-for-this-share*"

ota:
  - platform: esphome
    password: "*deleted-for-this-share*"
  - platform: http_request #required for remote firmware updating
    on_begin:
      then:
        - logger.log: "OTA start"
    on_progress:
      then:
        - logger.log:
            format: "OTA progress %0.1f%%"
            args: ["x"]
    on_end:
      then:
        - logger.log: "OTA end"
    on_error:
      then:
        - logger.log:
            format: "OTA update error %d"
            args: ["x"]
    on_state_change:
      then:
        lambda: 'ESP_LOGD("ota", "State %d", state);'

http_request:  #required for remote firmware updating
  verify_ssl: false

update:
  - platform: http_request  #required for remote firmware updating
    id: oslo_stoep_sensors_esp32_firmware_update
    name: oslo_stoep_sensors_esp32_firmware_update
    source: https://*deleted-for-this-share*/manifest.json
    update_interval: 1440min

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: true  # might help with stability during updates

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Oslo-Stoep-Sensors-Esp32"
    password: !secret fallback_password

captive_portal:

i2c:
  - id: bus_a
    sda: 21
    scl: 22
    scan: true
  - id: bus_b
    sda: 18
    scl: 19
    scan: true

sensor:

 - platform: dht
   pin: 17
   temperature:
     name: Temperature
     accuracy_decimals: 1
   humidity:
     name: Humidity
     accuracy_decimals: 0
   update_interval: 60s

 - platform: max44009
   i2c_id: bus_a
   address: 0x4A
   name: Illuminance
   update_interval: 60s
   accuracy_decimals: 0
   mode: auto

 - platform: adc
   pin: 35
   name: Volume
   device_class: sound_pressure
   unit_of_measurement: "%"
   accuracy_decimals: 1
   update_interval: 10ms
   filters:
     - multiply: 303 #analog volt to scale of 0 to 100%
     - max:
        window_size: 6000
        send_every: 6000

 - platform: mpl115a2
   i2c_id: bus_b
   address: 0x60
   temperature:
     name: Temperature2
     accuracy_decimals: 1
   pressure:
     name: Pressure
   update_interval: 60s 
 
binary_sensor:
  - platform: gpio
    pin: 04
    name: Sound
    device_class: sound
    filters:
      - invert:
      - delayed_off: 60000ms
```

### Additional information

ESPHome CLI output:

INFO ESPHome 2024.12.2
INFO Reading configuration C:\Users\hopsm\oslo-stoep-sensors-esp32.yaml...
INFO Unable to import component mpl115a2.sensor: No module named 'esphome.components.mpl115a2'
Failed config

sensor.mpl115a2: [source C:\Users\hopsm\oslo-stoep-sensors-esp32.yaml:110]

  Platform not found: 'sensor.mpl115a2'.
  platform: mpl115a2
  i2c_id: bus_b
  address: 96
  temperature:
    name: Temperature2
    accuracy_decimals: 1
  pressure:
    name: Pressure
  update_interval: 60s

randybb, 2024-12-24:
no, this was never merged and it was closed by the author.

But it might work, you can test it by adding this to your config:
```yaml
external_components:
  - source: github://pr#6584
    components: mpl115a2
```

closed 2024-12-24 by randybb

CobusTaljaard, 2024-12-25:
Thanks @randybb