esphome/issues #6602: ESP32 with MAX98357 reboots on media play

bram2202, 2024-12-24:
### The problem

I'm trying to build a ESP based announcement speaker for use is Homeassistent.
I'm using a ESP32-WROOM-32 board and a DFRobot MAX98357 I2S Amplifier Module.

When i send a tekst-to-speech command the ESP receives it but nothing is played and than the ESP reboots.
The (Verbose) logs show nothing useful.

I added some capacitors to counter any peaks in power draw that the supply could not handle, should not be a issue. 

### Which version of ESPHome has the issue?

2024.12.2

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2024.12.5

### What platform are you using?

ESP32

### Board

eps32s

### Component causing the issue

i2c

### YAML Config

```yaml
esphome:
  name: esp-speaker
  friendly_name: esp-speaker

logger:
  level: VERY_VERBOSE

substitutions:
  friendly_name: "esp-speaker"

esp32:
  board: esp32dev
  framework:
    type: arduino

# Include wifi, api, ota etc
packages:
  common: !include common.yaml

i2s_audio:
  i2s_lrclk_pin: GPIO27
  i2s_bclk_pin: GPIO25 

media_player:
  - platform: i2s_audio
    id: announcer_media_player
    name: "${friendly_name} player"
    dac_type: external
    i2s_dout_pin: GPIO26
    mode: mono
```

### Anything in the logs that might be useful for us?

```txt
[11:51:37][I][app:100]: ESPHome version 2024.12.2 compiled on Dec 24 2024, 11:42:07
[11:51:37][C][wifi:600]: WiFi:
[11:51:37][C][wifi:428]:   Local MAC: 3C:61:05:13:3F:78
[11:51:37][C][wifi:433]:   SSID: [redacted]
[11:51:37][C][wifi:436]:   IP Address: 192.168.30.249
[11:51:37][C][wifi:440]:   BSSID: [redacted]
[11:51:37][C][wifi:441]:   Hostname: 'esp-speaker'
[11:51:37][C][wifi:443]:   Signal strength: -56 dB ▂▄▆█
[11:51:37][V][wifi:445]:   Priority: 0.0
[11:51:37][C][wifi:447]:   Channel: 1
[11:51:37][C][wifi:448]:   Subnet: 255.255.255.0
[11:51:37][C][wifi:449]:   Gateway: 192.168.30.1
[11:51:37][C][wifi:450]:   DNS1: 192.168.1.10
[11:51:37][C][wifi:451]:   DNS2: 1.1.1.1
[11:51:37][C][logger:185]: Logger:
[11:51:37][C][logger:186]:   Level: VERY_VERBOSE
[11:51:37][C][logger:188]:   Log Baud Rate: 115200
[11:51:37][C][logger:189]:   Hardware UART: UART0
[11:51:37][C][uptime.sensor:033]: Uptime Sensor 'esp-speaker Uptime'
[11:51:37][C][uptime.sensor:033]:   Device Class: 'duration'
[11:51:37][C][uptime.sensor:033]:   State Class: 'total_increasing'
[11:51:37][C][uptime.sensor:033]:   Unit of Measurement: 's'
[11:51:37][C][uptime.sensor:033]:   Accuracy Decimals: 0
[11:51:37][C][uptime.sensor:033]:   Icon: 'mdi:timer-outline'
[11:51:37][V][uptime.sensor:033]:   Unique ID: '3c6105133f78-uptime'
[11:51:37][C][uptime.sensor:034]:   Type: Seconds
[11:51:37][C][status:034]: Status Binary Sensor 'esp-speaker Status'
[11:51:37][C][status:034]:   Device Class: 'connectivity'
[11:51:37][C][restart:068]: Restart Switch 'esp-speaker Restart'
[11:51:37][C][restart:070]:   Icon: 'mdi:restart'
[11:51:37][C][restart:091]:   Restore Mode: always OFF
[11:51:37][C][version.text_sensor:021]: Version Text Sensor 'esp-speaker ESPHome Version'
[11:51:37][C][version.text_sensor:021]:   Icon: 'mdi:new-box'
[11:51:38][V][version.text_sensor:021]:   Unique ID: '3c6105133f78-version'
[11:51:38][C][mdns:116]: mDNS:
[11:51:38][C][mdns:117]:   Hostname: esp-speaker
[11:51:38][V][mdns:118]:   Services:
[11:51:38][V][mdns:120]:   - _esphomelib, _tcp, 6053
[11:51:38][V][mdns:122]:     TXT: friendly_name = esp-speaker
[11:51:38][V][mdns:122]:     TXT: version = 2024.12.2
[11:51:38][V][mdns:122]:     TXT: mac = 3c6105133f78
[11:51:38][V][mdns:122]:     TXT: platform = ESP32
[11:51:38][V][mdns:122]:     TXT: board = esp32dev
[11:51:38][V][mdns:122]:     TXT: network = wifi
[11:51:38][C][esphome.ota:073]: Over-The-Air updates:
[11:51:38][C][esphome.ota:074]:   Address: esp-speaker.local:3232
[11:51:38][C][esphome.ota:075]:   Version: 2
[11:51:38][C][esphome.ota:078]:   Password configured
[11:51:38][C][safe_mode:018]: Safe Mode:
[11:51:38][C][safe_mode:020]:   Boot considered successful after 60 seconds
[11:51:38][C][safe_mode:021]:   Invoke after 10 boot attempts
[11:51:38][C][safe_mode:023]:   Remain in safe mode for 300 seconds
[11:51:38][C][api:140]: API Server:
[11:51:38][C][api:141]:   Address: esp-speaker.local:6053
[11:51:38][C][api:145]:   Using noise encryption: NO
[11:51:38][C][wifi_signal.sensor:010]: WiFi Signal 'esp-speaker WiFi Signal Strength'
[11:51:38][C][wifi_signal.sensor:010]:   Device Class: 'signal_strength'
[11:51:38][C][wifi_signal.sensor:010]:   State Class: 'measurement'
[11:51:38][C][wifi_signal.sensor:010]:   Unit of Measurement: 'dBm'
[11:51:38][C][wifi_signal.sensor:010]:   Accuracy Decimals: 0
[11:51:38][V][wifi_signal.sensor:010]:   Unique ID: '3c6105133f78-wifisignal'
[11:51:38][C][audio:225]: Audio:
[11:51:38][C][audio:247]:   External DAC channels: 1
[11:51:38][C][audio:248]:   I2S DOUT Pin: 26
[11:51:41][VV][api.service:916]: on_media_player_command_request: MediaPlayerCommandRequest {
  key: 4169698641
  has_command: NO
  command: MEDIA_PLAYER_COMMAND_PLAY
  has_volume: NO
  volume: 0
  has_media_url: YES
  media_url: 'http://192.168.1.9:8123/api/tts_proxy/oRcc-tpLl8hNyFXGskYrxQ.mp3'
  has_announcement: YES
  announcement: YES
}
[11:51:41][D][media_player:065]: 'esp-speaker player' - Setting
[11:51:41][D][media_player:072]:   Media URL: http://192.168.1.9:8123/api/tts_proxy/oRcc-tpLl8hNyFXGskYrxQ.mp3
[11:51:41][D][media_player:078]:  Announcement: yes
[11:51:41][VV][api.service:324]: send_media_player_state_response: MediaPlayerStateResponse {
  key: 4169698641
  state: MEDIA_PLAYER_STATE_IDLE
  volume: 1
  muted: NO
}
[11:51:41][W][component:237]: Component api took a long time for an operation (62 ms).
[11:51:41][W][component:238]: Components should block for at most 30 ms.
```

bram2202, 2025-08-25:
I have not fixed it, still an issue as far as I know.
I must say i did not touch it for a few months so maybe something changed, but if you have the same it doesn't look like it.

popy2k14, 2025-08-25:
Thx for your answer.

Interestingly:
When i use the arduino framework it works and never crashes.
Will investigate further,

PS: i am using an 8MB external PSRAM (this one: ESP32-S3-N16R8 with 16MB Flash & 8 MB PRAM)

Maybe this would also help in your setup.