esphome/issues #6603: NEC protocol "command_repeats" implementation is incorrect

GRbit, 2024-12-24:
### The problem

Greetings, dear fellow esphome developers!
Thank you for the great tool that you all build. I think I found a bug in NEC protocol implementation, and I hope I was able to provide clear and complete issue description below.

According to the documentation of NEC protocol ([a good explanation can be found here https](https://techdocs.altium.com/display/FPGA/NEC%2bInfrared%2bTransmission%2bProtocol)) repeats of the command are not exactly the command itself, but a specific simple signal that is sent after the command. This specific signal being sent dictates the controlled device to repeat the command itself.

Current implementation in `esphome/components/remote_base/nec_protocol.cpp` ([link](https://github.com/esphome/esphome/blob/dev/esphome/components/remote_base/nec_protocol.cpp)) instead of it just repeats the bytes from the "command" part of the signal.

The same goes to the decoder, it can not decode more than one command because it awaits for the command itself to be repeated, not the specific "repeat" type of signal.

To encode repeated signal you need to add a pause (low signal) for 40.5ms after the address+command transmission, and then start sending "repeat codes".
Repeat codeitself is `HEADER_HIGH_US` as 1 or "high" signal, then a space with a duration 1/2 of `HEADER_LOW_US` of 0 or "low", one `BIT_HIGH_US` as 1 or "high", and the last part is a long pause of 96.1875ms.

Each frame is 108ms.
Here is a picture to illustrate it
[image]

I myself don't know C++ good enough to resolve this issues in an elegant and laconic way, but I suggest this issue could be marked as "ready to code" or something, and I'm willing to test the exact implementation with my devices to prove that solution works correctly.

### Which version of ESPHome has the issue?

2024.12.2 

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

Any

### What platform are you using?

BK72XX

### Board

CBU

### Component causing the issue

remote_transmitter.transmit_nec

### YAML Config

```yaml
Currently I have a problem with Bluetooth reset command, which is "repeating" the Bluetooth button for 3 seconds.

   - platform: template
     name: "Klipsh set to Bluetooth"
     on_press:
       - remote_transmitter.transmit_nec:
           address: 0x7F80
           command: 0xF20D
           command_repeats: 30
```
```

### Anything in the logs that might be useful for us?

```txt
This is the of the RAW decoder which reads repeats from the remote:

[I][remote.raw:028]: Received Raw: 9060, -4525, 549, -562, 563, -593, 531, -594, 562, -562, 563, -562, 562, -563, 561, -563, 562, -1687, 563, -1686, 563, -1687, 531, -1707, 542, -1687, 562, -1687, 562, -1687, 563, -1686, 563, -562, 562, -1687, 562, -563, 562, -1690, 559, 
[I][remote.raw:041]:   -1704, 545, -562, 563, -562, 562, -563, 562, -562, 563, -562, 562, -1687, 562, -563, 562, -562, 563, -1687, 562, -1690, 559, -1690, 559, -1687, 562
[I][remote.raw:041]: Received Raw: 9042, -2275, 553
[I][remote.raw:041]: Received Raw: 9044, -2246, 571
[I][remote.raw:041]: Received Raw: 9050, -2279, 531
[I][remote.raw:041]: Received Raw: 9049, -2279, 563
[I][remote.raw:041]: Received Raw: 7012, -4284, 556
[I][remote.raw:041]: Received Raw: 9044, -2249, 559
[I][remote.raw:041]: Received Raw: 9050, -2251, 559
[I][remote.raw:041]: Received Raw: 9056, -2282, 556
```
```

### Additional information

Just thank you for the great project, it is really a blessing to have it.

DjordjeMandic, 2024-12-26:
@GRbit could you read the https://github.com/esphome/esphome/pull/8015 ?

GRbit, 2024-12-26:
@DjordjeMandic thank you! That was really fast :smiley: 
I left a few comments (hope you'll find them useful), but I don't have time to review it thoroughly right now, I'll come back to it in a few days.