esphome/issues #6605: Error and fail with  "Take Control" over newly added device

catdogmaus, 2024-12-25:
### The problem

When attempting to "Take Control" over newly added device I get error:

```INFO Reading configuration /config/esphome/esphome-web-469370.yaml...
Failed config

wifi: [source /config/esphome/esphome-web-469370.yaml:22]
  
  Please specify at least an SSID or an Access Point to create.
  domain: .local
  reboot_timeout: 15min
  power_save_mode: LIGHT
  fast_connect: False
  enable_btm: False
  enable_rrm: False
  passive_scan: False
  enable_on_boot: True
  networks: []
  use_address: esphome-web-469370.local
```

I have this with several ESP32-D and ESP32 devices I did attempted to add to HA ESPHome dashboard.
As workaround - when I open yaml created after that error and add manually 

`wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password`

I am able to install it and proceed with actual setup.
I have latest version at moment 2024.12.2 installed

### Which version of ESPHome has the issue?

2024.12.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.12.5

### What platform are you using?

ESP32-IDF

### Board

ESP-wroom-32, ESP32-wroom-D

### YAML Config

```yaml
esphome:
  name: esphome-web-469370
  friendly_name: Moretti
  min_version: 2024.11.0
  name_add_mac_suffix: false

esp32:
  board: esp32dev
  framework:
    type: esp-idf

# Enable logging
logger:

# Enable Home Assistant API
api:

# Allow Over-The-Air updates
ota:
- platform: esphome

wifi:
```

### Anything in the logs that might be useful for us?

```txt
2024-12-25 14:43:47,922 INFO Running command 'esphome --dashboard run /config/esphome/esphome-web-469370.yaml --device OTA'
2024-12-25 14:43:49,215 INFO 304 GET /devices (0.0.0.0) 7.80ms
2024-12-25 14:43:50,141 INFO Process exited with return code 2
```

Erzi-RA, 2024-12-27:
I have the same problem.

smithey253, 2024-12-27:
Same issue here too, worked fine using same devices and steps a couple of weeks ago.

vacano, 2024-12-28:
Good morning guys. I am experiencing an issue with setting up a ne T-Display S3 device to implement with Home Assistant. After "Prepare for first use" it all seems well. I can access the device via web on the designated IP. After I try to take control I get the following error message 

"INFO ESPHome 2024.12.2
INFO Reading configuration /config/esphome/esphome-web-54bf24.yaml...
Failed config

wifi: [source /config/esphome/esphome-web-54bf24.yaml:22]

  Please specify at least an SSID or an Access Point to create.
  domain: .local
  reboot_timeout: 15min
  power_save_mode: LIGHT
  fast_connect: False
  enable_btm: False
  enable_rrm: False
  passive_scan: False
  enable_on_boot: True
  networks: []
  use_address: esphome-web-54bf24.local. 

When I check the *.yaml file I can see that there are no wifi credential stored even though after the initial setup I entered those. After updating yaml with the wifi, device is no longer accessible.Tried with several devices and I get the same issue. I am really stressed out now thinking it is me who is doing it wrong... Thank for your help.

Pikestein, 2024-12-28:
I have the same problem too.

paranoidi, 2025-01-08:
Same, HA 2025.1.1 and completely new ESP32 install.

Faasteer, 2025-01-09:
Unfortunately, I have the exact same problem and the whole add-on is unusable because of it.

Faasteer, 2025-01-09:
Unfortunately, I have the same problem. The configuration for WiFi is completely missing, as well as some other parts. I managed to complete everything according to the screens on the site from homeautomationguy, but it's really annoying. 😢

HA 2025.1.1, completely new ESP Home install, esp32dev board

darkpixel, 2025-01-10:
The whole setup process seems a bit horrid to me.

Like...you go to https://esphome.io/projects/ and install a voice assistant or a speaker or whatever...and it loads it, connects to wireless, and then gives you a nice "easy button" to add it into Home Assistant.

...and amazingly it works....but then the ESP Home addon starts complaining that you have "discovered" devices and wants you to "take control" of them....and I guess some functions on the devices don't work because they can't really talk to Home Assistant.....so you take control of them, it fails, you edit the config to add the wireless settings, and then push....and you're left with a blank install and a wiped device....or even two devices that are actually the same physical device.  One was the original device name, but then it gets a new name when taking control.  Or something.

Frustrating.

ssieb, 2025-01-10:
There is no requirement to "take control" of it.  If it's working, then why are you trying to change it?  You don't need the addon at all unless you're wanting to make your own configurations.

darkpixel, 2025-01-10:
If there's no requirement to "take control" of it, why does the interface have a persistent prompt to get you to take control of it?

Are you saying there's an ESP Home component built-in to HomeAssistant's devices and configuration that handles the devices, but there's also an identically named but sorta unrelated ESP Home add-on in HomeAssistant's add-on section that's for doing more advanced stuff that most people don't need and will be confused by having installed?

darkpixel, 2025-01-10:
Ok--that makes sense and explains why I was so confused and having so much trouble with it.  I removed the add-on and things continue to work.  I thought I had to have it installed for some sort of API communication between the devices and HA.

ssieb, 2025-01-10:
It has never been that.  There was a communication path added somewhat recently that allows HA to get the encryption keys and esphome version from the addon, but that's all.

nikee73, 2025-01-10:
Same problem for me. Cant take control of my new installed esp. It connected to wifi

paranoidi, 2025-01-17:
Added ESP32 and this time used visible SSID straight away and it failed again the same way so it's was not because of hidden ssid like I previously guessed.

Bigfellas, 2025-01-18:
Same problem for me on an esp32

kanschat, 2025-01-26:
Same problem with esp32-c3

jam-eth, 2025-01-29:
Same issue with a WeMos D1 mini (ESP8266EX) 

I tried to add my WiFi SSID and password to the yaml, not using !secrets and the config seemed to build and upload, but the device was never heard from again  . . .

masto, 2025-02-03:
It appears this was broken by https://github.com/esphome/esphome/commit/687553a285fc13d4b17a75a84050e5c0e6ba2cb1 in March 2024. Prior to this commit, the code in `import_config` [delegated](https://github.com/esphome/esphome/blob/97a71482a908f497e70d99369798fbeb38c1c5a5/esphome/components/dashboard_import/__init__.py#L130-L133) to [wizard_file](https://github.com/esphome/esphome/blob/5108b9a8b7ca7bc3b541fd743d81821c47a6c000/esphome/wizard.py#L162-L167), which wrote the ssid and password to the generated config.

[Now](https://github.com/esphome/esphome/blob/687553a285fc13d4b17a75a84050e5c0e6ba2cb1/esphome/components/dashboard_import/__init__.py#L100), devices initialized through ESPHome Web have `import_full_config` set, the YAML is just copied from GitHub verbatim (e.g. https://raw.githubusercontent.com/esphome/firmware/refs/heads/main/esphome-web/esp32.yaml) and the user ends up with an invalid config that can't be installed.

I was going to try to fix this, but while I was able to trace out what this code is doing, the intention is unclear to me. It's a very complex path with a lot of twists and turns to get from a new device to fully adopted.

I suspect maybe this only happens if you set up with ESPHome Web and then try to take control of the discovered device, which might explain why it's been broken for almost a year with relatively little fanfare. Nevertheless, this is the supported path that users are sent down if they don't have an SSL cert or WebSerial-capable browser and can't plug directly into the machine running ESPHome, so it is unfortunate that it is in this state.

opencollectoroutput, 2025-02-13:
For anyone else what worked for me was to create the config in builder and then load it with esphome web.
Steps:
Click "+ new device" in builder
click "open esphome web" and leave that tab open for later
click "+ new device" again, then click "continue"
name the configuration and click next
select the device type
click install
select "manual download"
once compiling is done select "Factory Format"
back in the esphome web tab connect to the serial port
click "install" and choose the downloaded file
click install
close the esphome web tab

The new device should show up in builder now.

Kn1ghtHOK, 2025-02-18:
I had this problem -- then I looked at the yaml. When you click the install button, it will fail like the original poster said, and then you have to click edit. The wifi field will be empty. Replace it with this:

wifi:
  ssid: (your ssid)
  password: (your password)

This is what worked for me

Kn1ghtHOK, 2025-02-18:
To fix this issue: after clicking install, it will fail, press edit and the wifi field will be empty. Replace it with this:

wifi:
  ssid: (your SSID)
  password: (your Password)

This is what worked for me.

vacano, 2025-02-19:
Nothing works for me I havent been so stressed in a while, I dont know how many times I have tried all over again and no success..

mikquadrat, 2025-03-14:
> It appears this was broken by [esphome/esphome@687553a](https://github.com/esphome/esphome/commit/687553a285fc13d4b17a75a84050e5c0e6ba2cb1) in March 2024. Prior to this commit, the code in `import_config` [delegated](https://github.com/esphome/esphome/blob/97a71482a908f497e70d99369798fbeb38c1c5a5/esphome/components/dashboard_import/__init__.py#L130-L133) to [wizard_file](https://github.com/esphome/esphome/blob/5108b9a8b7ca7bc3b541fd743d81821c47a6c000/esphome/wizard.py#L162-L167), which wrote the ssid and password to the generated config.
> 
> [Now](https://github.com/esphome/esphome/blob/687553a285fc13d4b17a75a84050e5c0e6ba2cb1/esphome/components/dashboard_import/__init__.py#L100), devices initialized through ESPHome Web have `import_full_config` set, the YAML is just copied from GitHub verbatim (e.g. https://raw.githubusercontent.com/esphome/firmware/refs/heads/main/esphome-web/esp32.yaml) and the user ends up with an invalid config that can't be installed.
> 
> I was going to try to fix this, but while I was able to trace out what this code is doing, the intention is unclear to me. It's a very complex path with a lot of twists and turns to get from a new device to fully adopted.
> 
> I suspect maybe this only happens if you set up with ESPHome Web and then try to take control of the discovered device, which might explain why it's been broken for almost a year with relatively little fanfare. Nevertheless, this is the supported path that users are sent down if they don't have an SSL cert or WebSerial-capable browser and can't plug directly into the machine running ESPHome, so it is unfortunate that it is in this state.

Thanks for summarizing this in a perfect way. It drove me crazy to adopt a new ESP "outside" the browser in which HA is running. Hope this can get changed back to the previous "fill the config YAML" instead of "copying" the config file. 
Does anyone have a recommendation how to flash the ESP from within the HA browser?