esphome/issues #6606: AC dimmer issue sharing pin

alnavasa, 2024-12-26:
### The problem

Can't connect and modify code on ad AC dimmer from here: https://esphome.io/components/output/ac_dimmer.html
Issue is that as per code on that link you should be able to share pin between multiple dimerized lights for zero_cross_pin, but ESPHOME detects it as an error, and can't past that. 

I was working on previous ESPHOME versions (maybe 2 - 3 years ago)

### Which version of ESPHome has the issue?

2024.12.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.12.5

### What platform are you using?

ESP8266

### Board

esp8266:   board: esp01_1m

### Component causing the issue

AC Dimmer Component

### YAML Config

```yaml
substitutions:
  node_name: pergola-dimmer
  device_name: Pérgola dimmer

esphome:
  name: $node_name
  comment: $device_name
  

esp8266:
  board: esp01_1m

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: !secret api_key
ota:
- platform: esphome
  password: !secret ota_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "$node_name"
    password: !secret fallback_password

captive_portal:
    
    
    

text_sensor:
  - platform: template
    name: Uptime $device_name
    id: uptime_human
    entity_category: diagnostic
    icon: mdi:clock-start
    disabled_by_default: true  # Deshabilitada por defecto

  - platform: wifi_info
    ip_address:
      name: $device_name IP
      icon: mdi:wifi-cog
    ssid:
      name: $device_name SSID
      icon: mdi:wifi-cog
    bssid:
      name: $device_name BSSID
      icon: mdi:wifi-cog

sensor:
  - platform: wifi_signal
    name: "WiFi Signal $device_name"
    update_interval: 60s
    disabled_by_default: true  # Deshabilitada por defecto 

  - platform: uptime
    name: Uptime $device_name System
    id: uptime_sensor
    update_interval: 60s
    entity_category: diagnostic
    disabled_by_default: true  # Deshabilitada por defecto
    on_raw_value:
      then:
        - text_sensor.template.publish:
            id: uptime_human
            state: !lambda |-
              int seconds = round(id(uptime_sensor).raw_state);
              int days = seconds / (24 * 3600);
              seconds = seconds % (24 * 3600);
              int hours = seconds / 3600;
              seconds = seconds % 3600;
              int minutes = seconds /  60;
              seconds = seconds % 60;
              return (
                (days ? to_string(days) + "d " : "") +
                (hours ? to_string(hours) + "h " : "") +
                (minutes ? to_string(minutes) + "m " : "") +
                (to_string(seconds) + "s")
              ).c_str();

status_led:
  pin:
    number: GPIO2
    

button:
  - platform: restart
    name: "$device_name Restart"
    
  - platform: safe_mode
    name: "$device_name Room Restart (Safe Mode)"

binary_sensor:
  - platform: status
    name: "Status $device_name"
    

light:
  - platform: monochromatic
    output: dimmer1
    name: Cocina 1
    icon: mdi:faucet-variant
    id: cocina_1
    restore_mode: RESTORE_DEFAULT_OFF
    
  - platform: monochromatic
    output: dimmer2
    name: Cocina 2
    icon: mdi:faucet-variant
    id: cocina_2
    restore_mode: RESTORE_DEFAULT_OFF
    
  - platform: monochromatic
    output: dimmer3
    name: Dimmerized Light 3
    id: dimmerized_3
    restore_mode: RESTORE_DEFAULT_OFF
    disabled_by_default: true  # Deshabilitada por defecto
    
  - platform: monochromatic
    output: dimmer4
    name: Dimmerized Light 4
    id: dimmerized_4
    restore_mode: RESTORE_DEFAULT_OFF
    disabled_by_default: true  # Deshabilitada por defect
    

output:
  - platform: ac_dimmer
    id: dimmer1
    gate_pin: GPIO12
    min_power: 0.15
    max_power: 1
    zero_cross_pin:
      number: GPIO13
      mode:
        input: true
      inverted: yes

  - platform: ac_dimmer
    id: dimmer2
    gate_pin: GPIO14
    min_power: 0.15
    max_power: 1
    zero_cross_pin:
      number: GPIO13
      mode:
        input: true
      inverted: yes

  - platform: ac_dimmer
    id: dimmer3
    gate_pin: GPIO4
    zero_cross_pin:
      number: GPIO13
      mode:
        input: true
      inverted: yes
    

  - platform: ac_dimmer
    id: dimmer4
    gate_pin: GPIO5
    zero_cross_pin:
      number: GPIO13
      mode:
        input: true
      inverted: yes
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2024.12.2
INFO Reading configuration /config/esphome/pergola-dimmer.yaml...
Failed config

output.ac_dimmer: [source /config/esphome/pergola-dimmer.yaml:168]
  
  Pin 13 is used in multiple places.
  platform: ac_dimmer
  id: dimmer1
  gate_pin: 
    number: 12
    mode: 
      output: True
      input: False
      open_drain: False
      pullup: False
      pulldown: False
      analog: False
    inverted: False
output.ac_dimmer: [source /config/esphome/pergola-dimmer.yaml:179]
  
  Pin 13 is used in multiple places.
  platform: ac_dimmer
  id: dimmer2
  gate_pin: 
    number: 14
    mode: 
      output: True
      input: False
      open_drain: False
      pullup: False
      pulldown: False
      analog: False
    inverted: False
output.ac_dimmer: [source /config/esphome/pergola-dimmer.yaml:190]
  
  Pin 13 is used in multiple places.
  platform: ac_dimmer
  id: dimmer3
  gate_pin: 
    number: 4
    mode: 
      output: True
      input: False
      open_drain: False
      pullup: False
      pulldown: False
      analog: False
    inverted: False
output.ac_dimmer: [source /config/esphome/pergola-dimmer.yaml:200]
  
  Pin 13 is used in multiple places.
  platform: ac_dimmer
  id: dimmer4
  gate_pin: 
    number: 5
    mode: 
      output: True
      input: False
      open_drain: False
      pullup: False
      pulldown: False
      analog: False
    inverted: False
```

### Additional information

discussion thread currently closed:

https://github.com/esphome/feature-requests/issues/278

alnavasa, 2024-12-26:
chat gpt suggests this bypass:

```
globals:
  - id: zero_cross_pin
    type: int
    initial_value: 13  # GPIO13

output:
  - platform: ac_dimmer
    id: dimmer1
    gate_pin: GPIO12
    min_power: 0.15
    max_power: 1
    zero_cross_pin:
      number: !lambda 'return id(zero_cross_pin);'
      mode:
        input: true
      inverted: yes

  - platform: ac_dimmer
    id: dimmer2
    gate_pin: GPIO14
    min_power: 0.15
    max_power: 1
    zero_cross_pin:
      number: !lambda 'return id(zero_cross_pin);'
      mode:
        input: true
      inverted: yes

  - platform: ac_dimmer
    id: dimmer3
    gate_pin: GPIO4
    min_power: 0.15
    max_power: 1
    zero_cross_pin:
      number: !lambda 'return id(zero_cross_pin);'
      mode:
        input: true
      inverted: yes

  - platform: ac_dimmer
    id: dimmer4
    gate_pin: GPIO5
    min_power: 0.15
    max_power: 1
    zero_cross_pin:
      number: !lambda 'return id(zero_cross_pin);'
      mode:
        input: true
      inverted: yes
```

red4dj, 2025-05-16:
Having this same issue. Docs clearly state "you can have several dimmers controlled with the same zero cross detector, in such case duplicate the zero_cross_pin config on each output" but device builder fails to compile.

red4dj, 2025-05-16:
I think the solution is as easy as adding `allow_other_uses: true` to suppress the warning. This should be noted in docs.
```yaml
    zero_cross_pin:
      number: D0
      allow_other_uses: true
```

closed 2026-09-11 by alnavasa