esphome/issues #6607: Dashboard unavailable (127.0.0.1) after moving ESPHome to external machine

IgnacioHR, 2024-12-26:
### The problem

Hi,

I decided to move ESPHome from inside Home Assistant (docker) into a separate virtual machine (I use proxmox at home).

Previously I had ESPHome integration installed in Home Assistant.
Once the virtual machine is working and everything looks fine I decided to remove ESPHome integration from Home Assistant because it is no longer needed.

After uninstalling the ESPHome integration I see two errors in the error registry when I start Home Assistant:

```
Registrador: homeassistant.components.esphome.dashboard
Fuente: components/esphome/dashboard.py:122
integración: ESPHome (documentación, problemas)
Ocurrió por primera vez: 17:12:05 (1 ocurrencias)
Última vez registrado: 17:12:05

Dashboard unavailable; skipping reauth: Cannot connect to host 127.0.0.1:62944 ssl:default [Connect call failed ('127.0.0.1', 62944)]
```

and 

```
Registrador: homeassistant.components.esphome.coordinator
Fuente: helpers/update_coordinator.py:392
integración: ESPHome (documentación, problemas)
Ocurrió por primera vez: 17:12:05 (1 ocurrencias)
Última vez registrado: 17:12:05

Error requesting ESPHome Dashboard data: Cannot connect to host 127.0.0.1:62944 ssl:default [Connect call failed ('127.0.0.1', 62944)]
```

How can I get rid of these errors? and/or could I just indicate the IP address of my virtual machine installation so it is pointing there rather than a virtual machine inside Home Assistant?

### Which version of ESPHome has the issue?

2024.12.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.12.2

### What platform are you using?

ESP32

### Board

N/A

### Component causing the issue

N/A

### YAML Config

```yaml
N/A, the problem is in the ESPHome integration, after removing it from Home Assistant
```

### Anything in the logs that might be useful for us?

```txt
The two errors indicated in the description
```

uli-rpi, 2024-12-27:
I am using exactly your configuration with an external configuration and Proxmox. It works fine, so when I get back home I can give you details.

IgnacioHR, 2024-12-30:
Hi @uli-rpi thanks for jumping in.

I finally solved it. 

I went to the source code and noticed from where the host and port are taken. 

The file is at `config/.storage/esphome.dashboard` and the change is obvious.

I've been unable to find a way in the GUI to change the values that's why I decided to go the hacker path.

Hope this information helps others

closed 2024-12-30 by IgnacioHR

uli-rpi, 2024-12-31:
@IgnacioHR 

My solution:
1. I have installed esphome as lxc container with proxmox using the script [from tteck ](https://community-scripts.github.io/ProxmoxVE/scripts)
2. I have also installed VS Code Server in the same LXC, also using a script from tteck
3. Esphome Dashboard is directly accessible with the IP of the LXC container port 6052, eg. http://192.168.178.100:6052
4. VS Code Server with port 8680, eg. http://192.168.178.100:8680
5. Afterwards only the source code (the yaml file(s)) has to be copied from the original location in the esphome addon (/config/esphome/xyz.yaml) into the LXC container to config/. 
6. Esphome will automatically recognize the new file and display it in the dashboard 
7. Add a new dashboard via the settings in HomeAssitant. Use the option Webpage, use the ip:port of the proxmox esphome installation, activate “show in sidebar”, repeat this with the ip:port of the VS Code Server.
8. Now you can access esphome within HomeAssitant via the side bar