esphome/issues #6608: HBridge restore mode does not work

felipecrs, 2024-12-26:
### The problem

I am trying to convert my breaker to use the new hbridge component:

- https://devices.esphome.io/devices/Tomzn-TOB9-VAP

But `restore_mode` doesn't work. Is this a known issue? There's nothing in documentation about it.

### Which version of ESPHome has the issue?

2024.12.2

### What type of installation are you using?

pip

### What platform are you using?

BK72XX

### Board

cbu

### Component causing the issue

hbridge

### YAML Config

```yaml
switch:
  - platform: hbridge
    id: hbridge0
    on_pin: GPIO24
    off_pin: GPIO26
    pulse_length: 60ms
    wait_time: 30ms
```

felipecrs, 2024-12-26:
@dwmw2, since you implemented the H-Bridge component (https://github.com/esphome/esphome/pull/7421), maybe you'd be the best one to answer this.

felipecrs, 2024-12-26:
My workaround:

```yaml
esphome:
  project:
    name: TOMZN.TOB9-VAP
    version: v1
  on_boot: 
    priority: 900
    then:
      switch.turn_off: hbridge0
```

felipecrs, 2024-12-26:
Assuming this is intended, I added this to the device page:

_Uses an [H-Bridge](https://esphome.io/components/switch/hbridge.html) for controlling the output instead of a standard relay. One key difference is that the H-Bridge retains the last position upon power off without the need of `restore_mode`_

- https://github.com/esphome/esphome-devices/pull/961

dwmw2, 2024-12-27:
The problem is that we don't know the physical state of the relay when restarting. One option is to force it on/off according to the state that software believes it's in, but mostly I guess we'd prefer to leave it where it is and if the software is wrong about whether it's on/off, that's better than actually *switching* it to match the software's belief.

dwmw2, 2024-12-27:
But for the on/off restore modes we should actually trigger it, I suppose?

felipecrs, 2024-12-27:
I believe support for all restore modes can be added.

1. Restoring the last known state should work 99.9% of the times. If the chip can remember the state it was before shutdown, the bridge would not move on its own. And then, trying to energize the pin on initialization to guarantee that position will cause no harm if it's already on such position (and honestly, even for the 0.1% of the cases where it's not, I'd say it's still worth it (or at least give users the choice).

2. As you mentioned, always on or always off is kind of no brainer too. If the users want, they should be able to guarantee a position upon initialization. The only caveat is that the bridge can flick on initialization, which should be documented.

That's my opinion.

closed 2025-03-07 by ssieb