esphome/issues #6609: ESPHome esp32_ble_tracker on_ble_advertise, manufacturer data is missing company ID?

wire67, 2024-12-27:
### The problem

ESPHome in `esp32_ble_tracker` , `on_ble_advertise` , manufacturer data is missing company ID?
That is the 2 first bytes of the payload.

Example: We can observe the raw manufacturer data `(0x)07FF4C0012029400` with nRF android app. 
`07` is the length
`FF` custom manufacturer data 
`4C00` company ID little endian 
`12029400` manufacturer data custom payload 
ESPHome reports only the part starting after the manufacturer ID, that is `(0x)12029400`. 
How can we get the company ID? 

### Which version of ESPHome has the issue?

2024.12.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.12

### What platform are you using?

ESP32

### Board

devkit

### Component causing the issue

esp32_ble_tracker

### YAML Config

```yaml
# Bluetooth Low Energy Tracker
esp32_ble_tracker:
  id: myesp32_ble_tracker
  on_ble_advertise:
    - lambda: |-
        if (x.get_rssi() > id(strongest_rssi)) {
          id(strongest_rssi) = x.get_rssi();
          id(strongest_address) = x.address_str();
          
          // Check for manufacturer data and publish it
          auto manufacturer_data = x.get_manufacturer_datas();
          if (manufacturer_data.size() > 0) {
            std::string data_str;
            for (auto byte : manufacturer_data[0].data) {
              char buf[3];
              sprintf(buf, "%02X", byte);
              data_str += buf;
            }
            id(manufacturer_data_sensor).publish_state(data_str);
            
            // Extract and publish manufacturer ID (company identifier)
            if (manufacturer_data[0].data.size() >= 2) {
              uint16_t company_id = (manufacturer_data[0].data[1] << 8) | manufacturer_data[0].data[0];
              char company_id_str[5];
              sprintf(company_id_str, "%04X", company_id);
              id(company_identifier_sensor).publish_state(company_id_str);
            }
          }
        }

globals:
  - id: strongest_rssi
    type: int
    initial_value: '-100'
  - id: strongest_address
    type: std::string
    initial_value: ""

sensor:
  - platform: template
    name: "RSSI"
    id: strongest_ble_rssi
    unit_of_measurement: "dBm"
    icon: "mdi:bluetooth"

text_sensor:
  - platform: template
    name: "Address"
    id: strongest_ble_address
    icon: "mdi:bluetooth"
  - platform: template
    name: "Manufacturer Data"
    id: manufacturer_data_sensor
    icon: "mdi:package-variant-closed"
  - platform: template
    name: "Company Identifier"
    id: company_identifier_sensor
    icon: "mdi:factory"

interval:
  - interval: 10s
    then:
      - lambda: |-
          // Report the strongest device to Home Assistant
          id(strongest_ble_rssi).publish_state(id(strongest_rssi));
          id(strongest_ble_address).publish_state(id(strongest_address).c_str());
          // Reset for the next scan
          id(strongest_rssi) = -77;

```
```

impartialcricket, 2025-01-26:
I'm using this logic and it prints all that nRF app on iPhone displays.

```
            ESP_LOGI("ble_adv", "  Advertised manufacturer data:");
            for (auto data : x.get_manufacturer_datas()) {
              ESP_LOGI("ble_adv", "    - %s: (%s)", data.uuid.to_string().c_str(), hexencode(data.data).c_str());
            }
```

I'm not affiliated with the project but hopefully the above assist with the issue.

wire67, 2025-01-27:
thanks so much @impartialcricket this does contain the manufacturer ID: `data.uuid.to_string().c_str()` 🎉