esphome/issues #6611: sht4x CRC8 Checksum invalid sensor fail after update

jdohn15, 2024-12-28:
### The problem

STS40 sensor not working and throwing error after update.

I updated esphome from a very old version, 2022.3.0, to the current 2024.12.2. I have a lot of esp devices on my network that use the [STS40](https://www.mouser.com/ProductDetail/403-STS40-CD1B-R3) sensor. It is the [SHT4X](https://esphome.io/components/sensor/sht4x.html) without the humidity and heater. In my old installation of esphome i just did not add the humidity component and it ran fine. 

Then i updated my esp32 so I could see the logs, and then it started throwing the error. Worked fine before. I tried forking and reverting the old sht4x code but it did not work. 

### Which version of ESPHome has the issue?

2024.12.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.12.5

### What platform are you using?

ESP32

### Board

custom pcb

### Component causing the issue

i2c SHT4X/STS40

### YAML Config

```yaml
esphome:
  name: jd-small-lamp
  on_boot:
    priority: -10  # Run after everything else initializes
    then:
      - number.set:
          id: max_fan_speed_slider
          value: !lambda 'return id(max_fan_speed);'
      - number.set:
          id: temp_threshold_slider
          value: !lambda 'return id(temp_threshold);'
      - number.set:
          id: temp_fan_increase_slider
          value: !lambda 'return id(temp_fan_increase);'
      - number.set:
          id: default_fan_speed_slider
          value: !lambda 'return id(default_fan_speed);'

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:
  level: VERBOSE  # Set global level to VERBOSE to allow detailed component logs
  logs:
    sensor: WARN        # Suppress detailed sensor logs
    fan: WARN           # Suppress detailed fan logs
    template.number: WARN  # Suppress slider-related logs
    wifi: INFO          # Explicitly include Wi-Fi logs

# Enable Home Assistant API
api:

ota:
  platform: esphome
  password: "xxxxxxxxxx"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  manual_ip:
    static_ip: 192.168.1.119  # EDIT: Update static IP
    gateway: 192.168.1.1
    subnet: 255.255.255.0

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Jd-Small-Lamp Fallback Hotspot"
    password: "xxxxxxxx"

captive_portal:
    
# I2C Configuration for STS40 Sensor
i2c:
  sda: GPIO21
  scl: GPIO22
  scan: true
  frequency: 10kHz  # recommended range is 50-200kHz
  timeout: 1ms

# STS40 Temperature Sensor
sensor:
  - platform: sht4x
    address: 0x46
    temperature:
      name: "jd-small-lamp_LEDboardTemp"  # EDIT: Update sensor name
      id: sts40_temp
      unit_of_measurement: "°C"
      filters:
        - lambda: |-
            static float last_temp = NAN;
            if (!isnan(last_temp) && abs(x - last_temp) > 3.0) {
              return last_temp;  // Filter out abrupt changes
            }
            last_temp = x;
            return x;
    update_interval: 1s
  - platform: sht4x
    address: 0x44  # Address for Driver board temperature sensor
    temperature:
      name: "jd-small-lamp_DriverBoardTemp"
      id: driver_board_temp
      unit_of_measurement: "°C"
      filters:
        - lambda: |-
            static float last_temp = NAN;
            if (!isnan(last_temp) && abs(x - last_temp) > 3.0) {
              return last_temp;  // Filter out abrupt changes
            }
            last_temp = x;
            return x;
    update_interval: 1s

globals:
  - id: max_fan_speed
    type: float
    initial_value: "50.0"  # Default maximum fan speed at 100% brightness
  - id: temp_threshold
    type: float
    initial_value: "50.0"  # Default temperature threshold in °C
  - id: temp_fan_increase
    type: float
    initial_value: "20.0"  # Default additional fan speed percentage when over temp
  - id: default_fan_speed
    type: float
    initial_value: "15.0"  # Default fan speed when light turns on

# Outputs
output:
  - platform: ledc
    pin: GPIO26
    frequency: 30000 Hz
    id: pwm_cool_white
  - platform: ledc
    pin: GPIO25
    frequency: 30000 Hz
    id: pwm_warm_white
  - platform: ledc
    pin: GPIO33
    frequency: 30000 Hz
    id: pwm_red
  - platform: ledc
    pin: GPIO27
    frequency: 25000 Hz
    id: pwm_fan_output
  - platform: ledc
    frequency: 3 Hz
    id: pwm_dummy_green
    pin: GPIO4
    inverted: true
  - platform: ledc
    frequency: 3 Hz
    id: pwm_dummy_blue
    pin: GPIO16
    inverted: true

# Light
light:
  - platform: rgbww
    name: "jd-small-lamp_light"  # EDIT: Update light name
    id: j_rgbww_light
    red: pwm_red
    green: pwm_dummy_green
    blue: pwm_dummy_blue
    cold_white: pwm_cool_white
    warm_white: pwm_warm_white
    cold_white_color_temperature: 3000 K
    warm_white_color_temperature: 2000 K
    default_transition_length: 300ms
    on_turn_on:
      then:
        - fan.turn_on:
            id: fanboy
            speed: !lambda 'return id(default_fan_speed);'  # Default fan speed from slider
    on_turn_off:
      then:
        - fan.turn_off:
            id: fanboy

# Fan
fan:
  - platform: speed
    id: "fanboy"  
    name: "jd-small-lamp_fan"  # EDIT: Update fan name
    output: pwm_fan_output

# Number Sliders
number:
  - platform: template
    name: "jd-small-lamp_Fan Speed at 100% Brightness"  # EDIT: Update slider name
    id: max_fan_speed_slider
    min_value: 15.0
    max_value: 100.0
    step: 1.0
    unit_of_measurement: "%"
    optimistic: true
    set_action:
      - lambda: |-
          id(max_fan_speed) = x;

  - platform: template
    name: "jd-small-lamp_OverTemp Threshold"  # EDIT: Update slider name
    id: temp_threshold_slider
    min_value: 30.0
    max_value: 90.0
    step: 1.0
    unit_of_measurement: "°C"
    optimistic: true
    set_action:
      - lambda: |-
          id(temp_threshold) = x;

  - platform: template
    name: "jd-small-lamp_Fan Speed Increase OverTemp"  # EDIT: Update slider name
    id: temp_fan_increase_slider
    min_value: 0.0
    max_value: 50.0
    step: 1.0
    unit_of_measurement: "%"
    optimistic: true
    set_action:
      - lambda: |-
          id(temp_fan_increase) = x;

  - platform: template
    name: "jd-small-lamp_Default Fan Speed"  # EDIT: Update slider name
    id: default_fan_speed_slider
    min_value: 0.0
    max_value: 50.0
    step: 1.0
    unit_of_measurement: "%"
    optimistic: true
    set_action:
      - lambda: |-
          id(default_fan_speed) = x;

# Update Fan Speed Based on Light Brightness and Temperature
interval:
  - interval: 100ms
    then:
      - lambda: |-
          if (id(j_rgbww_light).current_values.is_on()) {
            // Calculate overall brightness (0 to 1 scale)
            float overall_brightness = id(j_rgbww_light).current_values.get_brightness();

            // Use separate variables for fan speed calculations
            float target_fan_speed = id(default_fan_speed);  // Default fan speed
            float max_brightness_speed = id(max_fan_speed);  // Max fan speed at 100% brightness
            float current_temperature = id(sts40_temp).state;  // Current temperature

            // Adjust fan speed for overtemperature
            if (current_temperature > id(temp_threshold)) {
              if (overall_brightness >= 0.99) {
                // At 100% brightness, use max brightness speed + temperature increment
                target_fan_speed = max_brightness_speed + id(temp_fan_increase);
              } else {
                // Below 100% brightness, increment default fan speed by slider value
                target_fan_speed = id(default_fan_speed) + id(temp_fan_increase);
              }
            } else if (overall_brightness >= 0.99) {
              // At 100% brightness but no overtemperature, use max brightness speed
              target_fan_speed = max_brightness_speed;
            }

            // Cap the fan speed at 100%
            target_fan_speed = std::min(target_fan_speed, 100.0f);

            // Apply the fan speed
            auto call = id(fanboy).turn_on();
            call.set_speed(target_fan_speed);
            call.perform();
          } else {
            // Turn off fan when the light is off
            id(fanboy).turn_off();
          }
```

### Anything in the logs that might be useful for us?

```txt
[21:47:03][I][i2c.arduino:096]: Results from i2c bus scan:
[21:47:03][I][i2c.arduino:102]: Found i2c device at address 0x44
[21:47:03][I][i2c.arduino:102]: Found i2c device at address 0x46

[21:47:03][E][sensirion_i2c:026]: CRC8 Checksum invalid at pos 1! 0xFF != 0xAC
[21:47:03][D][sht4x:080]: Sensor read failed
```

### Additional information

No hardware change. No code change except changing pwm output on dummy colors and adding ota:
  platform: esphome. Sensors are being picked up. 

Seems very similar to this issue #3310  

Thank you!

jdohn15, 2024-12-28:
I got the sensors to work again with skipping crc validation in [i2c_sensirion.cpp](https://github.com/esphome/esphome/compare/dev...jdohn15:esphome:dev?diff=unified&w=#diff-51f2bef95767c36a96051e11cd72ed5e3a31aefbc9e2faca1e0f33d1de662437).

unsure what to make of that...

AbianTang, 2025-01-02:
frequency: 400kHz

AbianTang, 2025-01-02:
#  frequency: 10kHz  # recommended range is 50-200kHz
Or try deleting it.