esphome/issues #6613: Modbus Controller: Only Write but not Read Holding Register (0x06)

autox86, 2024-12-28:
### The problem

Hi Dev Team :)

**Uncomfortable situation:**
I would need an option to stopp reading the holding register as it only supports writing to it.

My device is having an holding registers with different commands (2000 - 0001/0005)

Modbus_controller tries reading these write-only holding register and receives a valid: 0x3 - Illegal value error

Here the command I like to send:
03 - Slave ID (Device)
06 - Write Holding Register

20 - 2000 is the address 
00 - of the register

00 -  Start the
01 -  Device

OR

00 - Stopp the
05 - Device

So actually I am sending these requests: (where 43 and CA are CRC)
03 06 **20 00 00 01** _43 CA_ 

But somehow Modbus Controller is also trying to do a read on the same holding register:
03.**03**.20.00.00.01.8E.28

**Side effect**
The controller stopps reacting on register 2000 after the first enable / disable.
Other register can still be controlled. (e.g. 1004)

Honestly appreciate your help. :)

### Which version of ESPHome has the issue?

2024.12.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.12.1

### What platform are you using?

ESP32

### Board

Astra from rocketcontroller.com

### Component causing the issue

Modbus_Controller Switch

### YAML Config

```yaml
switch:
- platform: modbus_controller
    modbus_controller_id: modbus_frequency_inverter
    address: 0x2000
    register_type: holding
    name: "FU An"
    write_lambda: |-
      ESP_LOGD("main","Modbus Switch incoming state = %f",x);
      // return false ; // use this to just change the value
      payload.push_back(0x03); // device address
      payload.push_back(0x06); // force single register
      payload.push_back(0x20); // high byte address of the coil
      payload.push_back(0x00); // low byte address of the coil
      payload.push_back(0x00); // command      
      payload.push_back(0x01); // command
      return true;

  - platform: modbus_controller
    modbus_controller_id: modbus_frequency_inverter
    address: 0x2000
    register_type: holding
    name: "FU Aus"
    write_lambda: |-
      ESP_LOGD("main","Modbus Switch incoming state = %f",x);
      // return false ; // use this to just change the value
      payload.push_back(0x03); // device address
      payload.push_back(0x06); // force single register
      payload.push_back(0x20); // high byte address of the coil
      payload.push_back(0x00); // low byte address of the coil
      payload.push_back(0x00); // command      
      payload.push_back(0x05); // command
      return true;
```

### Anything in the logs that might be useful for us?

```txt
[00:13:04][W][component:237]: Component modbus_controller took a long time for an operation (114 ms).
[00:13:04][W][component:238]: Components should block for at most 30 ms.
[00:13:04][V][modbus_controller:044]: Sending next modbus command to device 3 register 0x2000 count 1
[00:13:04][V][modbus:223]: Modbus write: 03.03.20.00.00.01.8E.28 (8)
[00:13:04][V][modbus_controller:569]: Command sent 3 0x2000 1 send_count: 1
[00:13:04][D][modbus:136]: Modbus error function code: 0x83 exception: 2
[00:13:04][E][modbus_controller:094]: Modbus error function code: 0x3 exception: 2 
[00:13:04][E][modbus_controller:103]: Modbus error - last command: function code=0x3  register address = 0x2000  registers count=1 payload size=0
[00:13:04][V][modbus:159]: Clearing buffer of 4 bytes - parse succeeded
[00:13:05][D][switch:016]: 'FU An' Turning OFF.
[00:13:05][D][main:398]: Modbus Switch incoming state = 1.815804
[00:13:05][V][modbus_controller.switch:062]: Value overwritten by lambda
[00:13:05][V][modbus_controller.switch:070]: Modbus Switch write raw: 03.06.20.00.00.01 (6)
[00:13:05][V][modbus_controller:044]: Sending next modbus command to device 3 register 0x00 count 0
[00:13:05][V][modbus:244]: Modbus write raw: 03.06.20.00.00.01 (6)
[00:13:05][V][modbus_controller:569]: Command sent 0 0x0 0 send_count: 1
[00:13:05][V][modbus_controller:081]: Modbus response queued
[00:13:05][V][modbus:159]: Clearing buffer of 7 bytes - parse succeeded
[00:13:05][V][modbus_controller:089]: Process modbus response for address 0x0 size: 4
[00:13:05][V][modbus_controller:389]: Command ACK 0x2000 0 
[00:13:08][V][modbus_controller:232]: Updating modbus component
[00:13:08][V][modbus_controller:199]: Range : 1000 Size: 4 (3) skip: 0
[00:13:08][V][modbus_controller:199]: Range : 1005 Size: 3 (3) skip: 0
[00:13:08][V][modbus_controller:199]: Range : 2000 Size: 1 (3) skip: 0
```

synthesis77, 2025-05-10:
I'm curious too -- and have a similar question.  I have a bunch of variables that *can* be read, but I don't want them to unless written to (or after an initial load).   (namely, the configuration variables for a solar inverter that change almost never)

gotnone, 2025-05-20:
I think I see a similar issue with a yaml configuration with a `register_type: coil` modbus switch.

```
modbus_controller:
- id: modbus_aircon
  address: 0x1   ## address of the Modbus slave device on the bus
  modbus_id: modbus1
  update_interval: 5s
  setup_priority: -10

switch:
  - platform: modbus_controller
    modbus_controller_id: modbus_aircon
    name: "aircon"
    register_type: coil
    address: 0x0002
    bitmask: 1
```

In that case the modbus_controller code is creating a read coil 0x1 command (and poorly at that, if the CRC errors on my modbus slave are any indication).  After hours of trying to understand the convoluted modbus_controller code, it seems the underlying problem is in the `ModbusController::update_range_` function in modbus_controller.cpp.  In that function every range, `r` where the `r.register_type == ModbusRegisterType::CUSTOM` condition is **FALSE** gets queued as a **read command**:
```
queue_command(ModbusCommandItem::create_read_command(this, r.register_type, r.start_address, r.register_count));
```

In the case for a `platform: modbus_controller` switch this creates a read coil command 0x1 instead of a write single coil command 0x5.  In your case it is making a 0x3 instead of a 0x6.

There is no way to know from a `SensorItem` or a `RegisterSet` item if the underlying item is from a write type switch component.

The following is just a guess at how to fix this issue:

It seems like SensorItem might need to be extended to contain information of whether it is a write type "sensor" and then in `ModbusController::update_range_` this new member variable could be used to either send a `create_read_command` or to call the correct `create_write_{single/multiple}_{coil[s]/command}` command.  The `ModbusController::create_register_ranges_` function probably also needs to aggregate the registers depending on this new write type member variable.

It is possible that the `modbus_controller` `output` has the same problem

gotnone, 2025-05-22:
@autox86 I have an open PR that uses the alternative fix outlined above.  You can test it by grabbing the PR https://github.com/esphome/esphome/pull/8880 as an external component. You can now disable read polling / updates for the switch with the additional option `assumed_state: true`

```
external_components:
  - source: github://pr#8880
    components: [modbus_controller]

...

switch:
- platform: modbus_controller
    modbus_controller_id: modbus_frequency_inverter
    address: 0x2000
    register_type: holding
    name: "FU An"
    assumed_state: true
    write_lambda: |-
      ESP_LOGD("main","Modbus Switch incoming state = %f",x);
      // return false ; // use this to just change the value
      payload.push_back(0x03); // device address
      payload.push_back(0x06); // force single register
      payload.push_back(0x20); // high byte address of the coil
      payload.push_back(0x00); // low byte address of the coil
      payload.push_back(0x00); // command      
      payload.push_back(0x01); // command
      return true;
```

synthesis77, 2025-05-22:
@gotnone  nice, looks clean.  just switch though?  could be useful for number/etc too.  and for my slightly tangental issue, I think I'd need some sort of "read on-demand" coupling rather than just write-only (which might be a clearer name than assumed_state:).  do folks usually put comments here or in pull request?  I'm new to this project...  thanks!