esphome/issues #6621: Light 'turn on'-transition is delayed by transition length

CasperVerswijvelt, 2024-12-30:
### The problem

I have a simple ESPHome configuration with a SK6812 strip controlled with neopixelbus, which can be toggled on and off. I am using `default_transition_length` to ease the transition, which works beatifully when turning off the lamp. When turning it on, however, the light actually turning on is delayed by what seems the value passed to `default_transition_length`, 2 seconds in my example, after which it then still does the transition, but much quicker.

**Config:**
```
esphome:
  name: tile-lamp
  friendly_name: Tile lamp

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "redacted"

ota:
  - platform: esphome
    password: "redacted"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  ap:
    ssid: "Tile-Lamp Fallback Hotspot"
    password: "redacted"

captive_portal:

light:
  - platform: neopixelbus
    type: GRBW
    variant: SK6812
    pin: GPIO27
    num_leds: 16
    name: "light"
    id: "lamp"
    default_transition_length: 2s
    restore_mode: RESTORE_DEFAULT_OFF
```
**Video showing the issue:**
https://github.com/user-attachments/assets/55396a14-8723-49c8-af2c-d78e3bb1f331

Note that this shows the issue when controlled using home assistant, but I see the exact same behaviour when the light is controlled using an physical switch connected to the ESP32 using following code:

```
#esp32_touch:
#  setup_mode: true

binary_sensor:
  - platform: esp32_touch
    name: "Touch"
    pin: GPIO32
    threshold: 1150
    on_press:
      - light.toggle: lamp
```

### Which version of ESPHome has the issue?

 2024.12.2

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

 2024.12.2

### What platform are you using?

ESP32

### Board

AZDelivery ESP32 NodeMCU WiFi Dev Kit C CP2102

### Component causing the issue

light

### YAML Config

```yaml
esphome:
  name: tile-lamp
  friendly_name: Tile lamp

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "redacted"

ota:
  - platform: esphome
    password: "redacted"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  ap:
    ssid: "Tile-Lamp Fallback Hotspot"
    password: "redacted"

captive_portal:

light:
  - platform: neopixelbus
    type: GRBW
    variant: SK6812
    pin: GPIO27
    num_leds: 16
    name: "light"
    id: "lamp"
    default_transition_length: 2s
    restore_mode: RESTORE_DEFAULT_OFF
```

### Anything in the logs that might be useful for us?

```txt
Wi-Fi logging output as seen in video

[14:36:48][D][light:036]: 'light' Setting:
[14:36:48][D][light:047]:   State: ON
[14:36:48][D][light:085]:   Transition length: 2.0s
[14:36:51][D][light:036]: 'light' Setting:
[14:36:51][D][light:047]:   State: OFF
[14:36:51][D][light:085]:   Transition length: 2.0s
[14:36:55][D][light:036]: 'light' Setting:
[14:36:55][D][light:047]:   State: ON
[14:36:55][D][light:085]:   Transition length: 2.0s
[14:36:58][D][light:036]: 'light' Setting:
[14:36:58][D][light:047]:   State: OFF
[14:36:58][D][light:085]:   Transition length: 2.0s
```
```

mathisloge, 2024-12-31:
Seems related to my issue: https://github.com/esphome/issues/issues/6612

It's exactly the same behavior which I've observed

LostInCompilation, 2025-11-27:
The issue is still present in latest ESPHome (2025.11.1).