esphome/issues #6625: esp_adf.microphone set Warning flag: unspecified

lifuhuang, 2024-12-31:
### The problem

microphone does not work on box-3. Repeatedly getting the following message non-stop: 

[14:25:03][W][component:157]: Component esp_adf.microphone set Warning flag: unspecified
[14:25:03][W][component:170]: Component esp_adf.microphone cleared Warning flag

### Which version of ESPHome has the issue?

2024.12.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.12

### What platform are you using?

ESP32-IDF

### Board

S3-Box-3

### Component causing the issue

esp_adf.microphone 

### YAML Config

```yaml
substitutions:
  name: esp32-s3-box-3-2e078c
  friendly_name: ESP32 S3 Box 3 2e078c
  micro_wake_word_model: hey_jarvis
packages:
  esphome.voice-assistant: github://esphome/wake-word-voice-assistants/esp32-s3-box-3/esp32-s3-box-3.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:
  encryption:
    key: 3+ceTC4BqmusQRvmPeHhu43we6YZlU5Zf+JC1z9yz5U=
esp32:
  framework:
    type: esp-idf
    version: 4.4.5
    platform_version: 5.4.0

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
```

### Anything in the logs that might be useful for us?

```txt
[14:25:03][W][component:157]: Component esp_adf.microphone set Warning flag: unspecified
[14:25:03][W][component:170]: Component esp_adf.microphone cleared Warning flag
[14:25:03][W][component:157]: Component esp_adf.microphone set Warning flag: unspecified
[14:25:03][W][component:170]: Component esp_adf.microphone cleared Warning flag
[14:25:03][W][component:157]: Component esp_adf.microphone set Warning flag: unspecified
[14:25:03][W][component:170]: Component esp_adf.microphone cleared Warning flag
[14:25:03][W][component:157]: Component esp_adf.microphone set Warning flag: unspecified
[14:25:03][W][component:170]: Component esp_adf.microphone cleared Warning flag
[14:25:03][W][component:157]: Component esp_adf.microphone set Warning flag: unspecified
```

jonny190, 2025-01-17:
Getting the same issue

smashedr, 2025-01-17:
Have you tried using ESPHome version `2024.11.3`?
If it works on 11.3, but not 12+ then this is probably related to my issue: #6582

Marvin-D-Martian, 2025-01-21:
Getting the same issue

Non stop flooding the log with:

[17:05:43][W][component:157]: Component esp_adf.microphone set Warning flag: unspecified
[17:05:43][W][component:170]: Component esp_adf.microphone cleared Warning flag
[17:05:43][W][component:157]: Component esp_adf.microphone set Warning flag: unspecified
[17:05:43][W][component:170]: Component esp_adf.microphone cleared Warning flag
...

fabio-s-franco, 2025-01-23:
Also having it. Applying the workaround of #6852 doesn't get rid of the messages, but the microphone does work

closed 2026-05-23 by github-actions[bot]