esphome/issues #6626: Est-Poe-32 and One_Wire (DS18B20 became NA)

titou4307, 2025-01-02:
### The problem

Hello

I use Est-Poe-32 (with power by POE) and so the ethernet ....

I am trying to connect many DS18B20 Dallas but impossible to have the Dallas DS18B20 online under 2 bus. It may be function but after a time (many hour, or a little day or one or many days the DS18B20 became NA and into the frontend and so into Home Assistant I have no value of temperature

I am not understanding where is the error/ the problem

The PCB is this : 

https://shop.everythingsmart.io/products/est-poe-32-esp32-poe-development-board?srsltid=AfmBOorDnQ8CrK2taEHMfAglvfjhtK9l9N0qKI5YyhH7vDp29_ESWgHm

A solution I have found is to cut the power of the PCB (powered by POE) and reput power for to have the values of the temperature....but it function only for a time ....

Here a log when all function (when it not function, it appears into a line a message under Pin GPIOx "not devices found"

```
[09:11:58][I][app:100]: ESPHome version 2024.12.2 compiled on Dec 31 2024, 11:57:46
[09:11:58][C][logger:185]: Logger:
[09:11:58][C][logger:186]:   Level: DEBUG
[09:11:58][C][logger:188]:   Log Baud Rate: 115200
[09:11:58][C][logger:189]:   Hardware UART: UART0
[09:11:58][C][gpio.one_wire:020]: GPIO 1-wire bus:
[09:11:58][C][gpio.one_wire:021]:   Pin: GPIO13
[09:11:58][C][gpio.one_wire:080]:   Found devices:
[09:11:58][C][gpio.one_wire:082]:     0xd03cfc0457a9b028 (DS18B20)
[09:11:58][C][gpio.one_wire:082]:     0x243cbd0457a54b28 (DS18B20)
[09:11:58][C][gpio.one_wire:082]:     0xb03c940457a26b28 (DS18B20)
[09:11:58][C][gpio.one_wire:020]: GPIO 1-wire bus:
[09:11:58][C][gpio.one_wire:021]:   Pin: GPIO14
[09:11:58][C][gpio.one_wire:080]:   Found devices:
[09:11:58][C][gpio.one_wire:082]:     0xce3c6404577d3028 (DS18B20)
[09:11:58][C][gpio.one_wire:082]:     0xe03ca9045790ba28 (DS18B20)
[09:11:58][C][gpio.one_wire:082]:     0x523cba045700c528 (DS18B20)
[09:11:58][C][dallas.temp.sensor:029]: Dallas Temperature Sensor:
[09:11:58][C][dallas.temp.sensor:034]:   Address: 0xe03ca9045790ba28 (DS18B20)
[09:11:58][C][dallas.temp.sensor:035]:   Resolution: 12 bits
[09:11:58][C][dallas.temp.sensor:036]:   Update Interval: 120.0s
[09:11:58][C][dallas.temp.sensor:029]: Dallas Temperature Sensor:
[09:11:58][C][dallas.temp.sensor:034]:   Address: 0xce3c6404577d3028 (DS18B20)
[09:11:58][C][dallas.temp.sensor:035]:   Resolution: 12 bits
[09:11:58][C][dallas.temp.sensor:036]:   Update Interval: 120.0s
[09:11:58][C][dallas.temp.sensor:029]: Dallas Temperature Sensor:
[09:11:58][C][dallas.temp.sensor:034]:   Address: 0x523cba045700c528 (DS18B20)
[09:11:58][C][dallas.temp.sensor:035]:   Resolution: 12 bits
[09:11:58][C][dallas.temp.sensor:036]:   Update Interval: 120.0s
[09:11:58][C][dallas.temp.sensor:029]: Dallas Temperature Sensor:
[09:11:58][C][dallas.temp.sensor:034]:   Address: 0x243cbd0457a54b28 (DS18B20)
[09:11:58][C][dallas.temp.sensor:035]:   Resolution: 12 bits
[09:11:58][C][dallas.temp.sensor:036]:   Update Interval: 120.0s
[09:11:58][C][dallas.temp.sensor:029]: Dallas Temperature Sensor:
[09:11:58][C][dallas.temp.sensor:034]:   Address: 0xb03c940457a26b28 (DS18B20)
[09:11:58][C][dallas.temp.sensor:035]:   Resolution: 12 bits
[09:11:58][C][dallas.temp.sensor:036]:   Update Interval: 120.0s
[09:11:58][C][dallas.temp.sensor:029]: Dallas Temperature Sensor:
[09:11:58][C][dallas.temp.sensor:034]:   Address: 0xd03cfc0457a9b028 (DS18B20)
[09:11:58][C][dallas.temp.sensor:035]:   Resolution: 12 bits
[09:11:58][C][dallas.temp.sensor:036]:   Update Interval: 120.0s
[09:11:58][C][ethernet:326]: Ethernet:
[09:11:58][C][ethernet:515]:   IP Address: 192.168.1.112
[09:11:58][C][ethernet:516]:   Hostname: 'surveillance-temperatures'
[09:11:58][C][ethernet:517]:   Subnet: 255.255.255.0
[09:11:58][C][ethernet:518]:   Gateway: 192.168.1.1
[09:11:58][C][ethernet:523]:   DNS1: 0.0.0.0
[09:11:58][C][ethernet:524]:   DNS2: 0.0.0.0
[09:11:58][C][ethernet:536]:   MAC Address: 64:B7:08:70:53:14
[09:11:58][C][ethernet:537]:   Is Full Duplex: YES
[09:11:58][C][ethernet:538]:   Link Speed: 100
[... 213 lines cut ...]
    update_interval: 120s
  - platform: dallas_temp
    one_wire_id: bus2_cellier
    address: 0x523cba045700c528
    name: "temperature 3 Congélateur"
    update_interval: 120s
  - platform: dallas_temp
    one_wire_id: bus1_garage
    address: 0x243cbd0457a54b28
    name: "temperature 5 Radiateurs"
    update_interval: 120s
  - platform: dallas_temp
    one_wire_id: bus1_garage
    address: 0xb03c940457a26b28
    name: "temperature 6 Plancher départ"
    update_interval: 120s
  - platform: dallas_temp
    one_wire_id: bus1_garage
    address: 0xd03cfc0457a9b028
    name: "temperature 7 Plancher retour"
    update_interval: 120s

 web_server:
   port: 80
```

ssieb, 2025-01-02:
When it stops working, do both lines stop working?
What is in the logs when that happens?

titou4307, 2025-01-02:
Hello @ssieb 
First : happy new year 2025 with healthy 

At this time I have no error and the log are for example : 

```
[09:11:58][C][gpio.one_wire:021]:   Pin: GPIO13
[09:11:58][C][gpio.one_wire:080]:   Found devices
```
 When it stop working the log here (for example) : 

```
[09:11:58][C][gpio.one_wire:021]:   Pin: GPIO13
[09:11:58][C][gpio.one_wire:080]:   No devices found
```

And it is not only on GPIO13 but it could be on GPIO14 (not all the time on same GPIO)

and I'am not sure 100% but I have seen a case with 5 / 6 DS18B20 with NA Value

titou4307, 2025-01-02:
Hello
New situation .... Here the logs : 

The 3 DS18B20 on the two buses appears but error cause NA Value on frontend web esphome

```
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Starting log output from 192.168.1.112 using esphome API
INFO Successfully connected to surveillance-temperatures @ 192.168.1.112 in 0.001s
INFO Successful handshake with surveillance-temperatures @ 192.168.1.112 in 0.096s
[21:42:26][I][app:100]: ESPHome version 2024.12.2 compiled on Dec 31 2024, 11:57:46
[21:42:26][C][logger:185]: Logger:
[21:42:26][C][logger:186]:   Level: DEBUG
[21:42:26][C][logger:188]:   Log Baud Rate: 115200
[21:42:26][C][logger:189]:   Hardware UART: UART0
[21:42:26][C][gpio.one_wire:020]: GPIO 1-wire bus:
[21:42:26][C][gpio.one_wire:021]:   Pin: GPIO13
[21:42:26][C][gpio.one_wire:080]:   Found devices:
[21:42:26][C][gpio.one_wire:082]:     0xd03cfc0457a9b028 (DS18B20)
[21:42:26][C][gpio.one_wire:082]:     0x243cbd0457a54b28 (DS18B20)
[21:42:26][C][gpio.one_wire:082]:     0xb03c940457a26b28 (DS18B20)
[21:42:26][C][gpio.one_wire:020]: GPIO 1-wire bus:
[21:42:26][C][gpio.one_wire:021]:   Pin: GPIO14
[21:42:26][C][gpio.one_wire:080]:   Found devices:
[21:42:26][C][gpio.one_wire:082]:     0xce3c6404577d3028 (DS18B20)
[21:42:26][C][gpio.one_wire:082]:     0xe03ca9045790ba28 (DS18B20)
[21:42:26][C][gpio.one_wire:082]:     0x523cba045700c528 (DS18B20)
[21:42:26][C][dallas.temp.sensor:029]: Dallas Temperature Sensor:
[21:42:26][C][dallas.temp.sensor:034]:   Address: 0xe03ca9045790ba28 (DS18B20)
[21:42:26][C][dallas.temp.sensor:035]:   Resolution: 12 bits
[21:42:26][C][dallas.temp.sensor:036]:   Update Interval: 120.0s
[21:42:26][C][dallas.temp.sensor:029]: Dallas Temperature Sensor:
[21:42:26][C][dallas.temp.sensor:034]:   Address: 0xce3c6404577d3028 (DS18B20)
[21:42:26][C][dallas.temp.sensor:035]:   Resolution: 12 bits
[21:42:26][C][dallas.temp.sensor:036]:   Update Interval: 120.0s
[21:42:26][C][dallas.temp.sensor:029]: Dallas Temperature Sensor:
[21:42:26][C][dallas.temp.sensor:034]:   Address: 0x523cba045700c528 (DS18B20)
[21:42:26][C][dallas.temp.sensor:035]:   Resolution: 12 bits
[21:42:26][C][dallas.temp.sensor:036]:   Update Interval: 120.0s
[21:42:26][C][dallas.temp.sensor:029]: Dallas Temperature Sensor:
[21:42:26][C][dallas.temp.sensor:034]:   Address: 0x243cbd0457a54b28 (DS18B20)
[21:42:26][C][dallas.temp.sensor:035]:   Resolution: 12 bits
[21:42:26][C][dallas.temp.sensor:036]:   Update Interval: 120.0s
[21:42:26][C][dallas.temp.sensor:029]: Dallas Temperature Sensor:
[21:42:26][C][dallas.temp.sensor:034]:   Address: 0xb03c940457a26b28 (DS18B20)
[21:42:26][C][dallas.temp.sensor:035]:   Resolution: 12 bits
[21:42:26][C][dallas.temp.sensor:036]:   Update Interval: 120.0s
[21:42:26][C][dallas.temp.sensor:029]: Dallas Temperature Sensor:
[21:42:26][C][dallas.temp.sensor:034]:   Address: 0xd03cfc0457a9b028 (DS18B20)
[21:42:26][C][dallas.temp.sensor:035]:   Resolution: 12 bits
[21:42:26][C][dallas.temp.sensor:036]:   Update Interval: 120.0s
[21:42:26][C][ethernet:326]: Ethernet:
[21:42:26][C][ethernet:515]:   IP Address: 192.168.1.112
[21:42:26][C][ethernet:516]:   Hostname: 'surveillance-temperatures'
[21:42:26][C][ethernet:517]:   Subnet: 255.255.255.0
[21:42:26][C][ethernet:518]:   Gateway: 192.168.1.1
[21:42:26][C][ethernet:523]:   DNS1: 0.0.0.0
[21:42:26][C][ethernet:524]:   DNS2: 0.0.0.0
[21:42:26][C][ethernet:536]:   MAC Address: 64:B7:08:70:53:14
[21:42:26][C][ethernet:537]:   Is Full Duplex: YES
[21:42:26][C][ethernet:538]:   Link Speed: 100
[21:42:26][C][ethernet:345]:   Power Pin: 12
[21:42:26][C][ethernet:347]:   MDC Pin: 23
[21:42:26][C][ethernet:348]:   MDIO Pin: 18
[21:42:26][C][ethernet:349]:   PHY addr: 0
[21:42:26][C][ethernet:351]:   Type: LAN8720
[21:42:26][C][web_server:153]: Web Server:
[21:42:26][C][web_server:154]:   Address: 192.168.1.112:80
[21:42:26][C][mdns:116]: mDNS:
[... 2 lines cut ...]
[21:42:26][C][esphome.ota:074]:   Address: 192.168.1.112:3232
[21:42:26][C][esphome.ota:075]:   Version: 2
[21:42:26][C][esphome.ota:078]:   Password configured
[21:42:26][C][safe_mode:018]: Safe Mode:
[21:42:26][C][safe_mode:019]:   Boot considered successful after 60 seconds
[21:42:26][C][safe_mode:021]:   Invoke after 10 boot attempts
[21:42:26][C][safe_mode:022]:   Remain in safe mode for 300 seconds
[21:42:26][C][api:140]: API Server:
[21:42:26][C][api:141]:   Address: 192.168.1.112:6053
[21:42:26][C][api:143]:   Using noise encryption: YES
[21:42:50][W][component:170]: Component dallas_temp.sensor cleared Warning flag
[21:42:51][W][dallas.temp.sensor:139]: 'temperature 7 Plancher retour' - Scratch pad checksum invalid!
[21:42:51][W][component:157]: Component dallas_temp.sensor set Warning flag: scratch pad checksum invalid
[21:42:51][D][sensor:093]: 'temperature 7 Plancher retour': Sending state nan °C with 1 decimals of accuracy
[21:42:58][D][dallas.temp.sensor:054]: 'temperature 3 Congélateur': Got Temperature=-21.3°C
[21:42:58][D][sensor:093]: 'temperature 3 Congélateur': Sending state -21.31250 °C with 1 decimals of accuracy
[21:43:00][W][component:170]: Component dallas_temp.sensor cleared Warning flag
[21:43:01][W][dallas.temp.sensor:139]: 'temperature 6 Plancher départ' - Scratch pad checksum invalid!
[21:43:01][W][component:157]: Component dallas_temp.sensor set Warning flag: scratch pad checksum invalid
[21:43:01][D][sensor:093]: 'temperature 6 Plancher départ': Sending state nan °C with 1 decimals of accuracy
[21:43:05][D][dallas.temp.sensor:054]: 'temperature 2 Réfrigérateur': Got Temperature=4.7°C
[21:43:05][D][sensor:093]: 'temperature 2 Réfrigérateur': Sending state 4.68750 °C with 1 decimals of accuracy
[21:43:20][D][dallas.temp.sensor:054]: 'temperature 1 Cellier': Got Temperature=15.6°C
[21:43:20][D][sensor:093]: 'temperature 1 Cellier': Sending state 15.62500 °C with 1 decimals of accuracy
```

titou4307, 2025-01-02:
And more later : 

```
[22:46:56][I][app:100]: ESPHome version 2024.12.2 compiled on Dec 31 2024, 11:57:46
[22:46:56][C][logger:185]: Logger:
[22:46:56][C][logger:186]:   Level: DEBUG
[22:46:56][C][logger:188]:   Log Baud Rate: 115200
[22:46:56][C][logger:189]:   Hardware UART: UART0
[22:46:56][C][gpio.one_wire:020]: GPIO 1-wire bus:
[22:46:56][C][gpio.one_wire:021]:   Pin: GPIO13
[22:46:56][C][gpio.one_wire:080]:   Found devices:
[22:46:56][C][gpio.one_wire:082]:     0xd03cfc0457a9b028 (DS18B20)
[22:46:56][C][gpio.one_wire:082]:     0x243cbd0457a54b28 (DS18B20)
[22:46:56][C][gpio.one_wire:082]:     0xb03c940457a26b28 (DS18B20)
[22:46:56][C][gpio.one_wire:020]: GPIO 1-wire bus:
[22:46:56][C][gpio.one_wire:021]:   Pin: GPIO14
[22:46:56][W][gpio.one_wire:078]:   Found no devices!
[22:46:56][C][dallas.temp.sensor:029]: Dallas Temperature Sensor:
[22:46:56][C][dallas.temp.sensor:034]:   Address: 0xe03ca9045790ba28 (DS18B20)
[22:46:56][C][dallas.temp.sensor:035]:   Resolution: 12 bits
[22:46:56][C][dallas.temp.sensor:036]:   Update Interval: 120.0s
[22:46:56][C][dallas.temp.sensor:029]: Dallas Temperature Sensor:
[22:46:56][C][dallas.temp.sensor:034]:   Address: 0xce3c6404577d3028 (DS18B20)
[22:46:56][C][dallas.temp.sensor:035]:   Resolution: 12 bits
[22:46:56][C][dallas.temp.sensor:036]:   Update Interval: 120.0s
[22:46:56][C][dallas.temp.sensor:029]: Dallas Temperature Sensor:
[22:46:56][C][dallas.temp.sensor:034]:   Address: 0x523cba045700c528 (DS18B20)
[22:46:56][C][dallas.temp.sensor:035]:   Resolution: 12 bits
[22:46:56][C][dallas.temp.sensor:036]:   Update Interval: 120.0s
[22:46:56][C][dallas.temp.sensor:029]: Dallas Temperature Sensor:
[22:46:56][C][dallas.temp.sensor:034]:   Address: 0x243cbd0457a54b28 (DS18B20)
[22:46:56][C][dallas.temp.sensor:035]:   Resolution: 12 bits
[22:46:56][C][dallas.temp.sensor:036]:   Update Interval: 120.0s
[22:46:56][C][dallas.temp.sensor:029]: Dallas Temperature Sensor:
[22:46:56][C][dallas.temp.sensor:034]:   Address: 0xb03c940457a26b28 (DS18B20)
[22:46:56][C][dallas.temp.sensor:035]:   Resolution: 12 bits
[22:46:56][C][dallas.temp.sensor:036]:   Update Interval: 120.0s
[22:46:56][C][dallas.temp.sensor:029]: Dallas Temperature Sensor:
[22:46:56][C][dallas.temp.sensor:034]:   Address: 0xd03cfc0457a9b028 (DS18B20)
[22:46:56][C][dallas.temp.sensor:035]:   Resolution: 12 bits
[22:46:56][C][dallas.temp.sensor:036]:   Update Interval: 120.0s
```
Dallas DS18B20 disappears on GPIO 14 !!

ssieb, 2025-01-02:
How long are the wires and how are the sensors arranged?

titou4307, 2025-01-02:
> How long are the wires and how are the sensors arranged?

Hello
The Est Poe 32 is at the top of the garage wall....
3 x DS18B20 are going to the boiler (on the garage near the Est Poe 32) so the cable length is approximately of 350cm....(max 400cm)
2 x DS18B20 are going into other piece to fridge and freezer with same length 350cm max 400cm
1 x DS18B20 going into the same piece fridge/freezer for take temp of the piece (length 200cm)

You think about what ?

titou4307, 2025-01-03:
Hello @ssieb 
I have not change the length of the 6 DS18B20

Before the Est Poe 32 the 6 DS18B20 was connected to the same model at a Sonoff THR320 and all was ok

I have a DS18B20 who take the same road like an electric wire (220 volts in France) mays be interference are explanation
I will change the "road" and made return here.....

titou4307, 2025-01-05:
Hello

I have properly install one DS18B20 who is very near an electric wire.....
I have verify the ethernet cable and the cable is OK with the 4 pairs and the GND

I have restart the Est-Poe-32 at 17h30 and all the DS18B20 appears into the logs, in the frontend Esphome and into Home Assistant.....

At 20h00 the DS18B20 have NAN Value : 

```
INFO ESPHome 2024.12.2
INFO Reading configuration /config/esphome/surveillance-temperatures.yaml...
WARNING GPIO12 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Starting log output from 192.168.1.112 using esphome API
INFO Successfully connected to surveillance-temperatures @ 192.168.1.112 in 0.001s
INFO Successful handshake with surveillance-temperatures @ 192.168.1.112 in 0.086s
[20:04:04][I][app:100]: ESPHome version 2024.12.2 compiled on Jan  5 2025, 18:22:15
[20:04:04][C][logger:185]: Logger:
[20:04:04][C][logger:186]:   Level: DEBUG
[20:04:04][C][logger:188]:   Log Baud Rate: 115200
[20:04:04][C][logger:189]:   Hardware UART: UART0
[20:04:04][C][gpio.one_wire:020]: GPIO 1-wire bus:
[20:04:04][C][gpio.one_wire:021]:   Pin: GPIO13
[20:04:04][C][gpio.one_wire:080]:   Found devices:
[20:04:04][C][gpio.one_wire:082]:     0xd03cfc0457a9b028 (DS18B20)
[20:04:04][C][gpio.one_wire:082]:     0x243cbd0457a54b28 (DS18B20)
[20:04:04][C][gpio.one_wire:082]:     0xb03c940457a26b28 (DS18B20)
[20:04:04][C][gpio.one_wire:020]: GPIO 1-wire bus:
[20:04:04][C][gpio.one_wire:021]:   Pin: GPIO14
[20:04:04][C][gpio.one_wire:080]:   Found devices:
[20:04:04][C][gpio.one_wire:082]:     0xce3c6404577d3028 (DS18B20)
[20:04:04][C][gpio.one_wire:082]:     0xe03ca9045790ba28 (DS18B20)
[20:04:04][C][gpio.one_wire:082]:     0x523cba045700c528 (DS18B20)
[20:04:04][C][dallas.temp.sensor:029]: Dallas Temperature Sensor:
[20:04:04][C][dallas.temp.sensor:034]:   Address: 0xe03ca9045790ba28 (DS18B20)
[20:04:04][C][dallas.temp.sensor:035]:   Resolution: 12 bits
[20:04:04][C][dallas.temp.sensor:036]:   Update Interval: 120.0s
[20:04:04][C][dallas.temp.sensor:029]: Dallas Temperature Sensor:
[20:04:04][C][dallas.temp.sensor:034]:   Address: 0xce3c6404577d3028 (DS18B20)
[20:04:04][C][dallas.temp.sensor:035]:   Resolution: 12 bits
[20:04:04][C][dallas.temp.sensor:036]:   Update Interval: 120.0s
[20:04:04][C][dallas.temp.sensor:029]: Dallas Temperature Sensor:
[20:04:04][C][dallas.temp.sensor:034]:   Address: 0x523cba045700c528 (DS18B20)
[20:04:04][C][dallas.temp.sensor:035]:   Resolution: 12 bits
[20:04:04][C][dallas.temp.sensor:036]:   Update Interval: 120.0s
[20:04:04][C][dallas.temp.sensor:029]: Dallas Temperature Sensor:
[20:04:04][C][dallas.temp.sensor:034]:   Address: 0x243cbd0457a54b28 (DS18B20)
[20:04:04][C][dallas.temp.sensor:035]:   Resolution: 12 bits
[20:04:04][C][dallas.temp.sensor:036]:   Update Interval: 120.0s
[20:04:04][C][dallas.temp.sensor:029]: Dallas Temperature Sensor:
[20:04:04][C][dallas.temp.sensor:034]:   Address: 0xd03cfc0457a9b028 (DS18B20)
[20:04:04][C][dallas.temp.sensor:035]:   Resolution: 12 bits
[20:04:04][C][dallas.temp.sensor:036]:   Update Interval: 120.0s
[20:04:04][C][dallas.temp.sensor:029]: Dallas Temperature Sensor:
[20:04:04][C][dallas.temp.sensor:034]:   Address: 0xb03c940457a26b28 (DS18B20)
[20:04:04][C][dallas.temp.sensor:035]:   Resolution: 12 bits
[20:04:04][C][dallas.temp.sensor:036]:   Update Interval: 120.0s
[20:04:04][C][ethernet:326]: Ethernet:
[20:04:04][C][ethernet:515]:   IP Address: 192.168.1.112
[20:04:04][C][ethernet:516]:   Hostname: 'surveillance-temperatures'
[20:04:04][C][ethernet:517]:   Subnet: 255.255.255.0
[20:04:04][C][ethernet:518]:   Gateway: 192.168.1.1
[20:04:04][C][ethernet:523]:   DNS1: 0.0.0.0
[20:04:04][C][ethernet:524]:   DNS2: 0.0.0.0
[20:04:04][C][ethernet:536]:   MAC Address: 64:B7:08:70:53:14
[20:04:04][C][ethernet:537]:   Is Full Duplex: YES
[20:04:04][C][ethernet:538]:   Link Speed: 100
[20:04:04][C][ethernet:345]:   Power Pin: 12
[... 167 lines cut ...]
[20:12:38][D][dallas.temp.sensor:054]: 'temperature 2 Réfrigérateur': Got Temperature=3.4°C
[20:12:38][D][sensor:093]: 'temperature 2 Réfrigérateur': Sending state 3.37500 °C with 1 decimals of accuracy
[20:12:43][D][dallas.temp.sensor:054]: 'temperature 5 Radiateurs': Got Temperature=52.8°C
[20:12:43][D][sensor:093]: 'temperature 5 Radiateurs': Sending state 52.81250 °C with 1 decimals of accuracy
[20:12:50][D][dallas.temp.sensor:054]: 'temperature 3 Congélateur': Got Temperature=-21.0°C
[20:12:50][D][sensor:093]: 'temperature 3 Congélateur': Sending state -21.00000 °C with 1 decimals of accuracy
[20:12:50][D][dallas.temp.sensor:054]: 'temperature 7 Plancher retour': Got Temperature=23.1°C
[20:12:50][D][sensor:093]: 'temperature 7 Plancher retour': Sending state 23.06250 °C with 1 decimals of accuracy
[20:13:07][D][dallas.temp.sensor:054]: 'temperature 6 Plancher départ': Got Temperature=39.5°C
[20:13:07][D][sensor:093]: 'temperature 6 Plancher départ': Sending state 39.50000 °C with 1 decimals of accuracy
[20:14:38][D][dallas.temp.sensor:054]: 'temperature 1 Cellier': Got Temperature=16.0°C
[20:14:38][D][sensor:093]: 'temperature 1 Cellier': Sending state 16.00000 °C with 1 decimals of accuracy
[20:14:38][D][dallas.temp.sensor:054]: 'temperature 2 Réfrigérateur': Got Temperature=3.4°C
[... cut ...]
[20:14:50][D][sensor:093]: 'temperature 3 Congélateur': Sending state -21.00000 °C with 1 decimals of accuracy
[20:14:50][D][dallas.temp.sensor:054]: 'temperature 7 Plancher retour': Got Temperature=23.1°C
[20:14:50][D][sensor:093]: 'temperature 7 Plancher retour': Sending state 23.06250 °C with 1 decimals of accuracy
[20:15:07][D][dallas.temp.sensor:054]: 'temperature 6 Plancher départ': Got Temperature=40.2°C
[20:15:07][D][sensor:093]: 'temperature 6 Plancher départ': Sending state 40.18750 °C with 1 decimals of accuracy
```

I have no idea of the solution

ssieb, 2025-01-05:
Can you reset the power to the sensor and not the ESP to see if that fixes it?
Also, try adding this to your config:
```yaml
external_components:
  - source:
      type: git
      url: https://github.com/ssieb/esphome
      ref: onewire
    components: [ gpio, one_wire, dallas_temp ]
    refresh: 1min
```

titou4307, 2025-01-05:
Hi

After cutting power of the 6 DS18B20 and waiting a time (more than 3 minutes....) I have this logs : 

```
INFO ESPHome 2024.12.2
INFO Reading configuration /config/esphome/surveillance-temperatures.yaml...
WARNING GPIO12 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Starting log output from 192.168.1.112 using esphome API
INFO Successfully connected to surveillance-temperatures @ 192.168.1.112 in 0.002s
INFO Successful handshake with surveillance-temperatures @ 192.168.1.112 in 0.074s
[00:10:23][I][app:100]: ESPHome version 2024.12.2 compiled on Jan  5 2025, 18:22:15
[00:10:23][C][logger:185]: Logger:
[00:10:23][C][logger:186]:   Level: DEBUG
[00:10:23][C][logger:188]:   Log Baud Rate: 115200
[00:10:23][C][logger:189]:   Hardware UART: UART0
[00:10:23][C][gpio.one_wire:020]: GPIO 1-wire bus:
[00:10:23][C][gpio.one_wire:021]:   Pin: GPIO13
[00:10:23][C][gpio.one_wire:080]:   Found devices:
[00:10:23][C][gpio.one_wire:082]:     0xd03cfc0457a9b028 (DS18B20)
[00:10:23][C][gpio.one_wire:082]:     0x243cbd0457a54b28 (DS18B20)
[00:10:23][C][gpio.one_wire:082]:     0xb03c940457a26b28 (DS18B20)
[00:10:23][C][gpio.one_wire:020]: GPIO 1-wire bus:
[00:10:23][C][gpio.one_wire:021]:   Pin: GPIO14
[00:10:23][C][gpio.one_wire:080]:   Found devices:
[00:10:23][C][gpio.one_wire:082]:     0xce3c6404577d3028 (DS18B20)
[00:10:23][C][gpio.one_wire:082]:     0xe03ca9045790ba28 (DS18B20)
[00:10:23][C][gpio.one_wire:082]:     0x523cba045700c528 (DS18B20)
[00:10:23][W][dallas.temp.sensor:139]: 'temperature 5 Radiateurs' - Scratch pad checksum invalid!
[00:10:23][W][component:157]: Component dallas_temp.sensor set Warning flag: scratch pad checksum invalid
[00:10:23][D][sensor:093]: 'temperature 5 Radiateurs': Sending state nan °C with 1 decimals of accuracy
[00:10:23][C][dallas.temp.sensor:029]: Dallas Temperature Sensor:
[00:10:23][C][dallas.temp.sensor:034]:   Address: 0xe03ca9045790ba28 (DS18B20)
[00:10:23][C][dallas.temp.sensor:035]:   Resolution: 12 bits
[00:10:23][C][dallas.temp.sensor:036]:   Update Interval: 120.0s
[00:10:23][C][dallas.temp.sensor:029]: Dallas Temperature Sensor:
[00:10:23][C][dallas.temp.sensor:034]:   Address: 0xce3c6404577d3028 (DS18B20)
[00:10:23][C][dallas.temp.sensor:035]:   Resolution: 12 bits
[00:10:23][C][dallas.temp.sensor:036]:   Update Interval: 120.0s
[00:10:23][C][dallas.temp.sensor:029]: Dallas Temperature Sensor:
[00:10:23][C][dallas.temp.sensor:034]:   Address: 0x523cba045700c528 (DS18B20)
[00:10:23][C][dallas.temp.sensor:035]:   Resolution: 12 bits
[00:10:23][C][dallas.temp.sensor:036]:   Update Interval: 120.0s
[00:10:23][C][dallas.temp.sensor:029]: Dallas Temperature Sensor:
[00:10:23][C][dallas.temp.sensor:034]:   Address: 0x243cbd0457a54b28 (DS18B20)
[00:10:23][C][dallas.temp.sensor:035]:   Resolution: 12 bits
[00:10:23][C][dallas.temp.sensor:036]:   Update Interval: 120.0s
[00:10:23][C][dallas.temp.sensor:029]: Dallas Temperature Sensor:
[00:10:23][C][dallas.temp.sensor:034]:   Address: 0xd03cfc0457a9b028 (DS18B20)
[00:10:23][C][dallas.temp.sensor:035]:   Resolution: 12 bits
[00:10:23][C][dallas.temp.sensor:036]:   Update Interval: 120.0s
[00:10:23][C][dallas.temp.sensor:029]: Dallas Temperature Sensor:
[00:10:23][C][dallas.temp.sensor:034]:   Address: 0xb03c940457a26b28 (DS18B20)
[00:10:23][C][dallas.temp.sensor:035]:   Resolution: 12 bits
[00:10:23][C][dallas.temp.sensor:036]:   Update Interval: 120.0s
[00:10:23][C][ethernet:326]: Ethernet:
[00:10:23][C][ethernet:515]:   IP Address: 192.168.1.112
[00:10:23][C][ethernet:516]:   Hostname: 'surveillance-temperatures'
[00:10:23][C][ethernet:517]:   Subnet: 255.255.255.0
[00:10:23][C][ethernet:518]:   Gateway: 192.168.1.1
[00:10:23][C][ethernet:523]:   DNS1: 0.0.0.0
[00:10:23][C][ethernet:524]:   DNS2: 0.0.0.0
[00:10:23][C][ethernet:536]:   MAC Address: 64:B7:08:70:53:14
[00:10:23][C][ethernet:537]:   Is Full Duplex: YES
[00:10:23][C][ethernet:538]:   Link Speed: 100
[00:10:23][C][ethernet:345]:   Power Pin: 12
[00:10:23][C][ethernet:347]:   MDC Pin: 23
[00:10:23][C][ethernet:348]:   MDIO Pin: 18
[... 43 lines cut ...]
[00:12:35][W][dallas.temp.sensor:139]: 'temperature 6 Plancher départ' - Scratch pad checksum invalid!
[00:12:35][W][component:157]: Component dallas_temp.sensor set Warning flag: scratch pad checksum invalid
[00:12:35][D][sensor:093]: 'temperature 6 Plancher départ': Sending state nan °C with 1 decimals of accuracy
[00:12:40][D][dallas.temp.sensor:054]: 'temperature 3 Congélateur': Got Temperature=-20.9°C
[00:12:40][D][sensor:093]: 'temperature 3 Congélateur': Sending state -20.93750 °C with 1 decimals of accuracy
[00:12:46][D][dallas.temp.sensor:054]: 'temperature 2 Réfrigérateur': Got Temperature=3.6°C
[00:12:46][D][sensor:093]: 'temperature 2 Réfrigérateur': Sending state 3.56250 °C with 1 decimals of accuracy

```

I have add the external components here like you say me : 

```
external_components:
  - source:
      type: git
      url: https://github.com/ssieb/esphome
      ref: onewire
    components: [ gpio, one_wire, dallas_temp ]
    refresh: 1min

```

I will give the new logs after a time
Thanks for your helps

titou4307, 2025-01-05:
Hi....
None DS18B20 Found : 

```
INFO ESPHome 2024.12.2
INFO Reading configuration /config/esphome/surveillance-temperatures.yaml...
INFO Updating https://github.com/ssieb/esphome@onewire
WARNING GPIO12 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Starting log output from 192.168.1.112 using esphome API
INFO Successfully connected to surveillance-temperatures @ 192.168.1.112 in 0.002s
INFO Successful handshake with surveillance-temperatures @ 192.168.1.112 in 0.067s
[00:22:47][I][app:100]: ESPHome version 2024.12.2 compiled on Jan  6 2025, 00:16:02
[00:22:47][C][logger:185]: Logger:
[00:22:47][C][logger:186]:   Level: DEBUG
[00:22:47][C][logger:188]:   Log Baud Rate: 115200
[00:22:47][C][logger:189]:   Hardware UART: UART0
[00:22:47][C][gpio.one_wire:020]: GPIO 1-wire bus:
[00:22:47][C][gpio.one_wire:021]:   Pin: GPIO13
[00:22:47][W][gpio.one_wire:078]:   Found no devices!
[00:22:47][C][gpio.one_wire:020]: GPIO 1-wire bus:
[00:22:47][C][gpio.one_wire:021]:   Pin: GPIO14
[00:22:47][W][gpio.one_wire:078]:   Found no devices!
[00:22:47][C][dallas.temp.sensor:029]: Dallas Temperature Sensor:
[00:22:47][C][dallas.temp.sensor:034]:   Address: 0xe03ca9045790ba28 (DS18B20)
[00:22:47][C][dallas.temp.sensor:035]:   Resolution: 12 bits
[00:22:47][C][dallas.temp.sensor:036]:   Update Interval: 120.0s
[00:22:47][C][dallas.temp.sensor:029]: Dallas Temperature Sensor:
[00:22:47][C][dallas.temp.sensor:034]:   Address: 0xce3c6404577d3028 (DS18B20)
[00:22:47][C][dallas.temp.sensor:035]:   Resolution: 12 bits
[00:22:47][C][dallas.temp.sensor:036]:   Update Interval: 120.0s
[00:22:47][C][dallas.temp.sensor:029]: Dallas Temperature Sensor:
[00:22:47][C][dallas.temp.sensor:034]:   Address: 0x523cba045700c528 (DS18B20)
[00:22:47][C][dallas.temp.sensor:035]:   Resolution: 12 bits
[00:22:47][C][dallas.temp.sensor:036]:   Update Interval: 120.0s
[00:22:47][C][dallas.temp.sensor:029]: Dallas Temperature Sensor:
[00:22:47][C][dallas.temp.sensor:034]:   Address: 0x243cbd0457a54b28 (DS18B20)
[00:22:47][C][dallas.temp.sensor:035]:   Resolution: 12 bits
[00:22:47][C][dallas.temp.sensor:036]:   Update Interval: 120.0s
[00:22:47][C][dallas.temp.sensor:029]: Dallas Temperature Sensor:
[00:22:47][C][dallas.temp.sensor:034]:   Address: 0xd03cfc0457a9b028 (DS18B20)
[00:22:47][C][dallas.temp.sensor:035]:   Resolution: 12 bits
[00:22:47][C][dallas.temp.sensor:036]:   Update Interval: 120.0s
[00:22:47][C][dallas.temp.sensor:029]: Dallas Temperature Sensor:
[00:22:47][C][dallas.temp.sensor:034]:   Address: 0xb03c940457a26b28 (DS18B20)
[00:22:47][C][dallas.temp.sensor:035]:   Resolution: 12 bits
[00:22:47][C][dallas.temp.sensor:036]:   Update Interval: 120.0s
[00:22:47][C][ethernet:326]: Ethernet:
[00:22:47][C][ethernet:515]:   IP Address: 192.168.1.112
[00:22:47][C][ethernet:516]:   Hostname: 'surveillance-temperatures'
[00:22:47][C][ethernet:517]:   Subnet: 255.255.255.0
[00:22:47][C][ethernet:518]:   Gateway: 192.168.1.1
[00:22:47][C][ethernet:523]:   DNS1: 0.0.0.0
[00:22:47][C][ethernet:524]:   DNS2: 0.0.0.0
[00:22:47][C][ethernet:536]:   MAC Address: 64:B7:08:70:53:14
[00:22:47][C][ethernet:537]:   Is Full Duplex: YES
[00:22:47][C][ethernet:538]:   Link Speed: 100
[00:22:47][C][ethernet:345]:   Power Pin: 12
[00:22:47][C][ethernet:347]:   MDC Pin: 23
[00:22:47][C][ethernet:348]:   MDIO Pin: 18
[00:22:47][C][ethernet:349]:   PHY addr: 0
[00:22:47][C][ethernet:351]:   Type: LAN8720
[00:22:47][C][web_server:153]: Web Server:
[00:22:47][C][web_server:154]:   Address: 192.168.1.112:80
[00:22:47][C][mdns:116]: mDNS:
[00:22:47][C][mdns:117]:   Hostname: surveillance-temperatures
[00:22:47][C][esphome.ota:073]: Over-The-Air updates:
[00:22:47][C][esphome.ota:074]:   Address: 192.168.1.112:3232
[00:22:47][C][esphome.ota:075]:   Version: 2
[... 12 lines cut ...]
[00:24:07][W][component:170]: Component dallas_temp.sensor cleared Warning flag
[00:24:08][W][dallas.temp.sensor:139]: 'temperature 6 Plancher départ' - Scratch pad checksum invalid!
[00:24:08][W][component:157]: Component dallas_temp.sensor set Warning flag: scratch pad checksum invalid
[00:24:08][D][sensor:093]: 'temperature 6 Plancher départ': Sending state nan °C with 1 decimals of accuracy
[00:24:10][W][component:170]: Component dallas_temp.sensor cleared Warning flag
[00:24:11][W][dallas.temp.sensor:139]: 'temperature 5 Radiateurs' - Scratch pad checksum invalid!
[00:24:11][W][component:157]: Component dallas_temp.sensor set Warning flag: scratch pad checksum invalid
[00:24:11][D][sensor:093]: 'temperature 5 Radiateurs': Sending state nan °C with 1 decimals of accuracy
[00:24:17][W][component:170]: Component dallas_temp.sensor cleared Warning flag
[00:24:18][W][dallas.temp.sensor:139]: 'temperature 1 Cellier' - Scratch pad checksum invalid!
[00:24:18][W][component:157]: Component dallas_temp.sensor set Warning flag: scratch pad checksum invalid
[00:24:18][D][sensor:093]: 'temperature 1 Cellier': Sending state nan °C with 1 decimals of accuracy
[00:24:38][W][component:170]: Component dallas_temp.sensor cleared Warning flag
[00:24:39][W][dallas.temp.sensor:139]: 'temperature 3 Congélateur' - Scratch pad checksum invalid!
[00:24:39][W][component:157]: Component dallas_temp.sensor set Warning flag: scratch pad checksum invalid
[... cut ...]
[00:24:41][W][component:170]: Component dallas_temp.sensor cleared Warning flag
[00:24:42][W][dallas.temp.sensor:139]: 'temperature 2 Réfrigérateur' - Scratch pad checksum invalid!
[00:24:42][W][component:157]: Component dallas_temp.sensor set Warning flag: scratch pad checksum invalid
[00:24:42][D][sensor:093]: 'temperature 2 Réfrigérateur': Sending state nan °C with 1 decimals of accuracy
```

After reboot (stop POE and put POE....) no changes, no DS18B20 found

So I erase the external component ?

ssieb, 2025-01-05:
Yes, but that suggests an issue with your wiring.  Do you have a logic analyzer?

titou4307, 2025-01-05:
After erase extrenel components, the 6x DS18B20 are detected....

Sorry, I not understand "a logic analyzer"... I found "oscilloscope" with translation google ? Sorry I am french....

If it is this material or similar, sorry I have not

titou4307, 2025-01-06:
Hello
The signal of the 2 x 3 x DS18B20 are connected in the GPIO13/GPIO14 via a resistor 4,7 kohm between the power (5v) and the signal

May be the 2 x 3 circuits are too long, many the two resistors need to be changed by 2 x resistors like 2.2 kohm ??

ssieb, 2025-01-06:
You could try that.

titou4307, 2025-01-07:
> You could try that.

I will.....

I am surprise but the install has functioned during more time.... (only this morning the problem happen one more time....)
Usually it function only many hours (.....)

But I have not change anything

titou4307, 2025-01-15:
Hello
Impossible to adjust anything....

I have change the Est-Poe-32 by another one flashed under Tasmota lite
I have the 6 DS18B20 at this time, I will see if it stable  .....

So if the value are stable so it may be the hardware (first Est-Po-32) or the software (Esphome bug)

titou4307, 2025-01-26:
Hi at all....

Since the two weeks my Est-Poe-32 under Tasmota with the 6 DS18B20 give all the températures all the times without bug...
I have never lost the value
At this time my wiring of the 6 DS18B20 can be validated like good I think....

Etape 2 : 

I have erase my Est-Poe-32 Tasmota and flash with Esphome and not change the wiring of the 6 DS18B20
I will seen in the future days and week if I lost value at anytime .....

If not I can may be say this is the first Est-Po-32 who is defectuous
If yes I can may be say this is the software Esphome who have a bug

May be this need to be confirmed by flash the first Est-Poe-32 with Tasmota ??

titou4307, 2025-01-29:
Hi at all

After 3 days, the errors return (DS18B20 disappears and appear Na Value...

In the logs of Esphome : 
```
[22:19:24][W][component:170]: Component dallas_temp.sensor cleared Warning flag
[22:19:25][W][dallas.temp.sensor:139]: 'temperature 2 Réfrigérateur' - Scratch pad checksum invalid!
[22:19:25][W][component:157]: Component dallas_temp.sensor set Warning flag: scratch pad checksum invalid
```

I have no solution and I am 99% sure that is a bug of Esphome .....

So I have not solution for solve this problem

jaysalazar, 2025-01-30:
This worked for me. Prior to adding this I could not get my ESP32 to find the sensor at all.

external_components:
  - source:
      type: git
      url: https://github.com/ssieb/esphome
      ref: onewire
    components: [ gpio ]
    refresh: 1min

titou4307, 2025-02-03:
At this time I sure the bug come since Esphome....

@jaysalazar 

If I will try your proposition of solution (thanks for this)
Do you think it possible to put more time for refresh ? I would like tu put maybe 2 or 5 minutes between.....

seime, 2025-02-23:
Just saw this issue on an Olimex ESP32 PoE board. Last 2 weeks 1 of 5 sensors (spread over 4 buses - this is the only sensor on the bus) started failing with `Component dallas_temp.sensor set Warning flag: scratch pad checksum invalid`.

Updated to 2025.2 without success. Power-cycling device helped though.