esphome/issues #6628: Irregular software/system restarts, no abnormalities found

AbianTang, 2025-01-02:
### The problem

Irregular software/system restarts, no abnormalities found。
[image]
[image]
[image]|Flash: 4096kB Speed:40MHz Mode:DOUT|Chip: 0x00a7e3df|SDK: 2.2.2-dev(38a443e)|Core: 3.1.2|Boot: 31|Mode: 1|CPU: 80|Flash: 0x001660c4|Reset: Software/System restart|Software/System restart

### Which version of ESPHome has the issue?

2024.12.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.7

### What platform are you using?

ESP8266

### Board

nodemcuv2

### YAML Config

```yaml
captive_portal:

i2c:
- id: bus1
  sda: D5
  scl: D6
  scan: True
  frequency: 400kHz

light:
  - platform: status_led
    name: "Switch state"
    pin:
     number: D4
     inverted: true

ld2410:

sensor:  

  - platform: bmp280_i2c
    i2c_id: bus1
    temperature:
      name: "BMP280 Temperature"
      oversampling: 16x
      accuracy_decimals: 4
    pressure:
      name: "BMP280 Pressure"
      accuracy_decimals: 4
    address: 0x76
    update_interval: 15s 
    

  - platform: aht10
    variant: AHT20
    i2c_id: bus1
    temperature:
      name: "Living Room Temperature"
      id: temperature1
      accuracy_decimals: 4
    humidity:
      name: "Living Room Humidity"
      id: humidity1
      accuracy_decimals: 4
    update_interval: 15s
    

  - platform: template
    name: "Dew Point Magnus-Tetens"
    unit_of_measurement: "°C"
    id: dew_point_Magnus_Tetens
    update_interval: 16s
    lambda: |-
      float temperature = id(temperature1).state;
      float humidity = id(humidity1).state;

      // Check if the sensor values are valid
      if (isnan(temperature) || isnan(humidity) || humidity < 0 || humidity > 100) {
        return NAN;  // Handle invalid sensor values
      }

      // Define constants for Magnus-Tetens equation
      const float a = 17.271;
      const float b = 237.7;

      // Calculate alpha
      float alpha = (a * temperature) / (b + temperature) + log(humidity / 100.0);

      // Calculate Dew Point
      return (b * alpha) / (a - alpha);  

  - platform: wifi_signal # Reports the WiFi signal strength/RSSI in dB
    name: "WiFi Signal dB"
    id: wifi_signal_db
    update_interval: 60s
    entity_category: "diagnostic"
    device_class: "signal_strength"

  - platform: copy # Reports the WiFi signal strength in %
    source_id: wifi_signal_db
    name: "WiFi Signal Percent"
    filters:
      - lambda: return min(max(2 * (x + 100.0), 0.0), 100.0);
    unit_of_measurement: "Signal %"
    entity_category: "diagnostic"
    device_class: "signal_strength"

  - platform: adc
    pin: vcc
    name: "VCC Voltage"
    id: vcc_voltage
    update_interval: 180s    

  - platform: uptime
    name: Uptime
    type: seconds
    id: sys_uptime
    update_interval: 60s

  - platform: ld2410
    light:
      name: light
    moving_distance:
      name : Moving Distance
    still_distance:
      name: Still Distance
    moving_energy:
      name: Move Energy
    still_energy:
      name: Still Energy
    detection_distance:
      name: Detection Distance
    g0:
      move_energy:
        name: g0 move energy
      still_energy:
        name: g0 still energy
    g1:
      move_energy:
        name: g1 move energy
      still_energy:
        name: g1 still energy
    g2:
      move_energy:
        name: g2 move energy
      still_energy:
        name: g2 still energy
    g3:
      move_energy:
        name: g3 move energy
      still_energy:
        name: g3 still energy
    g4:
      move_energy:
        name: g4 move energy
      still_energy:
        name: g4 still energy
    g5:
      move_energy:
        name: g5 move energy
      still_energy:
        name: g5 still energy
    g6:
      move_energy:
        name: g6 move energy
      still_energy:
        name: g6 still energy
    g7:
      move_energy:
        name: g7 move energy
      still_energy:
        name: g7 still energy
    g8:
      move_energy:
        name: g8 move energy
      still_energy:
        name: g8 still energy        
  - platform: debug
    free:
      name: "Heap Free"
    fragmentation:
      name: "Heap Fragmentation"
    block:
      name: "Heap Max Block"
    loop_time:
      name: "Loop Time"
   

switch:
  - platform: gpio
    pin: D1
    name: "Living Room Open window"
    device_class: switch
    id: relay1
    interlock: [relay2]
    on_turn_on:
    - delay: 14s
    - switch.turn_off: relay1
    - logger.log: "Switch Turned On!"
    

  - platform: gpio
    pin: D2
    name: "Living Room Closs window"
    device_class: switch  
    id: relay2
    interlock: [relay1]
    on_turn_on:
    - delay: 15s
    - switch.turn_off: relay2
    - logger.log: "Switch Turned Off!"

  - platform: ld2410
    engineering_mode:
      name: "engineering mode"
    bluetooth:
      name: "control bluetooth"    

binary_sensor:
  - platform: ld2410
    has_target:
      name: Presence
    has_still_target:
      name: Still Target
    out_pin_presence_status:
      name: out pin presence status
  - platform: gpio
    pin: D3
    name: gpio out pin presence
    device_class: presence 

debug:
  update_interval: 1s

text_sensor:
  - platform: wifi_info
    ip_address:
      name: ESP IP Address

  - platform: ld2410
    version:
      name: "firmware version"
    mac_address:
      name: "mac address"   

  - platform: debug
    device:
      name: "Device Info"
    reset_reason:
      name: "Reset Reason"    

      

uart:
  tx_pin: GPIO1
  rx_pin: GPIO3
  baud_rate: 256000
  parity: NONE
  stop_bits: 1         
  

select:
  - platform: ld2410
    distance_resolution:
      name: "distance resolution"
    baud_rate:
      name: "baud rate"
    light_function:
      name: light function
    out_pin_level:
      name: out pin level
```

### Anything in the logs that might be useful for us?

```txt
2024.12.2|Flash: 4096kB Speed:40MHz Mode:DOUT|Chip: 0x00a7e3df|SDK: 2.2.2-dev(38a443e)|Core: 3.1.2|Boot: 31|Mode: 1|CPU: 80|Flash: 0x001660c4|Reset: Software/System restart|Software/System restart

日志记录器: aioesphomeapi.connection
来源: components/esphome/manager.py:846
首次出现: 2025年1月1日 19:41:26 (5 次总计出现)
上次记录: 19:11:40

diy-window @ 192.168.31.136: disconnect request failed: [Errno 104] Connection reset by peer
```

ssieb, 2025-01-03:
You will need to get serial logs to find out why it's restarting, but that will be difficult since you're using the uart.  Did you disable serial logging?  You didn't include that part.

The graph is interesting.  The heap graph becomes unusually stable before it reboots.  Does it always do that?
Your available RAM is quite low.  I wouldn't be surprised if that's the problem.

AbianTang, 2025-01-03:
> 您需要获取串行日志来找出它重启的原因，但由于您使用的是 uart，因此这将很困难。您是否禁用了串行日志记录？您没有包括该部分。
> 
> 这张图很有趣。堆图在重新启动之前变得异常稳定。它总是这样吗？ 你的可用 RAM 很低。如果这是问题所在，我不会感到惊讶。
exception:4 flag:3 (Software
Watchdog)
epc1:0x4026f762 epc2:0x00000000
ерс3:0x0000

closed 2026-05-08 by github-actions[bot]