esphome/issues #6629: Respeaker Lite (with ESP32-S3) - Deprecated APIs etc.

ronanhalpenny, 2025-01-02:
### The problem

- Trying to install Respeaker Lite.
- Device is visible and "Online" in ESPHome, within HA.
- YAML code provided by Seeed Studio guide is failing, referring to deprecated RMT drivers and deprecated I2S APIs (log attached).

### Which version of ESPHome has the issue?

2024.12.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

Core 2024.12.5, SV 2024.12.0, OS 14.1

### What platform are you using?

ESP32

### Board

Respeaker Lite with ESP32-S3

### YAML Config

```yaml
esphome:
  name: esp32s3
  friendly_name: ReSpeakerv3
  platformio_options:
    board_build.flash_mode: dio
    board_build.mcu: esp32s3
  # on_boot:
  #   then:
  #     - if:
  #            condition:
  #              switch.is_on: 
  #            then:
  #              - voice_assistant.start_continuous:

esp32:
  board: esp32-s3-devkitc-1
  variant: esp32s3
  framework:
    type: esp-idf
    version: recommended

logger:
  level: VERY_VERBOSE

api:

wifi:
  ssid: "..."
  password: "..."

captive_portal:

external_components:
  - source: github://QingWind6/ESPHome_XIAO-ESP32S3

i2s_audio_xiao:
  i2s_lrclk_pin: GPIO7
  i2s_bclk_pin: GPIO8
  i2s_mclk_pin: GPIO9

microphone:
[... 285 lines cut ...]
                 from src/esphome/components/i2s_audio_xiao/microphone/i2s_audio_microphone.cpp:1:
/data/cache/platformio/packages/framework-espidf/components/driver/deprecated/driver/i2s.h:27:2: warning: #warning "This set of I2S APIs has been deprecated, please include 'driver/i2s_std.h', 'driver/i2s_pdm.h' or 'driver/i2s_tdm.h' instead. if you want to keep using the old APIs and ignore this warning, you can enable 'Suppress leagcy driver deprecated warning' option under 'I2S Configuration' menu in Kconfig" [-Wcpp]
   27 | #warning "This set of I2S APIs has been deprecated, \
      |  ^~~~~~~
src/esphome/components/i2s_audio_xiao/i2s_audio_xiao.cpp: In member function 'virtual void esphome::i2s_audio_xiao::I2SAudioComponent::setup()':
src/esphome/components/i2s_audio_xiao/i2s_audio_xiao.cpp:15:24: error: 'I2S_NUM_MAX' was not declared in this scope; did you mean 'I2S_NUM_1'?
   15 |   if (next_port_num >= I2S_NUM_MAX) {
      |                        ^~~~~~~~~~~
      |                        I2S_NUM_1
Compiling .pioenvs/esp32s3/src/esphome/components/i2s_audio_xiao/speaker/i2s_audio_speaker.cpp.o
In file included from src/esphome/components/i2s_audio_xiao/speaker/../i2s_audio_xiao.h:5,
                 from src/esphome/components/i2s_audio_xiao/speaker/i2s_audio_speaker.h:5,
                 from src/esphome/components/i2s_audio_xiao/speaker/i2s_audio_speaker.cpp:1:
/data/cache/platformio/packages/framework-espidf/components/driver/deprecated/driver/i2s.h:27:2: warning: #warning "This set of I2S APIs has been deprecated, please include 'driver/i2s_std.h', 'driver/i2s_pdm.h' or 'driver/i2s_tdm.h' instead. if you want to keep using the old APIs and ignore this warning, you can enable 'Suppress leagcy driver deprecated warning' option under 'I2S Configuration' menu in Kconfig" [-Wcpp]
   27 | #warning "This set of I2S APIs has been deprecated, \
      |  ^~~~~~~
*** [.pioenvs/esp32s3/src/esphome/components/i2s_audio_xiao/i2s_audio_xiao.cpp.o] Error 1
src/esphome/components/i2s_audio_xiao/speaker/i2s_audio_speaker.cpp: In static member function 'static void esphome::i2s_audio_xiao::I2SAudioSpeaker::player_task(void*)':
src/esphome/components/i2s_audio_xiao/speaker/i2s_audio_speaker.cpp:54:24: error: 'I2S_MCLK_MULTIPLE_DEFAULT' was not declared in this scope; did you mean 'I2S_MCLK_MULTIPLE_512'?
   54 |       .mclk_multiple = I2S_MCLK_MULTIPLE_DEFAULT,
      |                        ^~~~~~~~~~~~~~~~~~~~~~~~~
      |                        I2S_MCLK_MULTIPLE_512
*** [.pioenvs/esp32s3/src/esphome/components/i2s_audio_xiao/speaker/i2s_audio_speaker.cpp.o] Error 1
========================= [FAILED] Took 105.07 seconds =========================
```

randybb, 2025-01-02:
Warnings are not something that are causing problems. Anyway, RMT component has been updated already in DEV.

This I2S warnings and declaration issues are related to the external component which you are using => out of scope of this project.

closed 2025-01-02 by randybb

ronanhalpenny, 2025-01-02:
OK, another user has explained that when they downgraded the ESPHome version to 2024.11.3, it worked as expected, hence I wondered if it was an ESPHome issue. Does this shed any relevant context? I am unable to create the file that is needed to update the device via ESPHome Web interface, because the process is failing, per above.

randybb, 2025-01-02:
esphome 2024.12 switched to idf 5.1.3 (was 4.4.8 previously), so the ext component needs to be updated.

you can switch to older IDF by overriding version/platform_version:
```yaml
esp32:
  framework:
    type: esp-idf
    version: 4.4.8
    platform_version: 5.4.0
```

ronanhalpenny, 2025-01-02:
I'm so appreciate of that, thank you. I had to tweak it slightly to add in board and variant but then it was fine.
`esp32:
  board: esp32-s3-devkitc-1
  variant: esp32s3
  framework:
    type: esp-idf
    version: 4.4.8
    platform_version: 5.4.0`