esphome/issues #6630: add_idf_sdkconfig_option being overridden by default value

olicooper, 2025-01-02:
### The problem

I am trying to use the `tinyusb` esp-idf component/library to communicate with an external serial device over USB with CDC-ACM but I am struggling to get the project to build. One of the issues is that I am unable to set sdkconfig options for the library including setting `CONFIG_TINYUSB_CDC_ENABLED`... The value is not set in the sdkconfig file despite setting it in the python build script.

```python
async def to_code(config):
    if core.CORE.using_esp_idf:
        # bug: this value isn't set
        esp32.add_idf_sdkconfig_option("CONFIG_TINYUSB_CDC_ENABLED", True)
        # I also tried adding the build flag to allow it to compile, but other errors appear
        cg.add_build_flag("-DCONFIG_TINYUSB_CDC_ENABLED=1")
        esp32.add_idf_component(
            name="tinyusb",
            repo="https://github.com/espressif/tinyusb.git",
            refresh=core.TimePeriod(days=1),
            ref="release/v0.17")
        esp32.add_idf_component(
            name="esp_tinyusb",
            repo="https://github.com/espressif/esp-usb.git",
            refresh=core.TimePeriod(days=1),
            path="device/esp_tinyusb")
```

### Which version of ESPHome has the issue?

2024.12.1

### What type of installation are you using?

pip

### What platform are you using?

ESP32-IDF

clydebarrow, 2025-05-03:
Just set it in the YAML:

```
esp32:
  board: esp32dev
  framework:
    type: esp-idf
    sdkconfig_options:
      CONFIG_TINYUSB_CDC_ENABLED: "y"
```

clydebarrow, 2025-05-03:
You may also want to look at https://github.com/esphome/esphome/pull/8334

closed 2025-05-03 by clydebarrow