esphome/issues #6632: build esp8266 error in docker container

intern, 2025-01-02:
### The problem

### ERROR log:
```
INFO ESPHome 2024.12.2
INFO Reading configuration /config/test.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing test (board: esp01_1m; framework: arduino; platform: platformio/espressif8266@4.2.1)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 1MB Flash
Dependency Graph
|-- ESPAsyncTCP-esphome @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.2.2
|-- DNSServer @ 1.1.1
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
|-- noise-c @ 0.1.6
|-- ArduinoJson @ 6.18.5
Compiling .pioenvs/test/src/esphome/components/api/api_connection.cpp.o
Compiling .pioenvs/test/src/esphome/components/api/api_frame_helper.cpp.o
Compiling .pioenvs/test/src/esphome/components/api/api_pb2.cpp.o
In file included from src/esphome/components/api/api_connection.h:5,
                 from src/esphome/components/api/api_connection.cpp:1:
src/esphome/components/api/api_frame_helper.h:2:10: fatal error: cstdint: No such file or directory
    2 | #include <cstdint>
      |          ^~~~~~~~~
compilation terminated.
Compiling .pioenvs/test/src/esphome/components/api/api_pb2_service.cpp.o
In file included from src/esphome/components/api/api_frame_helper.cpp:1:
src/esphome/components/api/api_frame_helper.h:2:10: fatal error: cstdint: No such file or directory
    2 | #include <cstdint>
      |          ^~~~~~~~~
compilation terminated.
*** [.pioenvs/test/src/esphome/components/api/api_connection.cpp.o] Error 1
*** [.pioenvs/test/src/esphome/components/api/api_frame_helper.cpp.o] Error 1
In file included from src/esphome/core/component.h:3,
                 from src/esphome/components/api/proto.h:3,
                 from src/esphome/components/api/api_pb2.h:5,
                 from src/esphome/components/api/api_pb2.cpp:3:
/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/cmath:41:10: fatal error: bits/c++config.h: No such file or directory
   41 | #include <bits/c++config.h>
      |          ^~~~~~~~~~~~~~~~~~
compilation terminated.
In file included from src/esphome/core/component.h:3,
                 from src/esphome/components/api/proto.h:3,
                 from src/esphome/components/api/api_pb2.h:5,
                 from src/esphome/components/api/api_pb2_service.h:5,
                 from src/esphome/components/api/api_pb2_service.cpp:3:
/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/cmath:41:10: fatal error: bits/c++config.h: No such file or directory
   41 | #include <bits/c++config.h>
      |          ^~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/test/src/esphome/components/api/api_pb2.cpp.o] Error 1
*** [.pioenvs/test/src/esphome/components/api/api_pb2_service.cpp.o] Error 1
========================== [FAILED] Took 3.29 seconds ==========================
```

### config file
```yaml
esphome:
  name: test
  friendly_name: test

esp8266:
  board: esp01_1m

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "1cuwJMjdG4oT5PrPMSJ6UfZG07a8OqryuOfhG9fKfw8="

ota:
  - platform: esphome
    password: "3d1741399ba9a49e636e5d303ba7fe51"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Test Fallback Hotspot"
    password: "nPEepz64P0G6"

switch:
  - platform: gpio
    pin: 4
    name: "Living Room Dehumidifier"

web_server:
  port: 80
captive_portal:
    
```

### esphome version

```
esphome  --version
Version: 2024.12.2
```

### Which version of ESPHome has the issue?

2024.12.2

### What type of installation are you using?

Docker

### What platform are you using?

ESP8266

### YAML Config

```yaml
esphome:
  name: test
  friendly_name: test

esp8266:
  board: esp01_1m

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "1cuwJMjdG4oT5PrPMSJ6UfZG07a8OqryuOfhG9fKfw8="

ota:
  - platform: esphome
    password: "3d1741399ba9a49e636e5d303ba7fe51"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Test Fallback Hotspot"
    password: "nPEepz64P0G6"

switch:
  - platform: gpio
    pin: 4
    name: "Living Room Dehumidifier"

web_server:
  port: 80
captive_portal:
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2024.12.2
INFO Reading configuration /config/test.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing test (board: esp01_1m; framework: arduino; platform: platformio/espressif8266@4.2.1)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 1MB Flash
Dependency Graph
|-- ESPAsyncTCP-esphome @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.2.2
|-- DNSServer @ 1.1.1
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
|-- noise-c @ 0.1.6
|-- ArduinoJson @ 6.18.5
Compiling .pioenvs/test/src/esphome/components/api/api_connection.cpp.o
Compiling .pioenvs/test/src/esphome/components/api/api_frame_helper.cpp.o
Compiling .pioenvs/test/src/esphome/components/api/api_pb2.cpp.o
In file included from src/esphome/components/api/api_connection.h:5,
                 from src/esphome/components/api/api_connection.cpp:1:
src/esphome/components/api/api_frame_helper.h:2:10: fatal error: cstdint: No such file or directory
    2 | #include <cstdint>
      |          ^~~~~~~~~
compilation terminated.
Compiling .pioenvs/test/src/esphome/components/api/api_pb2_service.cpp.o
In file included from src/esphome/components/api/api_frame_helper.cpp:1:
src/esphome/components/api/api_frame_helper.h:2:10: fatal error: cstdint: No such file or directory
    2 | #include <cstdint>
      |          ^~~~~~~~~
compilation terminated.
*** [.pioenvs/test/src/esphome/components/api/api_connection.cpp.o] Error 1
*** [.pioenvs/test/src/esphome/components/api/api_frame_helper.cpp.o] Error 1
In file included from src/esphome/core/component.h:3,
                 from src/esphome/components/api/proto.h:3,
                 from src/esphome/components/api/api_pb2.h:5,
                 from src/esphome/components/api/api_pb2.cpp:3:
/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/cmath:41:10: fatal error: bits/c++config.h: No such file or directory
   41 | #include <bits/c++config.h>
      |          ^~~~~~~~~~~~~~~~~~
compilation terminated.
In file included from src/esphome/core/component.h:3,
                 from src/esphome/components/api/proto.h:3,
                 from src/esphome/components/api/api_pb2.h:5,
                 from src/esphome/components/api/api_pb2_service.h:5,
                 from src/esphome/components/api/api_pb2_service.cpp:3:
/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/cmath:41:10: fatal error: bits/c++config.h: No such file or directory
   41 | #include <bits/c++config.h>
      |          ^~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/test/src/esphome/components/api/api_pb2.cpp.o] Error 1
*** [.pioenvs/test/src/esphome/components/api/api_pb2_service.cpp.o] Error 1
========================== [FAILED] Took 3.29 seconds ==========================
```

lissyx, 2025-01-17:
Still happening with 2024.12.4

lissyx, 2025-01-20:
With 2024.12.4, and after cleaning `/data/{build,cache}` within the docker container, it's reinstalling the deps and now it works