esphome/issues #6633: Modbus has multiple timing issues

exciton, 2025-01-03:
### The problem

I've been working through a number of modbus timing issues which occur when trying to poll multiple devices in quick succession (in this case 2x sdm_meter devices). I tested these devices both using the sdm_meter component, and using a modbus_controller. There are problems with each.

The symptoms that appear include:
- CRC error when beginning of response is cut off. 
- Timeout when end of response is cut off (almost always at 120 bytes)
- Various errors that stem from that (retries can cascade the issue)

The issues:

- Frame delay timeout (3.5 characters from spec) is not honoured, incorrectly merging unrelated bytes
- Turnaround delay (application specific) is not present, allowing tx too soon after rx or tx
- UART buffer (120 bytes by default) causes timeouts for long responses which get broken into pieces
- TX is synchronous (with flush) causing long component loop execution times
- Waiting for response didn't care if response was from correct address
- Waiting for response is used for server mode which cannot receive responses
- waiting_for_response() only blocks when waiting, doesn't block for frame delay or turnaround delay
- Multiple components inherit from ModbusDevice but don't block for waiting_for_response() : sdm_meter, havells_solar, kuntze, pzemac, pzemdc, selec_meter

I have built a patch which solves each of them (some elegantly, some less so). 

### Which version of ESPHome has the issue?

2024.12.2

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Component causing the issue

modbus

### YAML Config

```yaml
esphome:
  name: power-system
  friendly_name: Power System

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:
  baud_rate: 0
  level: VERBOSE
  logs:
    modbus: VERBOSE
    modbus_controller: VERBOSE
    sensor: DEBUG

# Enable Home Assistant API
api:
  encryption:
    key: !secret api_key 

ota:
  - platform: esphome
    password: !secret ota_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

external_components:
  - source: components

i2c:
  sda: 16
  scl: 17
  frequency: 800kHz

uart:
  tx_pin: 1
  rx_pin: 3
  baud_rate: 9600
  debug:

modbus:
  flow_control_pin: 4
  send_wait_time: 200ms
  turnaround_time: 200ms #This is from the patched version

# modbus_controller:
#  - address: 2
#    id: well_power_meter
#    update_interval: 2s 
#    #command_throttle: 200ms
#    max_cmd_retries: 0
#  - address: 3
#    id: pressure_power_meter
#    update_interval: 2s 
#    #command_throttle: 200ms
#    max_cmd_retries: 0
  

sensor:

  - platform: sdm_meter
    address: 2
    update_interval: 1s
    phase_a:
      current:
        name: "Well Pump Current"
      voltage:
        name: "Well Pump Voltage"
  - platform: sdm_meter
    address: 3
    update_interval: 1s
    phase_a:
      current:
        name: "Pressure Pump Current"
      voltage:
        name: "Pressure Pump Voltage"
  # - platform: modbus_controller
  #   modbus_controller_id: well_power_meter
  #   name: "Well Pump Voltage"
  #   address: 0x0000
  #   register_count: 2
  #   unit_of_measurement: "V"
  #   register_type: read
  #   value_type: FP32
  #   accuracy_decimals: 3
  #   device_class: voltage
  # - platform: modbus_controller
  #   modbus_controller_id: well_power_meter
  #   name: "Well Pump Current"
  #   address: 0x0006
  #   register_count: 100
  #   unit_of_measurement: "A"
  #   register_type: read
  #   value_type: FP32
  #   accuracy_decimals: 3
  #   device_class: current
  # - platform: modbus_controller
  #   modbus_controller_id: pressure_power_meter
  #   name: "Pressure Pump Voltage"
  #   address: 0x0000
  #   register_count: 2
  #   unit_of_measurement: "V"
  #   register_type: read
  #   value_type: FP32
  #   accuracy_decimals: 3
  #   device_class: voltage
  # - platform: modbus_controller
  #   modbus_controller_id: pressure_power_meter
  #   name: "Pressure Pump Current"
  #   address: 0x0006
  #   register_count: 2
  #   unit_of_measurement: "A"
  #   register_type: read
  #   value_type: FP32
  #   accuracy_decimals: 3
  #   device_class: current
  # - platform: modbus_controller
  #   modbus_controller_id: pressure_power_meter
  #   name: "Pressure Pump Active Power"
  #   address: 0x000C
  #   register_count: 2
  #   unit_of_measurement: "W"
  #   register_type: read
  #   value_type: FP32
  #   accuracy_decimals: 3
  #   device_class: power
  # - platform: modbus_controller
  #   modbus_controller_id: pressure_power_meter
  #   name: "Pressure Pump Aparent Power"
  #   address: 0x0012
  #   register_count: 2
  #   unit_of_measurement: "W"
  #   register_type: read
  #   value_type: FP32
  #   accuracy_decimals: 3
  #   device_class: power
```

KrzysztofHajdamowicz, 2025-05-20:
I guess I had one of these issues.
I have 9 SDM meters on single bus.

ESPHome tried to DDoS the bus with all requests at once without spreading them evenly.
In the end, I've wrote custom scheduler for that: https://community.home-assistant.io/t/esp32-9x-eastron-sdm-does-not-refresh-all-entities/892593/2

dominikstupka, 2025-05-21:
Hi, is there a plan to include this fix in an ESPHome release? I am experiencing the same issues with Modbus. The biggest complication is that ESPHome does not respect the silence interval when communicating with multiple devices. Thanks a lot!

exciton, 2025-05-22:
You can use the code here https://github.com/esphome/esphome/pull/8032 to fix this until it's merged:

```yaml
external_components:
    - source: github://pr#8032
      components: [ modbus, uart, modbus_controller ]

uart:
  rx_full_threshold: 1

modbus:
  send_wait_time: 600ms
  turnaround_time: 100ms
```

dominikstupka, 2025-05-22:
> You can use the code here [esphome/esphome#8032](https://github.com/esphome/esphome/pull/8032) to fix this until it's merged:
> 
> external_components:
>     - source: github://pr#8032
>       components: [ modbus, uart, modbus_controller ]
> 
> uart:
>   rx_full_threshold: 1
> 
> modbus:
>   send_wait_time: 600ms
>   turnaround_time: 100ms

Hi Bonnen, thanks for the guide! I also appreciate your improvements to the Modbus component. I think your modification significantly improves its functionality.

ruimarinho, 2025-06-28:
@exciton I'm not getting any data if I use the modbus change with this PR. Does anything look suspicious to you in the following log? It works reasonable well — with some of the timing issues mentioned on this PR — with the main branch (2025.6.2).

```
[00:39:28][V][modbus_controller:232]: Updating modbus component
[00:39:28][V][modbus_controller:199]: Range : BCEA Size: 1 (0) skip: 0
[00:39:28][V][modbus_controller:199]: Range : C Size: 1 (4) skip: 0
[00:39:28][V][modbus_controller:199]: Range : 1 Size: 1 (4) skip: 0
[00:39:28][V][modbus_controller:199]: Range : 3 Size: 3 (4) skip: 0
[00:39:28][V][modbus_controller:199]: Range : 4 Size: 3 (4) skip: 0
[00:39:28][V][modbus_controller:199]: Range : 5 Size: 3 (4) skip: 0
[00:39:28][V][modbus_controller:199]: Range : 6 Size: 2 (4) skip: 0
[00:39:28][V][modbus_controller:199]: Range : A Size: 1 (4) skip: 0
[00:39:28][V][modbus_controller:199]: Range : D Size: 5 (4) skip: 0
[00:39:28][V][modbus_controller:044]: Sending next modbus command to device 1 register 0xBCEA count 1
[00:39:28][V][modbus_controller:569]: Command sent 0 0xBCEA 1 send_count: 1
[00:39:28][V][modbus:239]: Modbus write: 01.04.00.80.00.01.30.22 (8) 13315ms after last send
[00:39:29][V][modbus:077]: Modbus received first Byte 1 (0X1) 34ms after last send
[00:39:29][V][modbus:321]: Clearing buffer of 13 bytes - parse succeeded 53ms after last send
[00:39:29][V][modbus_controller:044]: Sending next modbus command to device 1 register 0xBCEA count 1
[00:39:29][V][modbus_controller:569]: Command sent 0 0xBCEA 1 send_count: 2
[00:39:29][V][modbus:239]: Modbus write: 01.04.00.80.00.01.30.22 (8) 906ms after last send
[00:39:29][V][modbus:077]: Modbus received first Byte 1 (0X1) 35ms after last send
[00:39:29][V][modbus:321]: Clearing buffer of 13 bytes - parse succeeded 55ms after last send
[00:39:30][V][modbus_controller:044]: Sending next modbus command to device 1 register 0xBCEA count 1
[00:39:30][V][modbus_controller:569]: Command sent 0 0xBCEA 1 send_count: 3
[00:39:30][V][modbus:239]: Modbus write: 01.04.00.80.00.01.30.22 (8) 906ms after last send
[00:39:30][V][modbus:077]: Modbus received first Byte 1 (0X1) 33ms after last send
[00:39:30][V][modbus:321]: Clearing buffer of 13 bytes - parse succeeded 52ms after last send
[00:39:31][V][modbus_controller:044]: Sending next modbus command to device 1 register 0xBCEA count 1
[00:39:31][V][modbus_controller:569]: Command sent 0 0xBCEA 1 send_count: 4
[00:39:31][V][modbus:239]: Modbus write: 01.04.00.80.00.01.30.22 (8) 912ms after last send
[00:39:31][V][modbus:077]: Modbus received first Byte 1 (0X1) 32ms after last send
[00:39:31][V][modbus:321]: Clearing buffer of 13 bytes - parse succeeded 51ms after last send
[00:39:32][W][modbus_controller:027]: Modbus device=1 set offline
[00:39:32][D][modbus_controller:040]: Modbus command to device=1 register=0xBCEA no response received - removed from send queue
[00:39:32][V][modbus_controller:044]: Sending next modbus command to device 1 register 0x0C count 1
[00:39:32][V][modbus_controller:569]: Command sent 4 0xC 1 send_count: 1
[00:39:32][V][modbus:239]: Modbus write: 01.04.00.0C.00.01.F1.C9 (8) 919ms after last send
[00:39:32][V][modbus:077]: Modbus received first Byte 1 (0X1) 32ms after last send
[00:39:32][V][modbus:321]: Clearing buffer of 9 bytes - parse succeeded 35ms after last send
[00:39:33][V][modbus_controller:044]: Sending next modbus command to device 1 register 0x0C count 1
[00:39:33][V][modbus_controller:569]: Command sent 4 0xC 1 send_count: 2
[00:39:33][V][modbus:239]: Modbus write: 01.04.00.0C.00.01.F1.C9 (8) 907ms after last send
[00:39:33][V][modbus:077]: Modbus received first Byte 1 (0X1) 33ms after last send
[00:39:33][V][modbus:321]: Clearing buffer of 8 bytes - timeout 52ms after last send
[00:39:33][V][modbus:077]: Modbus received first Byte 93 (0X5d) 72ms after last send
[00:39:33][V][modbus:321]: Clearing buffer of 1 bytes - timeout 93ms after last send
[00:39:36][V][modbus:042]: Stop waiting for response from 1 2503ms after last send
[00:39:36][V][modbus_controller:044]: Sending next modbus command to device 1 register 0x0C count 1
[00:39:36][V][modbus_controller:569]: Command sent 4 0xC 1 send_count: 3
[00:39:36][V][modbus:239]: Modbus write: 01.04.00.0C.00.01.F1.C9 (8) 2536ms after last send
[00:39:36][V][modbus:077]: Modbus received first Byte 1 (0X1) 120ms after last send
[00:39:36][V][modbus:321]: Clearing buffer of 9 bytes - parse succeeded 124ms after last send
[00:39:36][V][modbus_controller:044]: Sending next modbus command to device 1 register 0x0C count 1
[00:39:36][V][modbus_controller:569]: Command sent 4 0xC 1 send_count: 4
[00:39:36][V][modbus:239]: Modbus write: 01.04.00.0C.00.01.F1.C9 (8) 907ms after last send
[00:39:37][V][modbus:077]: Modbus received first Byte 1 (0X1) 32ms after last send
[00:39:37][V][modbus:321]: Clearing buffer of 9 bytes - parse succeeded 35ms after last send
[00:39:37][D][modbus_controller:040]: Modbus command to device=1 register=0x0C no response received - removed from send queue
[00:39:37][V][modbus_controller:044]: Sending next modbus command to device 1 register 0x01 count 1
[00:39:37][V][modbus_controller:569]: Command sent 4 0x1 1 send_count: 1
[00:39:37][V][modbus:239]: Modbus write: 01.04.00.01.00.01.60.0A (8) 922ms after last send
[00:39:37][V][modbus:077]: Modbus received first Byte 1 (0X1) 66ms after last send
[00:39:38][V][modbus:321]: Clearing buffer of 17 bytes - parse succeeded 92ms after last send
[... cut ...]
[00:39:49][V][modbus:321]: Clearing buffer of 17 bytes - parse succeeded 118ms after last send
[00:39:50][V][modbus_controller:044]: Sending next modbus command to device 1 register 0x05 count 3
[00:39:50][V][modbus_controller:569]: Command sent 4 0x5 3 send_count: 3
[00:39:50][V][modbus:239]: Modbus write: 01.04.00.05.00.03.A0.0A (8) 900ms after last send
[00:39:50][V][modbus:077]: Modbus received first Byte 1 (0X1) 69ms after last send
[00:39:50][V][modbus:321]: Clearing buffer of 17 bytes - parse succeeded 90ms after last send
```

k0cka1, 2025-10-03:
@exciton appreciate your work on this. I was also having similar issues, tried your pr and it fixed basically all my issues. One thing I'm still trying to figure out is, if I set 3 of my slave devices to update every 250ms, it seems like one of them is not updating every cycle. Do you have any idea why is that? Is updating 3 devices that often too much? I tried to play with the settings, tried every combination but this is the best I could do. This is part of my current yaml configuration: 
`external_components:
  - source: github://pr#8103
    components: [uart]
  - source: github://pr#8032
    components: [modbus, modbus_controller]
    refresh: 1h
    
uart:
  id: uart_1
  tx_pin: 32 #TX
  rx_pin: 35 #RX
  baud_rate: 115200
  stop_bits: 1
  data_bits: 8
  parity: NONE
  rx_full_threshold: 114
  flow_control_pin: GPIO4
  rx_buffer_size: 1024

modbus:
  #flow_control_pin: GPIO4
  id: modbus_1
  uart_id: uart_1
  role: client
  send_wait_time: 60ms
  turnaround_time: 60ms

modbus_controller:

  #3-GANG WALL SWITCH
  - id: modbus_slave_2
    address: 0x02
    modbus_id: modbus_1
    update_interval: 250ms
    allow_duplicate_commands: True
    max_cmd_retries: 1

  #3-GANG WALL SWITCH
  - id: modbus_slave_3
    address: 0x03
    modbus_id: modbus_1
    update_interval: 250ms
    max_cmd_retries: 1
    allow_duplicate_commands: True

  #3-GANG WALL SWITCH
  - id: modbus_slave_4
    address: 0x04
    modbus_id: modbus_1
    update_interval: 250ms
    max_cmd_retries: 1
    allow_duplicate_commands: True`
and here are the extracts from slave logs...

SLAVE 1:
`21:14:36
[modbus:174]
Got Modbus frame from unknown address 0x03!
21:14:36
[modbus:174]
Got Modbus frame from unknown address 0x04!
21:14:36
[modbus_controller:110]
Received read holding/input registers for device 0x2. FC: 0x4. Start address: 0x15. Number of registers: 0x1.
21:14:36
[modbus_controller:124]
Matched register. Address: 0x15. Value type: 1. Register count: 1. Value: 0.
21:14:36
[modbus:174]
[... 215 lines cut ...]
Got Modbus frame from unknown address 0x03!
21:15:20
[modbus:174]
Got Modbus frame from unknown address 0x02!
21:15:20
[modbus:174]
Got Modbus frame from unknown address 0x03!
21:15:20
[modbus_controller:110]
Received read holding/input registers for device 0x4. FC: 0x4. Start address: 0x15. Number of registers: 0x1.
21:15:20
[modbus_controller:124]
Matched register. Address: 0x15. Value type: 1. Register count: 1. Value: 0.
21:15:20
[modbus:174]
Got Modbus frame from unknown address 0x02!
21:15:21
[modbus:174]
Got Modbus frame from unknown address 0x03!
21:15:21
[modbus:174]
Got Modbus frame from unknown address 0x02!
21:15:21
[modbus:174]
`

exciton, 2025-10-03:
Can you add the log from the client (master)? All the clues will be there.

I expect your send_wait_time is too short on the client. Try increasing it to 2000ms. There's no performance advantage to a short send_wait_time, especially if all your devices are actually online. If it's short then the client doesn't wait long enough to receive a response that does actually come later, and it causes it to request again.

k0cka1, 2025-10-03:
@exciton I have set send_wait_time to 2000ms, max_cmd_retries to 0 and allow_duplicate_commands to false. And this is the log from the master: 

```
22:33:14
[V]
[modbus:084]
Received first byte 3 (0X3) 19ms after last send
22:33:14
[V]
[modbus:360]
Clearing buffer of 7 bytes - parse succeeded 24ms after last send
22:33:14
[V]
[modbus:271]
Write: 02.04.00.15.00.01.20.3D (8) 112ms after last send
22:33:14
[V]
[modbus:084]
Received first byte 2 (0X2) 18ms after last send
22:33:14
[V]
[modbus:360]
Clearing buffer of 7 bytes - parse succeeded 23ms after last send
22:33:14
[V]
[modbus:271]
Write: 03.04.00.05.00.01.20.29 (8) 116ms after last send
22:33:14
[V]
[modbus:084]
Received first byte 3 (0X3) 18ms after last send
22:33:14
[V]
[modbus:360]
Clearing buffer of 7 bytes - parse succeeded 22ms after last send
22:33:14
[V]
[modbus:271]
Write: 02.04.00.15.00.01.20.3D (8) 108ms after last send
22:33:14
[V]
[modbus:084]
Received first byte 2 (0X2) 18ms after last send
22:33:14
[V]
[modbus:360]
Clearing buffer of 7 bytes - parse succeeded 22ms after last send
22:33:14
[V]
[modbus:271]
Write: 03.04.00.05.00.01.20.29 (8) 116ms after last send
22:33:15
[V]
[modbus:084]
Received first byte 3 (0X3) 18ms after last send
22:33:15
[V]
[modbus:360]
Clearing buffer of 7 bytes - parse succeeded 22ms after last send
22:33:15
[V]
[modbus:271]
Write: 04.04.00.15.00.01.20.5B (8) 106ms after last send
22:33:15
[V]
[modbus:084]
Received first byte 4 (0X4) 19ms after last send
22:33:15
[V]
[modbus:360]
Clearing buffer of 7 bytes - parse succeeded 24ms after last send
22:33:15
[V]
[modbus:271]
Write: 02.04.00.15.00.01.20.3D (8) 110ms after last send
22:33:15
[V]
[modbus:084]
Received first byte 2 (0X2) 19ms after last send
22:33:15
[V]
[modbus:360]
Clearing buffer of 7 bytes - parse succeeded 23ms after last send
22:33:15
[V]
[modbus:271]
Write: 03.04.00.05.00.01.20.29 (8) 118ms after last send
22:33:15
[V]
[modbus:084]
Received first byte 3 (0X3) 14ms after last send
22:33:15
[V]
[modbus:360]
Clearing buffer of 7 bytes - parse succeeded 20ms after last send
22:33:15
[V]
[modbus:271]
Write: 02.04.00.15.00.01.20.3D (8) 112ms after last send
22:33:15
[V]
[modbus:084]
Received first byte 2 (0X2) 19ms after last send
22:33:15
[V]
[modbus:360]
Clearing buffer of 7 bytes - parse succeeded 24ms after last send
22:33:15
[V]
[modbus:271]
Write: 04.04.00.15.00.01.20.5B (8) 116ms after last send
22:33:15
[V]
[modbus:084]
Received first byte 4 (0X4) 19ms after last send
22:33:15
[V]
[modbus:360]
Clearing buffer of 7 bytes - parse succeeded 24ms after last send
22:33:15
[V]
[modbus:271]
Write: 02.04.00.15.00.01.20.3D (8) 116ms after last send
22:33:15
[V]
[modbus:084]
Received first byte 2 (0X2) 18ms after last send
22:33:15
[V]
[modbus:360]
Clearing buffer of 7 bytes - parse succeeded 22ms after last send
22:33:15
[V]
[modbus:271]
Write: 03.04.00.05.00.01.20.29 (8) 116ms after last send
22:33:15
[V]
[modbus:084]
Received first byte 3 (0X3) 19ms after last send
22:33:15
[V]
[modbus:360]
Clearing buffer of 7 bytes - parse succeeded 24ms after last send
22:33:15
[V]
[modbus:271]
Write: 02.04.00.15.00.01.20.3D (8) 108ms after last send
22:33:15
[V]
[modbus:084]
Received first byte 2 (0X2) 17ms after last send
22:33:15
[V]
[modbus:360]
Clearing buffer of 7 bytes - parse succeeded 21ms after last send
22:33:16
[V]
[modbus:271]
Write: 03.04.00.05.00.01.20.29 (8) 115ms after last send
22:33:16
[V]
[modbus:084]
Received first byte 3 (0X3) 19ms after last send
22:33:16
[V]
[modbus:360]
Clearing buffer of 7 bytes - parse succeeded 24ms after last send
22:33:16
[V]
[modbus:271]
Write: 02.04.00.15.00.01.20.3D (8) 109ms after last send
22:33:16
[V]
[modbus:084]
Received first byte 2 (0X2) 19ms after last send
22:33:16
[V]
[modbus:360]
Clearing buffer of 7 bytes - parse succeeded 24ms after last send
22:33:16
[V]
[modbus:271]
Write: 03.04.00.05.00.01.20.29 (8) 114ms after last send
22:33:16
[V]
[modbus:084]
Received first byte 3 (0X3) 19ms after last send
22:33:16
[V]
[modbus:360]
Clearing buffer of 7 bytes - parse succeeded 24ms after last send
22:33:16
[V]
[modbus:271]
Write: 02.04.00.15.00.01.20.3D (8) 107ms after last send
22:33:16
[V]
[modbus:084]
Received first byte 2 (0X2) 18ms after last send
```

k0cka1, 2025-10-03:
@exciton I was able to go to turnaround_time: 30ms with send_wait_time: 200ms without warnings or CRC errors. It seems like its a little bit better but it still not completly balanced :/ I also tried to set update_interval to 250ms for slave 1, 260ms for slave 2 and 270ms for slave 3 but it didn't make a difference. This is the log from the slave 3 

```
23:48:16
[modbus:174]
Got Modbus frame from unknown address 0x03!
23:48:16
[modbus_controller:110]
Received read holding/input registers for device 0x4. FC: 0x4. Start address: 0x15. Number of registers: 0x1.
23:48:16
[modbus_controller:124]
Matched register. Address: 0x15. Value type: 1. Register count: 1. Value: 511.
23:48:16
[modbus:174]
Got Modbus frame from unknown address 0x02!
23:48:16
[modbus:174]
Got Modbus frame from unknown address 0x03!
23:48:16
[modbus:174]
Got Modbus frame from unknown address 0x02!
23:48:16
[modbus:174]
Got Modbus frame from unknown address 0x03!
23:48:16
[modbus_controller:110]
Received read holding/input registers for device 0x4. FC: 0x4. Start address: 0x15. Number of registers: 0x1.
23:48:16
[modbus_controller:124]
Matched register. Address: 0x15. Value type: 1. Register count: 1. Value: 511.
23:48:16
[modbus:174]
Got Modbus frame from unknown address 0x02!
23:48:16
[modbus:174]
Got Modbus frame from unknown address 0x03!
23:48:17
[modbus:174]
Got Modbus frame from unknown address 0x02!
23:48:17
[modbus:174]
Got Modbus frame from unknown address 0x03!
23:48:17
[modbus_controller:110]
Received read holding/input registers for device 0x4. FC: 0x4. Start address: 0x15. Number of registers: 0x1.
23:48:17
[modbus_controller:124]
Matched register. Address: 0x15. Value type: 1. Register count: 1. Value: 511.
23:48:17
[modbus:174]
Got Modbus frame from unknown address 0x02!
23:48:17
[modbus:174]
Got Modbus frame from unknown address 0x03!
23:48:17
[modbus:174]
Got Modbus frame from unknown address 0x02!
23:48:17
[modbus:174]
Got Modbus frame from unknown address 0x03!
23:48:17
[modbus_controller:110]
Received read holding/input registers for device 0x4. FC: 0x4. Start address: 0x15. Number of registers: 0x1.
23:48:17
[modbus_controller:124]
Matched register. Address: 0x15. Value type: 1. Register count: 1. Value: 511.
23:48:17
[modbus:174]
Got Modbus frame from unknown address 0x02!
23:48:18
[... 194 lines cut ...]
23:49:18
[V]
[modbus:360]
Clearing buffer of 7 bytes - parse succeeded 46ms after last send
23:49:18
[V]
[modbus:271]
Write: 02.04.00.15.00.01.20.3D (8) 104ms after last send
23:49:18
[V]
[modbus:084]
Received first byte 2 (0X2) 18ms after last send
23:49:18
[V]
[modbus:360]
Clearing buffer of 7 bytes - parse succeeded 44ms after last send
23:49:18
[V]
[modbus:271]
Write: 03.04.00.05.00.01.20.29 (8) 101ms after last send
23:49:18
[V]
[modbus:084]

```

exciton, 2025-10-03:
There's no need to reduce the send_wait_time - it won't help. 

You can see the client is able to get a request out about every 100-110ms, so your update interval needs to be at least 330ms (maybe 350 to be safe) to be sure they each get a turn.

k0cka1, 2025-10-04:
@exciton I ended up with 20ms turnaround_time, 400ms send_wait_time and I also had to compromise and set the update intervals to 300ms. This way it seems balanced but I thought I could do better :D Do you think there is a possibility to poll for example 10-15 devices every 300ms?