esphome/issues #6635: EAP-TLS does not work with framework arduino

Rudd-O, 2025-01-03:
### The problem

Various devices (Freenove ESP32-WROOM and M5Stack Atom) fail when using EAP-TLS and compiled with framework `arduino`, which correctly works when using framework `esp-idf`.

I no longer have logs from my router when the WROOM device was failing, but when the Atom fails, this is what my router says:

```
Fri Jan  3 13:16:43 2025 daemon.notice hostapd: wlan_cis_2g_ent: CTRL-EVENT-EAP-STARTED e8:9f:6d:06:b9:48
Fri Jan  3 13:16:43 2025 daemon.notice hostapd: wlan_cis_2g_ent: CTRL-EVENT-EAP-PROPOSED-METHOD vendor=0 method=1
Fri Jan  3 13:16:43 2025 daemon.notice hostapd: wlan_cis_2g_ent: CTRL-EVENT-EAP-STARTED e8:9f:6d:06:b9:48
Fri Jan  3 13:16:43 2025 daemon.notice hostapd: wlan_cis_2g_ent: CTRL-EVENT-EAP-PROPOSED-METHOD vendor=0 method=1
Fri Jan  3 13:16:44 2025 daemon.notice hostapd: wlan_cis_2g_ent: CTRL-EVENT-EAP-FAILURE2 e8:9f:6d:06:b9:48
Fri Jan  3 13:16:44 2025 daemon.warn hostapd: wlan_cis_2g_ent: STA e8:9f:6d:06:b9:48 IEEE 802.1X: authentication failed - EAP type: 13 (TLS)
Fri Jan  3 13:16:44 2025 daemon.info hostapd: wlan_cis_2g_ent: STA e8:9f:6d:06:b9:48 IEEE 802.1X: Supplicant used different EAP type: 3 (unknown)
Fri Jan  3 13:16:49 2025 daemon.info hostapd: wlan_cis_2g_ent: STA e8:9f:6d:06:b9:48 IEEE 802.11: deauthenticated due to local deauth request
```

Note the device is sending to the supplicant the wrong EAP type.

This should either be fixed, or documented that EAP-TLS simply doesn't work reliably with framework `arduino`

### Which version of ESPHome has the issue?

core-2024.12.2

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

Irrelevant

### What platform are you using?

ESP32

### Board

M5Stack Atom / Freenove ESP32-WROOM-S

### Component causing the issue

wifi

### YAML Config

```yaml
wifi:
  networks:
  - ssid: !secret wifi_ssid_ent
    eap:
      identity: bluetooth-proxy.dragonfear
      certificate_authority: wi-fi/ca.pem
      certificate: wi-fi/certs/bluetooth-proxy/cert.pem
      key: wi-fi/certs/bluetooth-proxy/key.pem
  - ssid: !secret wifi_recovery_ssid
    password: !secret wifi_recovery_password
```
```

### Anything in the logs that might be useful for us?

```txt
See above.
```

Rudd-O, 2025-01-03:
I confirm a different issue that prevents association using ESP8266 devices too:

Router logs:

```
Fri Jan  3 16:17:17 2025 daemon.info hostapd: wlan_cis_2g_ent: STA 34:98:7a:d9:3a:5e IEEE 802.11: authenticated
Fri Jan  3 16:17:17 2025 daemon.info hostapd: wlan_cis_2g_ent: STA 34:98:7a:d9:3a:5e IEEE 802.11: No WPA/RSN IE in association request
Fri Jan  3 16:17:18 2025 daemon.info hostapd: wlan_cis_2g_ent: STA 34:98:7a:d9:3a:5e IEEE 802.11: authenticated
Fri Jan  3 16:17:18 2025 daemon.info hostapd: wlan_cis_2g_ent: STA 34:98:7a:d9:3a:5e IEEE 802.11: No WPA/RSN IE in association request
Fri Jan  3 16:17:29 2025 daemon.info hostapd: wlan_cis_2g_ent: STA 34:98:7a:d9:3a:5e IEEE 802.11: authenticated
Fri Jan  3 16:17:29 2025 daemon.info hostapd: wlan_cis_2g_ent: STA 34:98:7a:d9:3a:5e IEEE 802.11: No WPA/RSN IE in association request
Fri Jan  3 16:17:29 2025 daemon.info hostapd: wlan_cis_2g_ent: STA 34:98:7a:d9:3a:5e IEEE 802.11: authenticated
Fri Jan  3 16:17:29 2025 daemon.info hostapd: wlan_cis_2g_ent: STA 34:98:7a:d9:3a:5e IEEE 802.11: No WPA/RSN IE in association request
Fri Jan  3 16:17:40 2025 daemon.info hostapd: wlan_cis_2g_ent: STA 34:98:7a:d9:3a:5e IEEE 802.11: authenticated
Fri Jan  3 16:17:40 2025 daemon.info hostapd: wlan_cis_2g_ent: STA 34:98:7a:d9:3a:5e IEEE 802.11: No WPA/RSN IE in association request
Fri Jan  3 16:17:40 2025 daemon.info hostapd: wlan_cis_2g_ent: STA 34:98:7a:d9:3a:5e IEEE 802.11: authenticated
Fri Jan  3 16:17:40 2025 daemon.info hostapd: wlan_cis_2g_ent: STA 34:98:7a:d9:3a:5e IEEE 802.11: No WPA/RSN IE in association request
Fri Jan  3 16:17:51 2025 daemon.info hostapd: wlan_cis_2g_ent: STA 34:98:7a:d9:3a:5e IEEE 802.11: authenticated
Fri Jan  3 16:17:51 2025 daemon.info hostapd: wlan_cis_2g_ent: STA 34:98:7a:d9:3a:5e IEEE 802.11: No WPA/RSN IE in association request
Fri Jan  3 16:17:52 2025 daemon.info hostapd: wlan_cis_2g_ent: STA 34:98:7a:d9:3a:5e IEEE 802.11: authenticated
Fri Jan  3 16:17:52 2025 daemon.info hostapd: wlan_cis_2g_ent: STA 34:98:7a:d9:3a:5e IEEE 802.11: No WPA/RSN IE in association request
```

ESP8266 logs:

```
[16:16:43][I][wifi:313]: WiFi Connecting to '(AP with WPA2-Enterprise)'...
[16:16:43][W][wifi_esp8266:513]: Event: Disconnected ssid='(AP with WPA2-Enterprise)' bssid=20:05:B6:FF:13:34 reason='Association Failed'
[16:16:43][W][wifi:653]: Error while connecting to network.
```

ESP32 devices with framework esp-idf work fine.

Rudd-O, 2025-01-07:
Issue #2128 seems to be blocking the completion of the ESP8266 part of this ticket.

Rudd-O, 2025-01-07:
Actually, my bad, the issue blocking the ESP8266 part of this is https://github.com/esphome/feature-requests/issues/2128 .

mwaibel01, 2025-05-12:
Same issue with arduino and 8266/esp32.
Works fine with esp-idf and ESP32.