esphome/issues #6636: MODBUS - Incorrectly read values ​​if there is a gap between registers

Dexter0007, 2025-01-03:
### The problem

I'm trying to read data from a Lumel nmid30-2 meter using Modbus.
When I read several registers consecutively, everything works correctly.
However, if I read several registers with a gap between them (commented out register_count: 14), I get incorrect sensor values.

### Which version of ESPHome has the issue?

2024.12.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.12.5

### What platform are you using?

ESP32

### Board

BoneIO

### Component causing the issue

modbus_controller

### YAML Config

```yaml
============================> GOOD:
uart:
    id: uart_modbus
    rx_pin: GPIO3
    tx_pin: GPIO1
    baud_rate: 9600
    stop_bits: 1

modbus:
    uart_id: uart_modbus
    id: mod_bus
    send_wait_time: 1000ms

modbus_controller:
  - id: lumel_modbus
    address: 0x0001
    modbus_id: mod_bus
    update_interval: 60s
    command_throttle: 1000ms

sensor:
  - platform: modbus_controller
    modbus_controller_id: lumel_modbus
    name: "Home Voltage L1"
    address: 0x0000
    device_class: voltage
    state_class: measurement
    unit_of_measurement: "V"
    register_type: read
    value_type: FP32
    accuracy_decimals: 2
  - platform: modbus_controller
    modbus_controller_id: lumel_modbus
    name: "Home Voltage L2"
    address: 0x0002
    device_class: voltage
    state_class: measurement
[... 281 lines cut ...]
[15:18:35][D][sensor:094]: 'Home Voltage L1': Sending state 0.54181 V with 2 decimals of accuracy
[15:18:35][D][modbus_controller.sensor:025]: Sensor new state: 0.54
[15:18:35][D][sensor:094]: 'Home Voltage L2': Sending state 0.54214 V with 2 decimals of accuracy
[15:18:35][D][modbus_controller.sensor:025]: Sensor new state: 0.76
[15:18:35][D][sensor:094]: 'Home Voltage L3': Sending state 0.76376 V with 2 decimals of accuracy
[15:18:35][E][modbus_controller:705]: not enough data for value
[15:18:35][D][modbus_controller.sensor:025]: Sensor new state: 0.00
[15:18:35][D][sensor:094]: 'Home Current L1': Sending state 0.00000 A with 2 decimals of accuracy
[15:18:35][D][modbus_controller.sensor:025]: Sensor new state: 0.01
[15:18:35][D][sensor:094]: 'Home Current L2': Sending state 0.01446 A with 2 decimals of accuracy
[15:18:35][D][modbus_controller.sensor:025]: Sensor new state: 0.00
[15:18:35][D][sensor:094]: 'Home Current L3': Sending state 0.00000 A with 2 decimals of accuracy
[15:18:35][D][modbus_controller.sensor:025]: Sensor new state: 2417851639229258349412352.00
[15:18:35][D][sensor:094]: 'Home Power L1': Sending state 2417851639229258349412352.00000 W with 2 decimals of accuracy
[15:18:35][D][modbus_controller.sensor:025]: Sensor new state: 0.00
[15:18:35][D][sensor:094]: 'Home Power L2': Sending state 0.00000 W with 2 decimals of accuracy
[15:18:35][D][modbus_controller.sensor:025]: Sensor new state: 2199023255552.00
[15:18:35][D][sensor:094]: 'Home Power L3': Sending state 2199023255552.00000 W with 2 decimals of accuracy
[15:18:36][D][modbus_controller.sensor:025]: Sensor new state: 240.23
[15:18:36][D][sensor:094]: 'Home Power Factor L1': Sending state 240.23157  with 2 decimals of accuracy
[15:18:36][D][modbus_controller.sensor:025]: Sensor new state: 240.42
[15:18:36][D][sensor:094]: 'Home Power Factor L2': Sending state 240.41922  with 2 decimals of accuracy
[15:18:36][D][modbus_controller.sensor:025]: Sensor new state: 239.65
[15:18:36][D][sensor:094]: 'Home Power Factor L3': Sending state 239.65027  with 2 decimals of accuracy
```

Dexter0007, 2025-05-05:
Not stale

nikito7, 2026-09-17:
Try with 2026.8.2