esphome/issues #6638: Platform not found: 'output.i2s_audio'.

CeyssensHelder, 2025-01-03:
### The problem

When trying to add a i2s speaker, it does not find the platform/module.

### Which version of ESPHome has the issue?

2024.12.2

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

### Board

WROOM32

### Component causing the issue

i2s_audio

### YAML Config

```yaml
i2s_audio:
  i2s_lrclk_pin: GPIO25
  i2s_bclk_pin: GPIO26

output:
  - platform: i2s_audio
    name: Speaker
    dac_type: external
    i2s_dout_pin: GPIO22
    mode: mono
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2024.12.2
INFO Reading configuration /config/test12345.yaml...
INFO Unable to import component i2s_audio.output: No module named 'esphome.components.i2s_audio.output'
Failed config

output.i2s_audio: [source /config/test12345.yaml:38]
  
  Platform not found: 'output.i2s_audio'.
  platform: i2s_audio
  name: Speaker
  dac_type: external
  i2s_dout_pin: GPIO22
  mode: mono
```

BigBobbas, 2025-01-06:
the component you need is `speaker:`  https://esphome.io/components/speaker/i2s_audio

maoawa, 2025-01-17:
the same

[image]

maoawa, 2025-01-17:
> the component you need is `speaker:` https://esphome.io/components/speaker/i2s_audio

```yaml
i2s_audio:
  i2s_lrclk_pin: GPIO27
  i2s_bclk_pin: GPIO25
  frame_size: 512
  sample_rate: 44100
  bits_per_sample: 16

speaker:
  - platform: i2s_audio
    dac_type: external
    i2s_dout_pin: GPIO26
```

I tried code like this but sadly still not working...
```
INFO ESPHome 2024.12.4
INFO Reading configuration esphome/sleepy-helper.yaml...
INFO Unable to import component i2s_audio.speaker: No module named 'esphome.components.i2s_audio.speaker'
Failed config

speaker.i2s_audio: [source esphome/sleepy-helper.yaml:122]
  
  Platform not found: 'speaker.i2s_audio'.
  platform: i2s_audio
  dac_type: external
  i2s_dout_pin: GPIO26
```

i2s_audio seems to be a built-in module of ESPHome, but I don't know why there's an error `INFO Unable to import component i2s_audio.speaker: No module named 'esphome.components.i2s_audio.speaker'`.
Tried reinstalling esphome by many package manager like homebrew, pip, etc. But that helped nothing...

maoawa, 2025-01-17:
> the component you need is `speaker:` https://esphome.io/components/speaker/i2s_audio

Solved this issue by using the docker version of esphome instead of pip or homebrew
Thanks for your kind answering ❤

maoawa, 2025-01-17:
Solved this issue by using the docker version of esphome instead of pip or homebrew!
Hope this could work for you ❤