esphome/issues #6639: ESP32-C6 cannot compile with I2C

PSMGoossens, 2025-01-03:
### The problem

When compiling a scatch of ESP32C6 with I2C support, we get a build error that `I2C_NUM_1 ` is not defined. This is caused by the introduction of Low Power and normal I2C interfaces

See ESP-IDF define [here](https://github.com/espressif/esp-idf/blob/b5ac4fbdf9e9fb320bb0a98ee4fbaa18f8566f37/components/soc/esp32c6/include/soc/soc_caps.h#L241)

Could the code be fixed that compiling for C6 with I2C will work?/

### Which version of ESPHome has the issue?

ESPHome 2024.12.2

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2024.12

### What platform are you using?

ESP32-IDF

### Board

ESP32 -c6 devkit

### Component causing the issue

i2c

### YAML Config

```yaml
esphome:
  name: esp-airquality
  friendly_name: ESP-AirQuality

esp32:
  board: esp32-c6-devkitm-1
  variant: esp32c6
  framework:
    type: esp-idf
    platform_version: 6.9.0
    version: 5.3.1
    
    sdkconfig_options:
      CONFIG_ESPTOOLPY_FLASHSIZE_4MB: y

# Enable logging
logger:
  level: DEBUG

# Enable Home Assistant API
api:
  encryption:
    key: "8iXBbbHx0q/pJ2M6vluoet2YYJt08Aisku5o7985xwE="

ota:
  - platform: esphome
    password: "cc6e98f638c7ea1d1ac2827b0fb20aac"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp-Airquality Fallback Hotspot"
    password: "oBrE5snHIMtw"

captive_portal:
    
switch:
   - platform: gpio
     name: "on board LED"
     pin: GPIO15 

i2c:
  sda: GPIO6
  scl: GPIO7
  scan: true
  id: bus_a  

# Example configuration entry
sensor:
  - platform: scd4x
    co2:
      name: "CO2"
    temperature:
      name: "Temperature"
    humidity:
      name: "Humidity"
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2024.12.2
INFO Reading configuration config/esp-airquality.yaml...
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING GPIO15 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Generating C++ source...
INFO Updating https://github.com/espressif/esp-protocols.git@mdns-v1.3.2
INFO Compiling app...
Processing esp-airquality (board: esp32-c6-devkitm-1; framework: espidf; platform: platformio/espressif32@6.9.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32C6 160MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.50301.0 (5.3.1) 
 - tool-cmake @ 3.16.4 
 - tool-ninja @ 1.7.1 
 - tool-riscv32-esp-elf-gdb @ 12.1.0+20221002 
 - tool-xtensa-esp-elf-gdb @ 12.1.0+20221002 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-riscv32-esp @ 13.2.0+20240530
Reading CMake configuration...
Dependency Graph
|-- noise-c @ 0.1.6
Compiling .pioenvs/esp-airquality/src/esphome/components/i2c/i2c_bus_esp_idf.cpp.o
Compiling .pioenvs/esp-airquality/src/esphome/components/mdns/mdns_component.cpp.o
Compiling .pioenvs/esp-airquality/src/esphome/components/mdns/mdns_esp32.cpp.o
Compiling .pioenvs/esp-airquality/src/esphome/components/mdns/mdns_esp8266.cpp.o
Compiling .pioenvs/esp-airquality/src/esphome/components/mdns/mdns_host.cpp.o
Compiling .pioenvs/esp-airquality/src/esphome/components/mdns/mdns_libretiny.cpp.o
Compiling .pioenvs/esp-airquality/src/esphome/components/mdns/mdns_rp2040.cpp.o
Compiling .pioenvs/esp-airquality/src/esphome/components/network/util.cpp.o
src/esphome/components/i2c/i2c_bus_esp_idf.cpp: In member function 'virtual void esphome::i2c::IDFI2CBus::setup()':
src/esphome/components/i2c/i2c_bus_esp_idf.cpp:21:42: error: 'I2C_NUM_1' was not declared in this scope; did you mean 'I2C_NUM_0'?
   21 |   next_port = (next_port == I2C_NUM_0) ? I2C_NUM_1 : I2C_NUM_MAX;
      |                                          ^~~~~~~~~
      |                                          I2C_NUM_0
Compiling .pioenvs/esp-airquality/src/esphome/components/ota/ota_backend.cpp.o
*** [.pioenvs/esp-airquality/src/esphome/components/i2c/i2c_bus_esp_idf.cpp.o] Error 1
========================== [FAILED] Took 8.12 seconds ==========================
```

### Additional information

It seems that ESP32-C6 is handling Low Power I2C differently with other defines.

PSMGoossens, 2025-01-03:
Workaround:
Adding this as a build flag to platformio:
`build_flags: "-DI2C_NUM_1=I2C_NUM_0"`

casse-boubou, 2025-01-05:
> Workaround: Adding this as a build flag to platformio: `build_flags: "-DI2C_NUM_1=I2C_NUM_0"`

Hi,
I have the same error, only after adding the build_flags as you said, my esp32-c6 no longer starts. I have to reflash the old firmware via USB. Could you provide the new YAML? I would like to check if I didn't make an intent error for build_flags.
THANKS

PSMGoossens, 2025-01-05:
@casse-boubou You also need to explicitly disable scanning for I2C:
``
i2c:
  scan: false
```

casse-boubou, 2025-01-05:
Yes that was it!! It works now! THANK YOU !!!!

ximex, 2025-02-19:
this PR will fix the first part of the problem: https://github.com/esphome/esphome/pull/8283
the `scan: false` is currently still necessary
```yaml
external_components:
  - source: github://pr#8283
    components: [ i2c ]
```

ximex, 2025-02-25:
maybe someone else could review and test the PR and approve if all is ok or add change requests

swoboda1337, 2025-02-26:
> Edit: got already merged now we need to debug whats the problem with `scan: true`

```
[C][i2c.idf:021]: Setting up I2C bus...
[I][i2c.idf:262]: Performing I2C bus recovery
[V][i2c.idf:078]: Scanning i2c bus for active devices...
[VV][i2c.idf:205]: 0x08 TX
[VV][i2c.idf:246]: TX to 08 failed: not acked
[VV][i2c.idf:205]: 0x09 TX
[VV][i2c.idf:246]: TX to 09 failed: not acked
[VV][i2c.idf:205]: 0x0A TX
[VV][i2c.idf:246]: TX to 0A failed: not acked
[VV][i2c.idf:205]: 0x0B TX
[VV][i2c.idf:246]: TX to 0B failed: not acked
[VV][i2c.idf:205]: 0x0C TX
[VV][i2c.idf:246]: TX to 0C failed: not acked
[VV][i2c.idf:205]: 0x0D TX
[VV][i2c.idf:246]: TX to 0D failed: not acked
[VV][i2c.idf:205]: 0x0E TX
[VV][i2c.idf:246]: TX to 0E failed: not acked
[VV][i2c.idf:205]: 0x0F TX
[VV][i2c.idf:246]: TX to 0F failed: not acked
[VV][i2c.idf:205]: 0x10 TX
[VV][i2c.idf:246]: TX to 10 failed: not acked
[VV][i2c.idf:205]: 0x11 TX
E (5762) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (5762) task_wdt:  - loopTask (CPU 0)
E (5762) task_wdt: Tasks currently running:
E (5762) task_wdt: CPU 0: loopTask
E (5762) task_wdt: Aborting.
```

It scans the first few really fast then gets stuck here:

https://github.com/espressif/esp-idf/blob/release/v5.3/components/driver/i2c/i2c.c#L688

swoboda1337, 2025-02-26:
This issue seems relevant: https://github.com/espressif/esp-idf/issues/13647
Seems there was a fix put into 5.3.2

swoboda1337, 2025-02-26:
Looks like the fix was to the new i2c not the legacy one. Will have to move to the new one and to 5.3.2 or later to fix scan.

randybb, 2025-02-26:
5.3.2 is not supported by platformio. maybe it can be overridden or we need to wait for https://github.com/esphome/esphome/pull/8106

swoboda1337, 2025-02-26:
Looked at the new i2c driver and its not great. Scan would be forced to run at 100khz and miss many devices because of that. Scan wouldn't work in the new driver anyways. 

https://github.com/espressif/esp-idf/issues/13395

trianglesis, 2025-03-19:
Using this board `esp32-c6-devkitc-1`:

I'm still unable to set the CO2 sensor to work,
The compilation is successful btw.
Logs in debug, and no warnings or issues, just NA as values.

<details>
  <summary>YAML config</summary>
  
```yaml
esphome:
  name: "esp32-c6-1"
  friendly_name: "ESP32_C6_1"
  # https://github.com/esphome/esphome/pull/1651
  platformio_options:
    build_flags: "-DI2C_NUM_1=I2C_NUM_0"

esp32:
  board: esp32-c6-devkitc-1
  flash_size: 8MB
  variant: esp32c6
  framework:
    type: esp-idf
    sdkconfig_options:
      CONFIG_ESPTOOLPY_FLASHSIZE_8MB: y
    version: "5.3.1"
    platform_version: 6.9.0

external_components:
  - source: github://pr#8283
    components: [ i2c ]

i2c:
  sda: GPIO06
  scl: GPIO07
  scan: false
  id: bus_a

sensor:
  - platform: scd4x
    i2c_id: bus_a
    address: 0x62
    id: scd41_esp32
    co2:
      name: "CO2"
    temperature:
      name: "Temperature"
    humidity:
      name: "Humidity"
```
  
</details>

<details>
  <summary>ESP dev board img</summary>
  
[image]
  
</details>

swoboda1337, 2025-03-20:
> 5.3.2 is not supported by platformio. maybe it can be overridden or we need to wait for [esphome/esphome#8106](https://github.com/esphome/esphome/pull/8106)

Turns out we can use 5.3.2 and even 5.4.0 through pioarduino. I created a PR: https://github.com/esphome/esphome/pull/8440
```
external_components:    
  - source: github://pr#8440
    components: [ esp32 ]
```
Can try scan with 5.3.2 or 5.4.0:
```
  framework:
    type: esp-idf
    version: 5.3.2
    platform_version: 53.03.11
```
```
  framework:    
    type: esp-idf
    version: 5.4.0
    platform_version: 54.03.20-rc2
```

ximex, 2025-03-22:
It's running with `5.3.2` but i still have to set `scan: false`
`5.4.0` variant is not compiling

swoboda1337, 2025-03-23:
Right, 

On 5.3.2 they only fixed the new i2c driver you can see a timeout check: https://github.com/espressif/esp-idf/blob/release/v5.3/components/esp_driver_i2c/i2c_master.c#L94

In the legacy driver they just moved the while: https://github.com/espressif/esp-idf/blob/master/components/driver/i2c/i2c.c#L688

Probably need to submit a PR to espressif and fix it ourselves. A PR to the new driver to allow setting the clock speed would be good too.

swoboda1337, 2025-03-23:
Ok I patched it and scan works now but you have to use my custom release:
```
esp32:
  board: esp32-c6-devkitc-1
  framework:
    type: esp-idf
    version: 5.3.2
    platform_version: 53.03.11
    source: "pioarduino/framework-espidf@https://github.com/swoboda1337/esp-idf/releases/download/5.3.2/esp-idf-v5.3.2.zip"

external_components:    
  - source: github://pr#8440
    components: [ esp32 ]
```

swoboda1337, 2025-03-29:
Today pioadruino released 5.4.1. I noticed the new i2c driver had new functionality that made it easy to integrate into ESPHome. I was able to integrate it. I have done limited testing on two i2c devices but it seems good. Everything should work including scan and even the second i2c port on c6 (I haven't tested this though). 

```
esp32:
  framework:
    type: esp-idf
    version: 5.4.1
    platform_version: 54.03.20

external_components:    
  - source: github://pr#8483
    components: [ i2c ]
  - source: github://pr#8480
    components: [ esp32 ]
```

If others want to test this PR it would be appreciated.

trianglesis, 2025-04-01:
> ```
> esp32:
>   framework:
>     type: esp-idf
>     version: 5.4.1
>     platform_version: 54.03.20
> 
> external_components:    
>   - source: github://pr#8483
>     components: [ i2c ]
>   - source: github://pr#8480
>     components: [ esp32 ]
> ```

Unfortunately I got this:

```log
src/esphome/core/helpers.cpp:47:10: fatal error: esp32/rom/crc.h: No such file or directory
   47 | #include "esp32/rom/crc.h"
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/esp32-c6-1/src/esphome/core/helpers.cpp.o] Error 1
========================= [FAILED] Took 44.31 seconds =========================
```

Updated ESPHome to latest from pip:

`Successfully installed aioesphomeapi-29.6.0 cryptography-44.0.2 esphome-2025.3.3 esptool-4.8.1 tornado-6.4.2 zeroconf-0.146.1`

ximex, 2025-05-31:
@trianglesis now it should already work without PR 8480 (already in `2025.5.1`) pls retest

Zipponia, 2025-06-07:
Hi, I see the many sketches around GitHub to get the Xiao Esp32 C6 working won't compile anymore.

What is the latest config to successfully use the Xiao esp32 C6 and with I2C ?

which version of esphome should I use ?

Zipponia, 2025-06-08:
> Ok I patched it and scan works on c6 now but you have to use my custom/modified release:
> 
> ```
> esp32:
>   board: esp32-c6-devkitc-1
>   framework:
>     type: esp-idf
>     version: 5.3.2
>     platform_version: 53.03.11
>     source: "pioarduino/framework-espidf@https://github.com/swoboda1337/esp-idf/releases/download/5.3.2/esp-idf-v5.3.2.zip"
> 
> external_components:    
>   - source: github://pr#8440
>     components: [ esp32 ]
> ```
> 
> ```
> [C][i2c.idf:083]: I2C Bus:
> [C][i2c.idf:084]:   SDA Pin: GPIO21
> [C][i2c.idf:085]:   SCL Pin: GPIO22
> [C][i2c.idf:086]:   Frequency: 50000 Hz
> [C][i2c.idf:092]:   Recovery: bus successfully recovered
> [I][i2c.idf:102]: Results from i2c bus scan:
> [I][i2c.idf:104]: Found no i2c devices!
> ```
> 
> PR is here: [espressif/esp-idf#15638](https://github.com/espressif/esp-idf/pull/15638)

This version compiles

however if I test 

`esp32:
  board: esp32-c6-devkitc-1
  framework:
    type: esp-idf
    version: 5.4.1
    platform_version: 54.03.20

external_components:    
  - source: github://pr#8483
    components: [ i2c ]`

this one 
I get 

> -- Found Git: /usr/bin/git (found version "2.39.5")
-- git rev-parse returned 'fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).'
-- The C compiler identification is GNU 14.2.0
-- The CXX compiler identification is GNU 14.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /data/cache/platformio/packages/toolchain-riscv32-esp/bin/riscv32-esp-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /data/cache/platformio/packages/toolchain-riscv32-esp/bin/riscv32-esp-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /data/cache/platformio/packages/toolchain-riscv32-esp/bin/riscv32-esp-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32c6
NOTICE: Updating lock file at /data/build/ambrogio/dependencies.lock
Processing 1 dependencies:
[1/1] idf (5.4.1)
-- Project sdkconfig file /data/build/ambrogio/sdkconfig.ambrogio
-- Configuring incomplete, errors occurred!

fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/root/.platformio/penv/.espidf-5.4.1/lib/python3.12/site-packages/kconfgen/__main__.py", line 5, in <module>
    from .core import main
  File "/root/.platformio/penv/.espidf-5.4.1/lib/python3.12/site-packages/kconfgen/core.py", line 29, in <module>
    import esp_idf_kconfig.gen_kconfig_doc as gen_kconfig_doc
  File "/root/.platformio/penv/.espidf-5.4.1/lib/python3.12/site-packages/esp_idf_kconfig/gen_kconfig_doc.py", line 14, in <module>
    import kconfiglib.core as kconfiglib
  File "/root/.platformio/penv/.espidf-5.4.1/lib/python3.12/site-packages/kconfiglib/__init__.py", line 3, in <module>
    from .core import Kconfig  # noqa F401
    ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/.platformio/penv/.espidf-5.4.1/lib/python3.12/site-packages/kconfiglib/core.py", line 27, in <module>
    from kconfiglib.report import PRAGMA_PREFIX
  File "/root/.platformio/penv/.espidf-5.4.1/lib/python3.12/site-packages/kconfiglib/report.py", line 29, in <module>
    from rich import print as rprint
ModuleNotFoundError: No module named 'rich'
CMake Error at /data/cache/platformio/packages/framework-espidf/tools/cmake/kconfig.cmake:209 (message):
  Failed to run kconfgen
  (/root/.platformio/penv/.espidf-5.4.1/bin/python;-m;kconfgen;--list-separator=semicolon;--kconfig;/data/cache/platformio/packages/framework-espidf/Kconfig;--sdkconfig-rename;/data/cache/platformio/packages/framework-espidf/sdkconfig.rename;--config;/data/build/ambrogio/sdkconfig.ambrogio;--env-file;/data/build/ambrogio/.pioenvs/ambrogio/config.env).
  Error 1
Call Stack (most recent call first):
  /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:681 (__kconfig_generate_config)
  /data/cache/platformio/packages/framework-espidf/tools/cmake/project.cmake:717 (idf_build_process)
  CMakeLists.txt:3 (project)

swoboda1337, 2025-06-08:
> however if I test
> 
> `esp32: board: esp32-c6-devkitc-1 framework: type: esp-idf version: 5.4.1 platform_version: 54.03.20
> 

Yes they broke `54.03.20` you can try either of these:
```
esp32:
  framework:
    type: esp-idf
    version: 5.4.1
    platform_version: "https://github.com/pioarduino/platform-espressif32.git#develop"
```
```
esp32:
  framework:
    type: esp-idf
    version: 5.5.0
    release: "20250510"
    platform_version: "https://github.com/pioarduino/platform-espressif32.git#develop"
```
might have to clean your platformio folder

Zipponia, 2025-06-11:
Thank you
it did compile yesterday but didn't find i2c devices. Today I discovered the port 6-7 is the secondary one while the primary one is 4-5. Wanted to try it but I now get an error

`In file included from src/esphome/components/sensor/sensor.h:3,
                 from src/esphome/core/application.h:20,
                 from src/esphome/core/application.cpp:1:
src/esphome/core/application.cpp: In member function 'void esphome::Application::teardown_components(uint32_t)':
src/esphome/core/application.cpp:222:21: error: format '%u' expects argument of type 'unsigned int', but argument 6 has type 'uint32_t' {aka 'long unsigned int'} [-Werror=format=]
  222 |       ESP_LOGW(TAG, "%s did not complete teardown within %u ms", component->get_component_source(), timeout_ms);
      |                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/esphome/core/log.h:76:36: note: in definition of macro 'ESPHOME_LOG_FORMAT'
   76 | #define ESPHOME_LOG_FORMAT(format) format
      |                                    ^~~~~~
src/esphome/core/log.h:154:28: note: in expansion of macro 'esph_log_w'
  154 | #define ESP_LOGW(tag, ...) esph_log_w(tag, __VA_ARGS__)
      |                            ^~~~~~~~~~
src/esphome/core/application.cpp:222:7: note: in expansion of macro 'ESP_LOGW'
  222 |       ESP_LOGW(TAG, "%s did not complete teardown within %u ms", component->get_component_source(), timeout_ms);
      |       ^~~~~~~~
src/esphome/core/application.cpp:222:59: note: format string is defined here
  222 |       ESP_LOGW(TAG, "%s did not complete teardown within %u ms", component->get_component_source(), timeout_ms);
      |                                                          ~^
      |                                                           |
      |                                                           unsigned int
      |                                                          %lu
Compiling .pioenvs/frontalino-primo/src/esphome/core/component_iterator.cpp.o
Compiling .pioenvs/frontalino-primo/src/esphome/core/controller.cpp.o
cc1plus: some warnings being treated as errors
*** [.pioenvs/frontalino-primo/src/esphome/core/application.cpp.o] Error 1
========================= [FAILED] Took 161.89 seconds =========================`

did the mistake to upgrade to latest dev.

swoboda1337, 2025-06-11:
> Thank you it did compile yesterday but didn't find i2c devices. Today I discovered the port 6-7 is the secondary one while the primary one is 4-5. Wanted to try it but I now get an error
>
> `In file [...]

That was fixed this morning: https://github.com/esphome/esphome/pull/9045

Zipponia, 2025-06-11:
Ok

Does the updated add on push every 24h because it is updated with a script ?