esphome/issues #6643: Esp32 connecting to unconfigured open wifi, not mine

LucaFulchir, 2025-01-04:
### The problem

I flashed esphome on a Shelly 2PM, configured to connect to my network.
Not the first time I flash and use shellys like this, but this time I'm not at home.

It doesn't seem connect, and the fallback ap does not come up.

Connecting it to the computer shows that esphome is instead connecting directly to an open wifi nearby that is not under my control.
I have no "failed attempt" towards my AP, it seems to try the open wifi first, that obviously works, and remains there.

After reading #3508 I tried both the single-wifi configuration and the multiple-network wifi config, with priority.

Does not work. Esphome still connects to the open wifi.

Connecting to unconfigured open wifi seems like a security problem, not merely a usability problem, but I can't find how to disable this behavior.

I also have not found any workaround of any kind.

### Which version of ESPHome has the issue?

2024.10.3

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Board

Shelly 2PM

### Component causing the issue

wifi

### YAML Config

```yaml
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_pwd
  ap:
    ssid: "Shelly-fallback"
    password: !secret wifi_fallback_pwd
    ap_imeout: 1min
  reboot_timeout: 5min
```

clydebarrow, 2025-01-05:
Assuming the open network has a different SSID to yours, that's odd, but setting `fast_connect: true` should work around the issue.

clydebarrow, 2025-01-06:
I tested this and could't reproduce the issue. Attach some logs.

LucaFulchir, 2025-01-08:
updated esphome:
```
$ esphome --version
Version: 2024.12.2
```

added an open wifi `OpenWrt`, channel 1, 20mhz

removed a lot of stuff from my yaml, currently only testing wifi:
```
substitutions:
  devicename: "connect"
  upper_devicename: "Shelly Plus 2PM"

# For PCB v0.1.9 with dual core ESP32
esphome:
  name: "${devicename}"

esp32:
  board: esp32doit-devkit-v1
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  platform: esphome
  password: !secret ota_password

wifi:
  ssid: "NotEventPresent"
  password: "WrongPassword"
  ap:
    ssid: "ESPHome-Default"
    password: "SomePwdHere"
    ap_timeout: 1min
  reboot_timeout: 5min

#captive_portal:

status_led:
  pin:
    number: GPIO0
    inverted: true

web_server:
  port: 80
  local: true
  include_internal: true
  version: 3
  auth:
    username: "AdminUser"
```

getting logs with: `minicom -D /dev/ttyUSB0 -C dump`

result:
```
[D][wifi:497]: Found networks:
[I][wifi:541]: - 'OpenWrt' (0A:22:33:44:AA:C0) ▂���
[D][wifi:542]:     Channel: 1
[D][wifi:543]:     RSSI: -23 dB
[D][wifi:546]: - 'SB7A' (02:22:33:44:AA:C0) ����
[D][wifi:546]: - 'SB7A-IoT' (00:22:33:44:AA:C0) ����
[D][wifi:546]: - 'SB7A-Guest' (06:22:33:44:AA:C0) ����
[D][wifi:546]: - 'DIRECT-ML-FireTV_f10d' (76:A7:EA:EB:7A:05) ����
[D][wifi:546]: - 'FRITZ!Powerline 540E' (50:E6:36:EE:3F:62) ����
[I][wifi:313]: WiFi Connecting to 'OpenWrt'...
[I][wifi:617]: WiFi Connected!
[C][wifi:428]:   Local MAC: D4:8A:FC:78:50:B4
[C][wifi:433]:   SSID: 'OpenWrt'
[C][wifi:436]:   IP Address: 192.168.2.156
[C][wifi:440]:   BSSID: 0A:22:33:44:AA:C0
[C][wifi:441]:   Hostname: 'connect'
[C][wifi:443]:   Signal strength: -26 dB ����
[C][wifi:447]:   Channel: 1
[C][wifi:448]:   Subnet: 255.255.255.0
[C][wifi:449]:   Gateway: 192.168.2.1
[C][wifi:450]:   DNS1: 192.168.2.1
[C][wifi:451]:   DNS2: 0.0.0.0
[D][wifi:626]: Disabling AP...
[C][web_server:100]: Setting up web server...
[C][api:026]: Setting up Home Assistant API server...
[I][app:062]: setup() finished successfully!
[W][component:170]: Component wifi cleared Warning flag
[W][component:157]: Component api set Warning flag: unspecified
[I][app:100]: ESPHome version 2024.12.2 compiled on Jan  1 1980, 00:00:00
[C][status_led:019]: Status LED:
[C][status_led:020]:   Pin: GPIO0
[C][wifi:600]: WiFi:
[C][wifi:428]:   Local MAC: D4:8A:FC:78:50:B4
[C][wifi:433]:   SSID: 'OpenWrt'
[C][wifi:436]:   IP Address: 192.168.2.156
[C][wifi:440]:   BSSID: 0A:22:33:44:AA:C0
[C][wifi:441]:   Hostname: 'connect'
[C][wifi:443]:   Signal strength: -25 dB ����
[C][wifi:447]:   Channel: 1
[C][wifi:448]:   Subnet: 255.255.255.0
[C][wifi:449]:   Gateway: 192.168.2.1
[C][wifi:450]:   DNS1: 192.168.2.1
[C][wifi:451]:   DNS2: 0.0.0.0
[C][logger:185]: Logger:
[C][logger:186]:   Level: DEBUG
[C][logger:188]:   Log Baud Rate: 115200
[C][logger:189]:   Hardware UART: UART0
[C][web_server:153]: Web Server:
[C][web_server:154]:   Address: connect.local:80
[C][mdns:116]: mDNS:
[C][mdns:117]:   Hostname: connect
[C][esphome.ota:073]: Over-The-Air updates:
[C][esphome.ota:074]:   Address: connect.local:3232
[C][esphome.ota:075]:   Version: 2
[C][esphome.ota:078]:   Password configured
[C][safe_mode:018]: Safe Mode:
[C][safe_mode:020]:   Boot considered successful after 60 seconds
[C][safe_mode:021]:   Invoke after 10 boot attempts
[C][safe_mode:023]:   Remain in safe mode for 300 seconds
[C][api:140]: API Server:
[C][api:141]:   Address: connect.local:6053
[C][api:145]:   Using noise encryption: NO
```

LucaFulchir, 2025-01-20:
Is there any extra info I can give, or do to help in debugging this?