esphome/issues #6651: Deep Sleep: Missing support for ESP_EXT1_WAKEUP_ANY_LOW mode for ESP32 variants

studiobts, 2025-01-06:
### The problem

Looking into the documentation of Sleep Modes, for ESP32-S2, ESP32-S3, ESP32-C6 or ESP32-H2 chips, there is are different level mode reported:

`
level_mode -- Select logic function used to determine wakeup condition: 

When target chip is ESP32:

- ESP_EXT1_WAKEUP_ALL_LOW: wake up when all selected GPIOs are low
- ESP_EXT1_WAKEUP_ANY_HIGH: wake up when any of the selected GPIOs is high 

When target chip is ESP32-S2, ESP32-S3, ESP32-C6 or ESP32-H2:

- ESP_EXT1_WAKEUP_ANY_LOW: wake up when any of the selected GPIOs is low
- ESP_EXT1_WAKEUP_ANY_HIGH: wake up when any of the selected GPIOs is high
`

In particular, ALL_LOW is not available, instead an ANY_LOW can be used, which is not available through ESPHome, giving me an error.

### Which version of ESPHome has the issue?

2024.*

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### YAML Config

```yaml
deep_sleep:
  id: dp
  sleep_duration: 7d
  wakeup_pin_mode: KEEP_AWAKE
  esp32_ext1_wakeup: 
    mode: ANY_LOW
    pins:
      - 13
      - 14
```

### Anything in the logs that might be useful for us?

```txt
Failed config

deep_sleep: [source /config/esphome/config.yaml:65]
  id: dp
  sleep_duration: 7d
  wakeup_pin_mode: KEEP_AWAKE
  esp32_ext1_wakeup: 
    
    Unknown value 'ANY_LOW', did you mean 'ALL_LOW'?.
    mode: ANY_LOW
    pins: 
      - 13
      - 14
```

studiobts, 2025-05-07:
The issue persists.

alainvdu69, 2025-06-09:
I confirm.
Usage of ALL_LOW issue a warning saying "deprecated ... use ANY_LOW" ... and using ANY_LOW issue a compilation failed.

closed 2025-07-08 by jesserockz