esphome/issues #6656: i2s_audio speaker stops plaining audio after 2024.12 ?

s00500, 2025-01-07:
### The problem

I am using an onju voice board with the config shown below. on 2024.10 and .11 it all works fine, on 2024.12 I do not get any audio output at all... what am I doing wrong ?

### Which version of ESPHome has the issue?

2024.12

### What type of installation are you using?

pip

### What platform are you using?

ESP32-IDF

### Board

Onju Voice

### Component causing the issue

i2s_audio

### YAML Config

```yaml
esphome:
  name: "${name}"
  friendly_name: "${friendly_name}"
  comment: "${device_description}"
  name_add_mac_suffix: false
  project:
    name: tetele.onju_voice_satellite
    version: "${project_version}"
  min_version: 2024.7.3
  platformio_options:
    board_build.flash_mode: dio
    board_build.arduino.memory_type: qio_opi
  on_boot:
    then:
      - light.turn_on:
          id: top_led
          effect: slow_pulse
          red: 100%
          green: 60%
          blue: 0%
      - wait_until:
          condition:
            wifi.connected:
      - light.turn_on:
          id: top_led
          effect: pulse
          red: 0%
          green: 100%
          blue: 0%
      - wait_until:
          condition:
            api.connected:
      - light.turn_on:
          id: top_led
          effect: none
          red: 0%
          green: 100%
          blue: 0%
      - delay: 1s
      - script.execute: reset_led

esp32:
  board: esp32-s3-devkitc-1
[... 447 lines cut ...]
            case 2:
              id(volume_up).set_threshold(newthresh);
              break;
            default:
              ESP_LOGE("touch_calibration", "Invalid button ID (%d)", button);
              return;
          }

switch:
  - platform: template
    name: Use Wake Word
    id: use_wake_word
    optimistic: true
    restore_mode: RESTORE_DEFAULT_ON
    on_turn_on:
      - script.execute: turn_on_wake_word
    on_turn_off:
      - script.execute: turn_off_wake_word
  - platform: gpio
    id: dac_mute
    restore_mode: ALWAYS_OFF
    pin:
      number: GPIO21
      inverted: True
```

fabio-garavini, 2025-01-20:
In ESPHome 2024.12 changed the default esp-idf version, you can try to roll it back until it's fixed.
You can find how to [here](https://github.com/esphome/issues/issues/6582#issuecomment-2603043256)

s00500, 2025-01-21:
Hey fabio, thanks for the tip and the link to the other issue
Funny I get the issue the other way around though, so no output, while people on the other thread struggle with no working microphone...

s00500, 2025-02-19:
Just adding in here: this did not make it work...

```
  framework:
    type: esp-idf
    version: 4.4.8
    platform_version: 5.4.0
```

I still do not get oaudio out with this config, while it works fine on 2024.11

s00500, 2025-02-19:
Any tipps on doing bisect on esphome ?

the issue certainly is introduced between 2024.11.3 (still working) and 2024.12.0b1 (non working)

fabio-garavini, 2025-03-05:
That's odd, do you have any error log?

s00500, 2025-03-05:
Well, not a lot of errors, simply no audio, here is a run through:

```
❯ esphome logs nabu2.yml
INFO ESPHome 2024.12.0b1
INFO Reading configuration nabu2.yml...
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
Found multiple options for logging, please choose one:
  [1] /dev/cu.usbserial-1340 (USB Serial)
  [2] Over The Air (onju-voice-2.local)
(number): 2
INFO Starting log output from onju-voice-2.local using esphome API
INFO Successfully connected to onju-voice-2 @ 192.168.1.239 in 0.368s
INFO Successful handshake with onju-voice-2 @ 192.168.1.239 in 0.072s
[12:56:08][I][app:100]: ESPHome version 2024.12.0b1 compiled on Feb 19 2025, 23:29:27
[12:56:08][I][app:102]: Project tetele.onju_voice_satellite version 1.0.0
[12:56:08][C][wifi:600]: WiFi:
[12:56:08][C][wifi:441]:   Hostname: 'onju-voice-2'
[12:56:08][C][wifi:443]:   Signal strength: -60 dB ▂▄▆█
[12:56:08][C][wifi:447]:   Channel: 11
[12:56:08][C][wifi:448]:   Subnet: 255.255.254.0
[12:56:08][C][wifi:449]:   Gateway: 192.168.0.254
[12:56:08][C][wifi:450]:   DNS1: 192.168.0.245
[12:56:08][C][wifi:451]:   DNS2: 0.0.0.0
[12:56:08][C][logger:185]: Logger:
[12:56:08][C][logger:186]:   Level: DEBUG
[12:56:08][C][logger:188]:   Log Baud Rate: 115200
[12:56:08][C][logger:189]:   Hardware UART: USB_SERIAL_JTAG
[12:56:08][C][template.number:050]: Template Number 'Touch threshold percentage'
[12:56:08][C][template.number:051]:   Optimistic: YES
[12:56:08][C][template.number:052]:   Update Interval: never
[12:56:08][C][esp32_rmt_led_strip:187]: ESP32 RMT LED Strip:
[12:56:08][C][esp32_rmt_led_strip:188]:   Pin: 11
[12:56:08][C][esp32_rmt_led_strip:189]:   Channel: 0
[12:56:08][C][esp32_rmt_led_strip:214]:   RGB Order: GRB
[12:56:08][C][esp32_rmt_led_strip:215]:   Max refresh rate: 0
[12:56:08][C][esp32_rmt_led_strip:216]:   Number of LEDs: 6
[12:56:08][C][switch.gpio:068]: GPIO Switch 'dac_mute'
[12:56:08][C][switch.gpio:091]:   Restore Mode: always OFF
[12:56:08][C][switch.gpio:031]:   Pin: GPIO21
[12:56:08][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Disable wake word'
[12:56:08][C][gpio.binary_sensor:016]:   Pin: GPIO38
[12:56:08][C][light:092]: Light 'leds'
[12:56:08][C][light:094]:   Default Transition Length: 0.0s
[12:56:08][C][light:095]:   Gamma Correct: 2.80
[12:56:08][C][light:092]: Light 'left_led'
[12:56:08][C][light:094]:   Default Transition Length: 0.1s
[12:56:08][C][light:095]:   Gamma Correct: 2.80
[12:56:08][C][light:092]: Light 'top_led'
[12:56:08][C][light:094]:   Default Transition Length: 0.1s
[12:56:08][C][light:095]:   Gamma Correct: 2.80
[12:56:08][C][light:092]: Light 'right_led'
[12:56:08][C][light:094]:   Default Transition Length: 0.1s
[12:56:08][C][light:095]:   Gamma Correct: 2.80
[12:56:08][C][template.switch:068]: Template Switch 'Use Wake Word'
[12:56:08][C][template.switch:091]:   Restore Mode: restore defaults to ON
[12:56:08][C][template.switch:057]:   Optimistic: YES
[12:56:08][C][psram:020]: PSRAM:
[12:56:08][C][psram:021]:   Available: YES
[12:56:08][C][psram:024]:   Size: 8191 KB
[12:56:08][C][esp32_touch:073]: Config for ESP32 Touch Hub:
[12:56:08][C][esp32_touch:074]:   Meas cycle: 0.80ms
[12:56:08][C][esp32_touch:075]:   Sleep cycle: 2.00ms
[12:56:08][C][esp32_touch:095]:   Low Voltage Reference: 0.8V
[12:56:08][C][esp32_touch:115]:   High Voltage Reference: 2.4V
[12:56:08][C][esp32_touch:135]:   Voltage Attenuation: 0V
[12:56:08][C][esp32_touch:169]:   Filter mode: IIR_16
[12:56:08][C][esp32_touch:170]:   Debounce count: 2
[12:56:08][C][esp32_touch:171]:   Noise threshold coefficient: 0
[... 103 lines cut ...]

[12:56:16][D][i2s_audio.speaker:116]: Started Speaker
[12:56:18][D][voice_assistant:635]: Event Type: 99
[12:56:19][D][voice_assistant:789]: TTS stream end
[12:56:19][D][voice_assistant:377]: End of audio stream received
[12:56:19][D][voice_assistant:512]: State changed from STREAMING_RESPONSE to RESPONSE_FINISHED
[12:56:19][D][voice_assistant:518]: Desired state set to RESPONSE_FINISHED
[12:56:19][D][esp-idf:000][speaker_task]: I (1136554399) I2S: DMA queue destroyed

[12:56:19][D][light:036]: 'top_led' Setting:
[12:56:19][D][light:051]:   Brightness: 60%
[12:56:19][D][light:059]:   Red: 100%, Green: 0%, Blue: 100%
[12:56:19][D][light:109]:   Effect: 'listening_ww'
[12:56:19][D][voice_assistant:512]: State changed from RESPONSE_FINISHED to IDLE
[12:56:19][D][voice_assistant:518]: Desired state set to IDLE
[12:56:19][D][i2s_audio.speaker:129]: Stopped Speaker
[12:56:20][D][micro_wake_word:399]: Resetting buffers and probabilities
[12:56:20][D][micro_wake_word:195]: State changed from IDLE to START_MICROPHONE
[12:56:20][D][micro_wake_word:107]: Starting Microphone
[12:56:20][D][micro_wake_word:195]: State changed from START_MICROPHONE to STARTING_MICROPHONE
[12:56:20][D][esp-idf:000]: I (1136554641) I2S: DMA Malloc info, datalen=blocksize=1024, dma_buf_count=4

[12:56:20][D][micro_wake_word:195]: State changed from STARTING_MICROPHONE to DETECTING_WAKE_WORD
^C[12:56:21][D][api.connection:199]: ESPHome Logs 2024.12.0b1 (192.168.0.111) requested disconnected
```

fabio-garavini, 2025-03-05:
Looking at your logs it seems everything is fine
Have you tried changing your logger level?

```yaml
logger:
  level: VERBOSE
```
or maybe even to `VERY_VERBOSE`
or upgrading to the latest esphome version 2025.2.2

s00500, 2025-03-05:
I was running on the old beta cause it was the first version where it failed (if you have tips on runnning bisect on esphome let me know =D)

Meanwhile I have updated to latest and set log level to verbose. Still same behavior, mic, wakeword, and stt working fine, no audio output working, when its playing I get a LOT of these messages:

```
[14:45:49][V][voice_assistant:446]: Speaker buffer full, trying again next loop
[14:45:49][V][voice_assistant:446]: Speaker buffer full, trying again next loop
[14:45:49][V][voice_assistant:446]: Speaker buffer full, trying again next loop
[14:45:49][V][voice_assistant:446]: Speaker buffer full, trying again next loop
[14:45:49][V][voice_assistant:446]: Speaker buffer full, trying again next loop
[14:45:49][V][voice_assistant:446]: Speaker buffer full, trying again next loop
[14:45:49][V][voice_assistant:446]: Speaker buffer full, trying again next loop
[14:45:49][V][voice_assistant:446]: Speaker buffer full, trying again next loop
[14:45:49][V][voice_assistant:817]: Received audio: 1024 bytes from API
[14:45:49][V][voice_assistant:446]: Speaker buffer full, trying again next loop
[14:45:49][V][voice_assistant:446]: Speaker buffer full, trying again next loop
[14:45:49][V][voice_assistant:446]: Speaker buffer full, trying again next loop
[14:45:49][V][voice_assistant:446]: Speaker buffer full, trying again next loop
[14:45:49][V][voice_assistant:446]: Speaker buffer full, trying again next loop
[14:45:49][V][voice_assistant:446]: Speaker buffer full, trying again next loop
[14:45:49][V][voice_assistant:446]: Speaker buffer full, trying again next loop
[14:45:49][V][voice_assistant:446]: Speaker buffer full, trying again next loop
[14:45:49][V][voice_assistant:446]: Speaker buffer full, trying again next loop
[14:45:49][V][voice_assistant:446]: Speaker buffer full, trying again next loop
[14:45:49][V][voice_assistant:446]: Speaker buffer full, trying again next loop
[14:45:49][V][voice_assistant:446]: Speaker buffer full, trying again next loop
[14:45:49][V][voice_assistant:446]: Speaker buffer full, trying again next loop
[14:45:49][V][voice_assistant:446]: Speaker buffer full, trying again next loop
[14:45:49][V][voice_assistant:446]: Speaker buffer full, trying again next loop
[14:45:49][V][voice_assistant:446]: Speaker buffer full, trying again next loop
[14:45:49][V][voice_assistant:446]: Speaker buffer full, trying again next loop
[14:45:49][V][voice_assistant:817]: Received audio: 1024 bytes from API
[14:45:49][V][voice_assistant:446]: Speaker buffer full, trying again next loop
[14:45:49][V][voice_assistant:446]: Speaker buffer full, trying again next loop
[14:45:49][V][voice_assistant:446]: Speaker buffer full, trying again next loop
[14:45:49][V][voice_assistant:446]: Speaker buffer full, trying again next loop
[14:45:49][V][voice_assistant:446]: Speaker buffer full, trying again next loop
[14:45:49][D][voice_assistant:635]: Event Type: 99
[14:45:49][D][voice_assistant:789]: TTS stream end
[14:45:49][V][voice_assistant:446]: Speaker buffer full, trying again next loop
[14:45:49][D][voice_assistant:377]: End of audio stream received
[14:45:49][D][voice_assistant:512]: State changed from STREAMING_RESPONSE to RESPONSE_FINISHED
[14:45:49][D][voice_assistant:518]: Desired state set to RESPONSE_FINISHED
[14:45:49][V][voice_assistant:446]: Speaker buffer full, trying again next loop
[14:45:49][V][voice_assistant:446]: Speaker buffer full, trying again next loop
[14:45:49][V][voice_assistant:446]: Speaker buffer full, trying again next loop
[14:45:49][V][voice_assistant:446]: Speaker buffer full, trying again next loop
[14:45:49][V][voice_assistant:446]: Speaker buffer full, trying again next loop
[14:45:49][V][voice_assistant:446]: Speaker buffer full, trying again next loop
[14:45:49][V][voice_assistant:446]: Speaker buffer full, trying again next loop
```

s00500, 2025-03-06:
Any other threads you could link me to?

fabio-garavini, 2025-03-06:
#5180