esphome/issues #6657: esphome does not even compile

boehser-enkel, 2025-01-08:
### The problem

Can't compile anything. 
Install -> manual download: the logfile just shows nothing
This happens with old and newly created devices.
Also tried to run a fresh install of esphome

### Which version of ESPHome has the issue?

2024.12.2

### What type of installation are you using?

Docker

### What platform are you using?

ESP8266

### Additional information

  esphome:
    container_name: ESPHome
    image: esphome/esphome
    volumes:
      - ./esphome/config:/config
      - /etc/localtime:/etc/localtime:ro
    restart: always
    privileged: true
    network_mode: host

rjodwyer, 2025-01-11:
same here at the moment, HA 2025.1.0 and ESPHome 2024.12.2.

there is a 403 in the logs, but not sure what it refers to: WARNING 403 GET /run (0.0.0.0) 0.74ms otherwise there is only gets

boehser-enkel, 2025-01-15:
My bad. Error did not occur locally but only remote, accessed via nginx reverse proxy.
I activated Websockets support which fixed the issue

closed 2025-01-15 by boehser-enkel

dekiesel, 2025-05-03:
> My bad. Error did not occur locally but only remote, accessed via nginx reverse proxy. I activated Websockets support which fixed the issue

How did you activate websocket support?