esphome/issues #6658: MAC address in logs and in the network are different when using ETHERNET W5500, nmap in homeassistant cannot find the esp32 device because of this

sermaster2477, 2025-01-08:
### The problem

I have a lot of devices based on esp32 in conjunction with w5500 and disabled wifi.
They are all connected to homeassistant and all behave the same in terms of mac address.
I think that they display the mac address from the wifi module in the log,
and the ethernet module is assigned a mac address calculated from the wifi module but increased.
In connection with this, nmap cannot correctly detect the device on the network, since the mac addresses do not match.
When using the wifi module for communication, no problems arise, devices are automatically detected by nmap.
I see the following picture LOG ESP32 tells me that the MAC address is C4:D8:D5:94:7E:18 and the router that issued the IP address,
reports that my ESP32 device uses the following mac address on the network C4:D8:D5:94:7E:1B why is this happening?
I think that this is automatically implemented inside the W5500 driver library to reduce problems with duplicating MAC addresses in the network.
In the ethernet component, there is no way to set the MAC address manually, like for example the IP address of the w5500 module.
I think that it would be correct to take into account which communication module is used and display the assigned MAC in the Module Log.
If WiFi is not declared, then you need to send the MAC address that is actually used in the network.

### Which version of ESPHome has the issue?

2024.12.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024

### What platform are you using?

ESP32

### Board

nodemcu

### Component causing the issue

ETHERNET W5500

### YAML Config

```yaml
esphome:
  name: mibc-esp32-prot
  friendly_name: mibc-esp32-prot
  comment: "mibc-esp32-prot"

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "U6+gp0HqRPejMf+XAjp6oJqt7DgcAJWWayeDIQcSFjQ="

ota:
  - platform: esphome
    password: "c45d8acd906fe2991d4d6ffc558440f4"

#wifi:
#  networks:
#  - hidden: true
#    ssid: !secret wifi_ssid
#    password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
#  ap:
#    ssid: "Mbic32Hp Fallback Hotspot"
#    password: "2kuub8gTKCcg"

#captive_portal:

ethernet:
  type: W5500
  clk_pin: GPIO18
  mosi_pin: GPIO23
  miso_pin: GPIO19
  cs_pin: GPIO15
  interrupt_pin: GPIO0
  reset_pin: GPIO5 

button:                  
  - platform: restart
    name: "Restart"
```

### Anything in the logs that might be useful for us?

```txt
there are no errors, the devices work perfectly. The MAC address is just different in the log and on the network.
```

sermaster2477, 2025-01-08:
INFO ESPHome 2024.11.2
INFO Reading configuration /config/esphome/test-rs485.yaml...
WARNING GPIO15 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO0 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO5 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Starting log output from 10.10.19.224 using esphome API
INFO Successfully connected to mibc-esp32-prot @ 10.10.19.224 in 0.003s
INFO Successful handshake with mibc-esp32-prot @ 10.10.19.224 in 0.070s
[17:27:41][I][app:100]: ESPHome version 2024.11.2 compiled on Jan  8 2025, 15:17:43
[17:27:41][C][logger:185]: Logger:
[17:27:41][C][logger:186]:   Level: DEBUG
[17:27:41][C][logger:188]:   Log Baud Rate: 115200
[17:27:41][C][logger:189]:   Hardware UART: UART0
[17:27:41][C][gpio.one_wire:020]: GPIO 1-wire bus:
[17:27:41][C][gpio.one_wire:021]:   Pin: GPIO32
[17:27:41][W][gpio.one_wire:078]:   Found no devices!
[17:27:41][C][restart.button:017]: Restart Button 'Restart'
[17:27:41][C][restart.button:017]:   Icon: 'mdi:restart'
[17:27:41][C][ethernet:326]: Ethernet:
[17:27:41][C][ethernet:515]:   IP Address: 10.10.19.224
[17:27:41][C][ethernet:516]:   Hostname: 'mibc-esp32-prot'
[17:27:41][C][ethernet:517]:   Subnet: 255.255.255.0
[17:27:41][C][ethernet:518]:   Gateway: 10.10.19.1
[17:27:41][C][ethernet:523]:   DNS1: 10.10.19.1
[17:27:41][C][ethernet:524]:   DNS2: 0.0.0.0
[17:27:41][C][ethernet:536]:   MAC Address: C4:D8:D5:94:7E:18
[17:27:41][C][ethernet:537]:   Is Full Duplex: YES
[17:27:41][C][ethernet:538]:   Link Speed: 100
[17:27:41][C][ethernet:329]:   CLK Pin: 18
[17:27:41][C][ethernet:330]:   MISO Pin: 19
[17:27:41][C][ethernet:331]:   MOSI Pin: 23
[17:27:41][C][ethernet:332]:   CS Pin: 15
[17:27:41][C][ethernet:339]:   IRQ Pin: 0
[17:27:41][C][ethernet:341]:   Reset Pin: 5
[17:27:41][C][ethernet:342]:   Clock Speed: 26 MHz
[17:27:41][C][ethernet:351]:   Type: W5500
[17:27:41][C][mdns:116]: mDNS:
[17:27:41][C][mdns:117]:   Hostname: mibc-esp32-prot
[17:27:41][C][esphome.ota:073]: Over-The-Air updates:
[17:27:41][C][esphome.ota:074]:   Address: mibc-esp32-prot.local:3232
[17:27:41][C][esphome.ota:075]:   Version: 2
[17:27:41][C][esphome.ota:078]:   Password configured
[17:27:41][C][safe_mode:018]: Safe Mode:
[17:27:41][C][safe_mode:020]:   Boot considered successful after 60 seconds
[17:27:41][C][safe_mode:021]:   Invoke after 10 boot attempts
[17:27:41][C][safe_mode:023]:   Remain in safe mode for 300 seconds
[17:27:41][W][safe_mode:029]: Last reset occurred too quickly; safe mode will be invoked in 2 restarts
[17:27:41][C][api:140]: API Server:
[17:27:41][C][api:141]:   Address: mibc-esp32-prot.local:6053
[17:27:41][C][api:143]:   Using noise encryption: YES

sermaster2477, 2025-01-08:
[image]

randybb, 2025-01-08:
Duplicate of https://github.com/esphome/issues/issues/5167

closed 2025-01-08 by randybb