esphome/issues #6661: esp8266 prometheus endpoint (/metrics) output is cut with increasing no of sensors

pgolawsk, 2025-01-09:
### The problem

I have couple of boards with esp8266 (ESP 12F module) and those works fine under HomeAssistant and just logging to them via web interface - all sensors are visible.
However for some of those boards I can see that when I'm adding more sensors (this issue seems to be highly correlated with no of sensors) the prometheus endpoint output for scrapping is cut. 

The more senses I add the less prometheus output I get. To the point when added so many of them that nothing is returned from `/metrics`.

On ESP32 I have not experienced such issues yet, even I have much more sensors for some YAMLs.

- Cut `/metrics` example 1:

```ini
#TYPE esphome_sensor_value gauge
#TYPE esphome_sensor_failed gauge
esphome_sensor_failed{id="test_esp12f-29_wifi_signal",area="Test",node="esp12f-29",friendly_name="Test",name="Test esp12f-29 WiFi Signal"} 0
esphome_sensor_value{id="test_esp12f-29_wifi_signal",area="Test",node="esp12f-29",friendly_name="Test",name="Test esp12f-29 WiFi Signal",unit="dBm"} -71
esphome_sensor_failed{id="test_esp12f-29_wifi_strength",area="Test",node="esp12f-29",friendly_name="Test",name="Test esp12f-29 WiFi Strength"} 0
esphome_sensor_value{id="test_esp12f-29_wifi_strength",area="Test",node="esp12f-29",friendly_name="Test",name="Test esp12f-29 WiFi Strength",unit="%"} 58
esphome_sensor_failed{id="test_temperature",area="Test",node="esp12f-29",friendly_name="Test",name="Test Temperature"} 0
esphome_sensor_value{id="test_temperature",area="Test",node="esp12f-29",friendly_name="Test",name="Test Temperature",unit="°C"} 21.08
esphome_sensor_failed{id="test_pressure",area="Test",node="esp12f-29",friendly_name="Test",name="Test Pressure"} 0
esphome_sensor_value{id="test_pressure",area="Test",node="esp12f-29",friendly_name="Test",name="Test Pressure",unit="hPa"} 1000.97
esphome_sensor_failed{id="test_humidity",area="Test",node="esp12f-29",friendly_name="Test",name="Test Humidity"} 0
esphome_sensor_value{id="test_humidity",area="Test",node="esp12f-29",friendly_name="Test",name="Test Humidity",unit="%"} 48.05
esphome_sensor_failed{id="test_gas_resistance",area="Test",node="esp12f-29",friendly_name="Test",name="Test Gas Resistance"} 0
esphome_sensor_value{id="test_gas_resistance",area="Test",node="esp12f-29",friendly_name="Test",name="Test Gas Resistance",unit="Ω"} 21400.84
esphome_sensor_failed{id="test_illuminance",area="Test",node="esp12f-29",friendly_name="Test",name="Test Illuminance"} 1
esphome_sensor_failed{id="esp12f-29_heap_free",area="Test",node="esp12f-29",friendly_name="Test",name="esp12f-29 Heap Free"} 0
esphome_sensor_value{id="esp12f-29_heap_free",area="Test",node="esp12f-29",friendly_name="Test",name="esp12f-29 Heap Free",unit="B"} 12304
esphome_sensor_failed{id="esp12f-29_heap_max_block",area="Test",node="esp12f-29",friendly_name="Test",name="esp12f-29 Heap Max Block"} 0
esphome_sensor_value{id="esp12f-29_heap_max_block",area="Test",node="esp12f-29",friendly_name="Test",name="esp12f-29 Heap Max Block",unit="B"} 6464
esphome_sensor_failed{id="esp12f-29_loop_time",area="Test",node="esp12f-29",friendly_name="Test",name="esp12f-29 Loop Time"} 0
esphome_sensor_value{id="esp12f-29_loop_time",area="Test",node="esp12f-29",friendly_name="Test",name="esp12f-29 Loop Time",unit="ms"} 26
#TYPE esphome_binary_sensor_value gauge
#TYPE esphome_binary_sensor_failed gauge
esphome_binary_sensor_failed{id="test2_doorbell",area="Test",node="esp12f-29",friendly_name="Test",name="Test2 Doorbell"} 0
esphome_binary_sensor_value{id="test2_doorbell",area="Test",node="esp12f-29",friendly_name="Test",name="Test2 Doorbell"} 0
#TYPE esphome_light_state gauge
#TYPE esphome_light_color gauge
#TYPE esphome_light_effect_active gauge
esphome_light_state{id="test_led",area="Test",node="esp12f-29",friendly_name="Test",name="Test LED"} 0
esphome_light_color{id="test_led",area="Test",node="esp12f-29",friendly_name="Test",name="Test LED",channel="brightness"} 0.00
esphome_light_color{id="test_led",area="Test",node="esp12f-29
```

 - Cut `/metrics` example 2:

```ini
#TYPE esphome_sensor_value gauge
#TYPE esphome_sensor_failed gauge
esphome_sensor_failed{id="test_esp12f-29_wifi_signal",area="Test",node="esp12f-29",friendly_name="Test",name="Test esp12f-29 WiFi Signal"} 0
esphome_sensor_value{id="test_esp12f-29_wifi_signal",area="Test",node="esp12f-29",friendly_name="Test",name="Test esp12f-29 WiFi Signal",unit="dBm"} -75
esphome_sensor_failed{id="test_esp12f-29_wifi_strength",area="Test",node="esp12f-29",friendly_name="Test",name="Test esp12f-29 WiFi Strength"} 0
esphome_sensor_value{id="test_esp12f-29_wifi_strength",area="Test",node="esp12f-29",friendly_name="Test",name="Test esp12f-29 WiFi Strength",unit="%"} 50
esphome_sensor_failed{id="test_temperature",area="Test",node="esp12f-29",friendly_name="Test",name="Test Temperature"} 0
esphome_sensor_value{id="test_temperature",area="Test",node="esp12f-29",friendly_name="Test",name="Test Temperature",unit="°C"} 21.09
esphome_sensor_failed{id="test_pressure",area="Test",node="esp12f-29",friendly_name="Test",name="Test Pressure"} 0
esphome_sensor_value{id="test_pressure",area="Test",node="esp12f-29",friendly_name="Test",name="Test Pressure",unit="hPa"} 1001.00
esphome_sensor_failed{id="test_humidity",area="Test",node="esp12f-29",friendly_name="Test",name="Test Humidity"} 0
esphome_sensor_value{id="test_humidity",area="Test",node="esp12f-29",friendly_name="Test",name="Test Humidity",unit="%"} 47.47
esphome_sensor_failed{id="test_gas_resistance",area="Test",node="esp12f-29",friendly_name="Test",name="Test Gas Resistance"} 0
esphome_sensor_value{id="test_gas_resistance",area="Test",node="esp12f-29",friendly_name="Test",name="Test Gas Resistance",unit="Ω"} 20766.43
esphome_sensor_failed{id="test_illuminance",area="Test",node="esp12f-29",friendly_name="Test",name="Test Illuminance"} 1
esphome_sensor_failed{id="esp12f-29_heap_free",area="Test",node="esp12f-29",friendly_name="Test",name="esp12f-29 Heap Free"} 0
esphome_sensor_value{id="esp12f-29_heap_free",area="Test",node="esp12f-29",friendly_name="Test",name="esp12f-29 Heap Free",unit="B"} 13632
esphome_sensor_failed{id="esp12f-29_heap_max_block",area="Test",node="esp12f-29",friendly_name="Test",name="esp12f-29 Heap Max Block"} 0
esphome_sensor_value{id="esp12f-29_heap_max_block",area="Test",node="esp12f-29",friendly_name="Test",name="esp12f-29 Heap Max Block",unit="B"} 6568
esphome_sensor_failed{id="esp12f-29_loop_time",area="Test",node="esp12f-29",friendly_name="Test",name="esp12f-29 Loop Time"} 0
esphome_sensor_value{id="esp12f-29_loop_time",area="Test",node="esp12f-29",friendly_name="Test",name="esp12f-29 Loop Time",unit="ms"} 31
#TYPE esphome_binary_sensor_value gauge
[... 904 lines cut ...]
[00:48:30][C][mqtt:715]:   Topic: 'measures/room/rtttl_play_it'
[00:48:30][C][mqtt:716]:   QoS: 0
[00:48:39][D][text_sensor:064]: 'Test esp12f-29 Uptime': Sending state '26s'
[00:48:39][D][sensor:093]: 'Test esp12f-29 Uptime [min]': Sending state 0.43217 min with 0 decimals of accuracy
[00:48:40][D][bme680.sensor:332]: Got temperature=23.3°C pressure=999.1hPa humidity=41.1% gas_resistance=4011.2Ω
[00:48:40][D][sensor:093]: 'Test Temperature': Sending state 20.93676 °C with 2 decimals of accuracy
[00:48:40][D][sensor:093]: 'Test Pressure': Sending state 1006.12018 hPa with 2 decimals of accuracy
[00:48:40][D][sensor:093]: 'Test Humidity': Sending state 45.48106 % with 2 decimals of accuracy
[00:48:40][D][sensor:093]: 'Test Gas Resistance': Sending state 4011.15405 Ω with 2 decimals of accuracy
[00:48:55][D][sensor:093]: 'Test esp12f-29 WiFi Signal': Sending state -77.00000 dBm with 0 decimals of accuracy
[00:49:09][D][text_sensor:064]: 'Test esp12f-29 Uptime': Sending state '56s'
[00:49:09][D][sensor:093]: 'Test esp12f-29 Uptime [min]': Sending state 0.93207 min with 0 decimals of accuracy
[00:49:10][D][bme680.sensor:332]: Got temperature=23.3°C pressure=999.1hPa humidity=40.1% gas_resistance=5174.5Ω
[00:49:10][D][sensor:093]: 'Test Temperature': Sending state 20.94560 °C with 2 decimals of accuracy
[00:49:10][D][sensor:093]: 'Test Pressure': Sending state 1006.14862 hPa with 2 decimals of accuracy
[00:49:10][D][sensor:093]: 'Test Humidity': Sending state 44.50609 % with 2 decimals of accuracy
[00:49:10][D][sensor:093]: 'Test Gas Resistance': Sending state 5174.46875 Ω with 2 decimals of accuracy
[00:49:22][I][safe_mode:041]: Boot seems successful; resetting boot loop counter
[00:49:25][D][sensor:093]: 'Test esp12f-29 WiFi Signal': Sending state -78.00000 dBm with 0 decimals of accuracy
[00:49:39][D][text_sensor:064]: 'Test esp12f-29 Uptime': Sending state '1m 26s'
[00:49:39][D][sensor:093]: 'Test esp12f-29 Uptime [min]': Sending state 1.43208 min with 0 decimals of accuracy
[00:49:40][D][bme680.sensor:332]: Got temperature=23.4°C pressure=999.2hPa humidity=42.0% gas_resistance=6270.8Ω
[00:49:40][D][sensor:093]: 'Test Temperature': Sending state 20.98666 °C with 2 decimals of accuracy
...
```

barcar, 2025-01-18:
I'm seeing this too with 2024-12. I have not changed the number of sensors on these esp8266 modules but /metrics is truncated which breaks things for Prometheus.

pgolawsk, 2025-01-23:
Hi @barcar ,
I confirm, Rising above I was sure no of sensors have impact on it, as adding 1 more sensor (like debug sensor) truncated more /metrics, so for sure there is correlation with no of sensors.
Observing this for a few more days with no changes re sensor numbers I see gaps in prometheus data (like 20-40% of each day) and while I was checking /metrics I discovered that /metrics endpoint is just hangs sometimes (page do dot load) while web server on the same esp works instantly with no delay (as normal).

I believe the issue could be created by how esp8266 memory is used/managed, hence a bit unstable behavior:( Maybe when continuous memory block is not available, then prometheus endpoint is truncated or hangs (?)

Hope that someone with knowledge can take a look in this issue?
That's really annoying to have metrics gaps:(

barcar, 2025-05-26:
This is still an issue.

pgolawsk, 2026-01-01:
> error still seems present. SONOFF S31 with ESP Home 2025.12.2
> 
> ```
> #TYPE esphome_sensor_value gauge
> #TYPE esphome_sensor_failed gauge
> esphome_sensor_failed{id="wifi_signal",node="powermon-sumppump",friendly_name="Power Monitor - Sump Pump",name="WiFi Signal"} 0
>  ...
> ```

Hi Derek,
There were some tweaks implemented in 2025.11 and 2025.12 which improved situation for my devices, but I see I have less sensors that in your /metrics ... With esp8266 and limited memory you need to limit prometheus /metrics output...

To improve your case I would recommend to:
1. Do not set "friendly_name" for the device - then you will have less characters each row... - that would have the biggest impact IMHO. 
2. If not enough, then consider to shorten sensor names and node name - that would make prometheus output less characters, so maybe all sensors will fit with available memory...
3.  If still not enough then ... (last resort option) reduce no of sensors in prometheus output - via setting some of then as internal and setting prometheus to skip internal ones.

Hope this helps.
Regards, Pawelo