esphome/issues #6664: binary_sensort not always updating in HA when having over 200 sensors

bkbartk, 2025-01-09:
### The problem

I have an esphome device with 200 sensors, and sometimes the state of a sensor changes, but if this is a binary_sensor the state is some scenario's isn't send to home assistant, which means HA thinks the state is different then the state in the esp modules webinterface.
this connectionplem seems to be related to all bit/bool like entities (switch,binarysenorm,etc) but I'm not sure.

### Which version of ESPHome has the issue?

2024.11.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.1.1

### What platform are you using?

ESP8266

### Component causing the issue

binary_sensor

### YAML Config

```yaml
the full content is too long for github so:
https://github.com/Mosibi/Midea-heat-pump-ESPHome/blob/master/heatpump.yaml
```

### Anything in the logs that might be useful for us?

```txt
no
```

### Additional information

I November I started a forum post for this issue,
https://community.home-assistant.io/t/esphome-entities-not-always-updating-in-homeassistant/802143

and a screenshot showing the actual issue
[image]

ssieb, 2025-01-09:
Try this PR: https://github.com/esphome/esphome/pull/7547

bkbartk, 2025-01-10:
thnx
I will add this
```
external_components:
  - source: github://pr#7547
    components: [ api ]
```
and try to upload it later, I can't access my device now,

bkbartk, 2025-01-10:
I updated my controller,
I checks all values before updating and off course they all matched,
Afterwards they still all match,
So I have to do a few more compares, But when can I be sure this fix actually works,

bkbartk, 2025-01-13:
no issues during all of my tests,
I think this PR indeed fixes stuff

closed 2025-01-13 by bkbartk