esphome/issues #6665: Config with platform adc no longer compiles with 2024.12.x

klein1jo, 2025-01-09:
### The problem

I have an ESP using the adc platform that no longer compiles since 2024.12.x.  The latest version I've tried is 2024.12.2.  My ESP is currently working with version 2024.11.3.  I am using esphome installed via pip.  I compile on the command line using "esphome run..."

### Which version of ESPHome has the issue?

2024.12.2

### What type of installation are you using?

pip

### What platform are you using?

ESP32

### Board

nodemcu-32s

### Component causing the issue

adc

### YAML Config

```yaml
esphome:
  name: radon-manometer
  platform: ESP32
  board: nodemcu-32s

sensor:
  - platform: adc
    id: pressure_sensor
    pin: GPIO39
    name: "vacuum"
    update_interval: 0.1s
    unit_of_measurement: "in"
    accuracy_decimals: 2
    attenuation: auto
    filters:
      - lambda: return ((x-0.1)/3.0*300.0-100.0)-1.8;
      - exponential_moving_average:
          alpha: 0.01
          send_every: 100
```

### Anything in the logs that might be useful for us?

```txt
/opt/esphome/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pioenvs/radon-manometer/src/esphome/components/adc/adc_sensor_common.cpp.o: in function `esphome::adc::ADCSensor::get_setup_priority() const':
/opt/esphome/.esphome/build/radon-manometer/src/esphome/components/adc/adc_sensor_common.cpp:21: multiple definition of `esphome::adc::ADCSensor::get_setup_priority() const'; .pioenvs/radon-manometer/src/esphome/components/adc/adc_sensor.cpp.o:/opt/esphome/.esphome/build/radon-manometer/src/esphome/components/adc/adc_sensor.cpp:145: first defined here
/opt/esphome/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pioenvs/radon-manometer/src/esphome/components/adc/adc_sensor_common.cpp.o: in function `esphome::adc::ADCSensor::get_setup_priority() const':
/opt/esphome/.esphome/build/radon-manometer/src/esphome/components/adc/adc_sensor_common.cpp:21: multiple definition of `non-virtual thunk to esphome::adc::ADCSensor::get_setup_priority() const'; .pioenvs/radon-manometer/src/esphome/components/adc/adc_sensor.cpp.o:/opt/esphome/.esphome/build/radon-manometer/src/esphome/components/adc/adc_sensor.cpp:145: first defined here
/opt/esphome/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pioenvs/radon-manometer/src/esphome/components/adc/adc_sensor_common.cpp.o: in function `esphome::adc::ADCSensor::update()':
/opt/esphome/.esphome/build/radon-manometer/src/esphome/components/adc/adc_sensor_common.cpp:9: multiple definition of `esphome::adc::ADCSensor::update()'; .pioenvs/radon-manometer/src/esphome/components/adc/adc_sensor.cpp.o:/opt/esphome/.esphome/build/radon-manometer/src/esphome/components/adc/adc_sensor.cpp:146: first defined here
/opt/esphome/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pioenvs/radon-manometer/src/esphome/components/adc/adc_sensor_common.cpp.o: in function `non-virtual thunk to esphome::adc::ADCSensor::update()':
adc_sensor_common.cpp:(.text._ZThn80_N7esphome3adc9ADCSensor6updateEv+0x0): multiple definition of `non-virtual thunk to esphome::adc::ADCSensor::update()'; .pioenvs/radon-manometer/src/esphome/components/adc/adc_sensor.cpp.o:adc_sensor.cpp:(.text._ZThn80_N7esphome3adc9ADCSensor6updateEv+0x0): first defined here
/opt/esphome/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pioenvs/radon-manometer/src/esphome/components/adc/adc_sensor_common.cpp.o: in function `esphome::adc::ADCSensor::set_sample_count(unsigned char)':
/opt/esphome/.esphome/build/radon-manometer/src/esphome/components/adc/adc_sensor_common.cpp:15: multiple definition of `esphome::adc::ADCSensor::set_sample_count(unsigned char)'; .pioenvs/radon-manometer/src/esphome/components/adc/adc_sensor.cpp.o:/opt/esphome/.esphome/build/radon-manometer/src/esphome/components/adc/adc_sensor.cpp:152: first defined here
/opt/esphome/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pioenvs/radon-manometer/src/esphome/components/adc/adc_sensor_esp32.cpp.o: in function `esphome::adc::ADCSensor::setup()':
/opt/esphome/.esphome/build/radon-manometer/src/esphome/components/adc/adc_sensor_esp32.cpp:24: multiple definition of `esphome::adc::ADCSensor::setup()'; .pioenvs/radon-manometer/src/esphome/components/adc/adc_sensor.cpp.o:/opt/esphome/.esphome/build/radon-manometer/src/esphome/components/adc/adc_sensor.cpp:46: first defined here
/opt/esphome/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pioenvs/radon-manometer/src/esphome/components/adc/adc_sensor_esp32.cpp.o: in function `non-virtual thunk to esphome::adc::ADCSensor::setup()':
adc_sensor_esp32.cpp:(.text._ZThn80_N7esphome3adc9ADCSensor5setupEv+0x0): multiple definition of `non-virtual thunk to esphome::adc::ADCSensor::setup()'; .pioenvs/radon-manometer/src/esphome/components/adc/adc_sensor.cpp.o:adc_sensor.cpp:(.text._ZThn80_N7esphome3adc9ADCSensor5setupEv+0x0): first defined here
/opt/esphome/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pioenvs/radon-manometer/src/esphome/components/adc/adc_sensor_esp32.cpp.o: in function `esphome::adc::ADCSensor::dump_config()':
/opt/esphome/.esphome/build/radon-manometer/src/esphome/components/adc/adc_sensor_esp32.cpp:57: multiple definition of `esphome::adc::ADCSensor::dump_config()'; .pioenvs/radon-manometer/src/esphome/components/adc/adc_sensor.cpp.o:/opt/esphome/.esphome/build/radon-manometer/src/esphome/components/adc/adc_sensor.cpp:96: first defined here
/opt/esphome/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pioenvs/radon-manometer/src/esphome/components/adc/adc_sensor_esp32.cpp.o: in function `non-virtual thunk to esphome::adc::ADCSensor::dump_config()':
adc_sensor_esp32.cpp:(.text._ZThn80_N7esphome3adc9ADCSensor11dump_configEv+0x0): multiple definition of `non-virtual thunk to esphome::adc::ADCSensor::dump_config()'; .pioenvs/radon-manometer/src/esphome/components/adc/adc_sensor.cpp.o:adc_sensor.cpp:(.text._ZThn80_N7esphome3adc9ADCSensor11dump_configEv+0x0): first defined here
/opt/esphome/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pioenvs/radon-manometer/src/esphome/components/adc/adc_sensor_esp32.cpp.o: in function `esphome::adc::ADCSensor::sample()':
/opt/esphome/.esphome/build/radon-manometer/src/esphome/components/adc/adc_sensor_esp32.cpp:84: multiple definition of `esphome::adc::ADCSensor::sample()'; .pioenvs/radon-manometer/src/esphome/components/adc/adc_sensor.cpp.o:/opt/esphome/.esphome/build/radon-manometer/src/esphome/components/adc/adc_sensor.cpp:177: first defined here
/opt/esphome/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pioenvs/radon-manometer/src/esphome/components/adc/adc_sensor_esp32.cpp.o: in function `non-virtual thunk to esphome::adc::ADCSensor::sample()':
adc_sensor_esp32.cpp:(.text._ZThn100_N7esphome3adc9ADCSensor6sampleEv+0x0): multiple definition of `non-virtual thunk to esphome::adc::ADCSensor::sample()'; .pioenvs/radon-manometer/src/esphome/components/adc/adc_sensor.cpp.o:adc_sensor.cpp:(.text._ZThn100_N7esphome3adc9ADCSensor6sampleEv+0x0): first defined here
collect2: error: ld returned 1 exit status
*** [.pioenvs/radon-manometer/firmware.elf] Error 1
```

randybb, 2025-01-09:
clean build files (`esphome clean ...`)

klein1jo, 2025-01-09:
> clean build files (`esphome clean ...`)

Doesn't work.  Went so far as to completely delete the .platformio and .esphome directories.  Problem persists.

randybb, 2025-01-09:
there must be something with your installation. maybe remove your venv and install it again -  adc works with the newer idf 5.1.5 which is in 2024.12 (previously 4.4) without issues, there is only a warning that legacy adc driver is deprecated.

klein1jo, 2025-01-09:
> there must be something with your installation. maybe remove your venv and install it again - adc works with the newer idf 5.1.5 which is in 2024.12 (previously 4.4) without issues, there is only a warning that legacy adc driver is deprecated.

I'm not using a venv.  I'm doing everything with --user.  So by deleting .platformio and .esphome from my user directory I effectively blew away my entire installation.  I'm also using the ads1115 platform and that one works just fine with 2024.12.x.  I included my yaml in my initial post.  Is there a modification I need to be making to my yaml to support adc past 2024.12.x?

closed 2025-01-09 by klein1jo