esphome/issues #6666: LVGL not drawing with waveshare_epaper

Tanemahuta, 2025-01-09:
### The problem

I can't get LVGL to work with display `waveshare_epaper`.

### Which version of ESPHome has the issue?

2024.12.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

all

### What platform are you using?

ESP32

### Board

S3 DevkitC N16R8

### Component causing the issue

lvgl

### YAML Config

```yaml
# Base configuration

# WiFi settings
wifi:
  fast_connect: True
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  ap:
    password: !secret wifi_ap_password

# OTA settings
ota:
  platform: esphome
  password: !secret ota_password

# Add captive portal
captive_portal:

esphome:
  name: test
  name_add_mac_suffix: True
  friendly_name: "test"
  platformio_options:
    build_flags: "-DBOARD_HAS_PSRAM"
    board_build.arduino.memory_type: qio_opi

esp32:
  board: esp32-s3-devkitc-1
  variant: ESP32S3
  flash_size: 16MB
  framework:
    type: arduino

psram:
  mode: octal
  speed: 40MHz

# Enable logging
logger:
[... 208 lines cut ...]
[19:03:03][V][lvgl:000]: lv_obj_update_layout: Layout update end 	(in lv_obj_pos.c line #317)
[19:03:03][V][lvgl:000]: lv_obj_update_layout: Layout update begin 	(in lv_obj_pos.c line #314)
[19:03:03][V][lvgl:000]: lv_obj_update_layout: Layout update end 	(in lv_obj_pos.c line #317)
[19:03:03][V][lvgl:000]: lv_obj_update_layout: Layout update begin 	(in lv_obj_pos.c line #314)
[19:03:03][V][lvgl:000]: lv_obj_update_layout: Layout update end 	(in lv_obj_pos.c line #317)
[19:03:03][V][lvgl:542]: realloc 0x0: 294
[19:03:03][V][lvgl:526]: allocate 48 - > 0x3d82a468
[19:03:03][V][lvgl:542]: realloc 0x0: 64
[19:03:03][V][lvgl:542]: realloc 0x3d82a4a8: 160
[19:03:03][V][lvgl:542]: realloc 0x0: 80
[19:03:03][V][lvgl:542]: realloc 0x3d82a330: 296
[19:03:03][V][lvgl:526]: allocate 36 - > 0x3d82a5b8
[19:03:03][V][lvgl:192]: draw buffer x1=0, y1=0, width=128, height=296
[19:03:04][V][lvgl:533]: free 0x3d82a330
[19:03:04][V][lvgl:533]: free 0x3d82a4a8
[19:03:04][V][lvgl:533]: free 0x3d82a558
[19:03:04][V][lvgl:533]: free 0x3d82a468
[19:03:04][V][lvgl:533]: free 0x3d82a5b8
```

### Additional information

Please note:
- when using a lambda (without lvgl), everything is drawn
- with LVGL the display is not even cleared

mjhilton, 2025-01-14:
Same problem. I'm trying a few different screen sizes (`1.54inv2`, `2.70inv2`) using the WaveShare e-Paper ESP32 Driver Board rev3. 

With the following basic LVGL config:

```
esphome:
  name: waveshare
  friendly_name: Waveshare

esp32:
  board: esp32dev
  framework:
    type: esp-idf

api:

ota:

wifi: 
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  power_save_mode: none
  reboot_timeout: 2min

spi:
  clk_pin: GPIO013
  mosi_pin: GPIO014

# LVGL
lvgl:
  log_level: DEBUG
  buffer_size: 25%
  widgets:
    - label:
        align: center
        text: 'Hello World!'

display:
  - platform: waveshare_epaper
    id: eink_display
    cs_pin: GPIO015
    dc_pin: GPIO027
    busy_pin: GPIO025
    reset_pin: GPIO026
    model: 1.54inv2
    auto_clear_enabled: false
    update_interval: never
    rotation: 270
```

Both screens work fine with the inbuilt display framework/lambda, but on LVGL I get a combo of fully black screen on the ePaper, and sometimes the ESP seems to go unresponsive, can't reliably connect for OTA updates/stuff in the logs about timeouts and failing to load images.

clydebarrow, 2025-01-14:
For e-paper displays the entire display has to be redrawn, you need an update interval other than `never`

clydebarrow, 2025-01-14:
> with no change - just black screen

Look at the OP's config WRT colors. You are drawing white on white, which becomes black on e-paper.

closed 2025-01-14 by clydebarrow

mjhilton, 2025-01-14:
Thanks, will try that one. I would have thought if the colours are inverted, then at least the text should display as white on the black background?

clydebarrow, 2025-01-14:
No, because neither is pure black. The E-paper display driver converts pure black (only!) to white, and everything else to black.

mjhilton, 2025-01-14:
Got it. I assume that means that the standard display rendering engine/lambda system just automagically sorts those colours out for you when dealing with ePaper, then?

clydebarrow, 2025-01-14:
If only using two colours (ON and OFF) counts as automagical, then yes. :-)