esphome/issues #6668: ESP32-S3-WROOM-2-N16R8V

wifispray, 2025-01-10:
### The problem

Cant find a working config for this trying the following doesnt seem to work.

### Which version of ESPHome has the issue?

latest

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Board

N16R8V

swoboda1337, 2025-01-10:
framework is esp-idf or arduino?

wifispray, 2025-01-10:
arduino

I had working on esp-idf but wanted
arduino

swoboda1337, 2025-01-10:
Yes but arduino is based on a really old version of idf which doesnt support some newer variants, I use idf for most of my devices. Docs say idf is recommended for s3: 

https://esphome.io/components/esp32.html#esp-idf-framework

wifispray, 2025-01-10:
Thanks for the tip!

What’s the best working config? I seem to get loads of errors about drivers
etc that are out of date or depreciating

swoboda1337, 2025-01-10:
Yeah will take time to upgrade to the new drivers. Can ignore the warnings for now. I believe RMT and ADC will be upgraded in the next release. Maybe others as well.

wifispray, 2025-01-11:
Ok, thanks.

Do you have a config to share?

Thanks
Neil

swoboda1337, 2025-01-15:
Not for that particular board I have one s3:
```
esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: esp-idf
```

nmcrae85, 2025-02-05:
Can anyone else help here, it seems that the WROOM2 isnt currently supported on esphome as far as I can see??

ssieb, 2025-08-04:
Just use the generic board as shown and set the flash size in `esp32:` section.