esphome/issues #6670: Template switch

VitisEK, 2025-01-11:
### The problem

For the folowing configuration i had an issue:

`
esp32:
  board: esp32dev
  framework:
    type: esp-idf
`
`
switch:
  - platform: template
    name: "Name"
    id: my_id
    optimistic: True
`
I have folowing issue:

`
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld: .pioenvs/plc-a64/src/main.cpp.o:(.literal._Z5setupv+0xb90): undefined reference to `_ZN7esphome9template_14TemplateSwitchC1Ev'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld: .pioenvs/plc-a64/src/main.cpp.o:(.literal._Z5setupv+0xb94): undefined reference to `_ZN7esphome9template_14TemplateSwitch14set_optimisticEb'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld: .pioenvs/plc-a64/src/main.cpp.o:(.literal._Z5setupv+0xb98): undefined reference to `_ZN7esphome9template_14TemplateSwitch17set_assumed_stateEb'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld: .pioenvs/plc-a64/src/main.cpp.o: in function `_Z5setupv':
/data/build/plc-a64/src/main.cpp:2534: undefined reference to `_ZN7esphome9template_14TemplateSwitchC1Ev'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld: .pioenvs/plc-a64/src/main.cpp.o: in function `_ZN7esphome7pcf857414PCF8574GPIOPin7set_pinEh':
/data/build/plc-a64/src/esphome/components/pcf8574/pcf8574.h:53: undefined reference to `_ZN7esphome9template_14TemplateSwitch14set_optimisticEb'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld: .pioenvs/plc-a64/src/main.cpp.o: in function `_ZN7esphome7pcf857414PCF8574GPIOPin9set_flagsENS_4gpio5FlagsE':
/data/build/plc-a64/src/esphome/components/pcf8574/pcf8574.h:55: undefined reference to `_ZN7esphome9template_14TemplateSwitch17set_assumed_stateEb'
`

### Which version of ESPHome has the issue?

2024.12.2

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2025.1.2

### What platform are you using?

ESP32

### Component causing the issue

Template switch

### YAML Config

```yaml
esp32:
  board: esp32dev
  framework:
    type: esp-idf

switch:
  - platform: template
    name: "Name"
    id: my_id
    optimistic: True
`
I have folowing issue:
```

### Anything in the logs that might be useful for us?

```txt
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld: .pioenvs/plc-a64/src/main.cpp.o:(.literal._Z5setupv+0xb90): undefined reference to `_ZN7esphome9template_14TemplateSwitchC1Ev'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld: .pioenvs/plc-a64/src/main.cpp.o:(.literal._Z5setupv+0xb94): undefined reference to `_ZN7esphome9template_14TemplateSwitch14set_optimisticEb'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld: .pioenvs/plc-a64/src/main.cpp.o:(.literal._Z5setupv+0xb98): undefined reference to `_ZN7esphome9template_14TemplateSwitch17set_assumed_stateEb'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld: .pioenvs/plc-a64/src/main.cpp.o: in function `_Z5setupv':
/data/build/plc-a64/src/main.cpp:2534: undefined reference to `_ZN7esphome9template_14TemplateSwitchC1Ev'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld: .pioenvs/plc-a64/src/main.cpp.o: in function `_ZN7esphome7pcf857414PCF8574GPIOPin7set_pinEh':
/data/build/plc-a64/src/esphome/components/pcf8574/pcf8574.h:53: undefined reference to `_ZN7esphome9template_14TemplateSwitch14set_optimisticEb'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld: .pioenvs/plc-a64/src/main.cpp.o: in function `_ZN7esphome7pcf857414PCF8574GPIOPin9set_flagsENS_4gpio5FlagsE':
/data/build/plc-a64/src/esphome/components/pcf8574/pcf8574.h:55: undefined reference to `_ZN7esphome9template_14TemplateSwitch17set_assumed_stateEb'
```

### Additional information

For arduino IDE it works fine, but I have another issue with another components so I prefere esp-idf framework.

ssieb, 2025-01-11:
Do a clean build.

VitisEK, 2025-01-12:
What a beginner's mistake. Thank you, it works.

closed 2025-01-12 by VitisEK