esphome/issues #6672: Component not found: event

koying, 2025-01-12:
### The problem

I can't seem to be able to use the event component. 
ESPHome errors out on `module 'esphome.components.mqtt' has no attribute 'MQTTEventComponent'` and then `Component not found: event`.
MQTT is not enabled on this config (not that I think it matters)

Google returns nothing like this issue, so I'm at loss, because MQTTEventComponent definitely seems to be in `mqtt/__init.py__` in the container.

```shell
root@theia:/config# cd /esphome/esphome/components/mqtt
root@theia:/esphome/esphome/components/mqtt# grep MQTTEventComponent  *
__init__.py:MQTTEventComponent = mqtt_ns.class_("MQTTEventComponent", MQTTComponent)
mqtt_event.cpp:MQTTEventComponent::MQTTEventComponent(event::Event *event) : event_(event) {}
mqtt_event.cpp:void MQTTEventComponent::send_discovery(JsonObject root, mqtt::SendDiscoveryConfig &config) {
mqtt_event.cpp:void MQTTEventComponent::setup() {
mqtt_event.cpp:void MQTTEventComponent::dump_config() {
mqtt_event.cpp:bool MQTTEventComponent::publish_event_(const std::string &event_type) {
mqtt_event.cpp:std::string MQTTEventComponent::component_type() const { return "event"; }
mqtt_event.cpp:const EntityBase *MQTTEventComponent::get_entity() const { return this->event_; }
mqtt_event.h:class MQTTEventComponent : public mqtt::MQTTComponent {
mqtt_event.h:  explicit MQTTEventComponent(event::Event *event);
```

### Which version of ESPHome has the issue?

2024.12.2

### What type of installation are you using?

Docker

### What platform are you using?

ESP8266

### YAML Config

```yaml
esphome:
  name: rf-bridge
  friendly_name: rf-bridge

esp8266:
  board: esp01_1m

packages:
  base: !include common/base_nomq.yaml

# Enable logging
logger:
  baud_rate: 0

button:
  # Restart the ESP
  - platform: restart
    name: "Restart"

uart:
  tx_pin: 1
  rx_pin: 3
  baud_rate: 19200

# Sensors for ESP version and WIFI information
text_sensor:
  # ESPHome version
  - platform: version
    hide_timestamp: true
    name: "ESPHome Version"
  # IP address and connected SSID
  - platform: wifi_info
    ip_address:
      name: "IP Address"
      icon: mdi:wifi
    ssid:
      name: "Connected SSID"
      icon: mdi:wifi-strength-2
sensor:
  # WiFi signal
  - platform: wifi_signal
    name: "WiFi Signal"
    update_interval: 120s
  - platform: uptime
    name: Sonoff RF Bridge Uptime

binary_sensor:
  - platform: status
    name: Sonoff RF Bridge Status

light:
  - platform: status_led
    name: "Switch state"
    pin: GPIO13

rf_bridge:
  on_code_received:
    then:
      - lambda: |-
          uint32_t device = (data.code & 0xFFFFFFF0) >> 4;
          uint32_t button = data.code & 0x0F;
          std::string sButton;
          switch(device) {
            case 0xff5f7:
              switch button: {
                case 2:
                  sButton = "C";
                  break;
              }
              id(REMOTE-TINY-ABCD-03).trigger(sButton);
              break;
          }

event:
  - platform: template
    name: REMOTE-WHITE-ABCD-01
    id: REMOTE-WHITE-ABCD-01
    icon: "mdi:remote"
    device_class: "button"
    event_types:
      - "A"
      - "B"  
      - "C"  
      - "D"  
  - platform: template
    name: REMOTE-TINY-ABCD-03
    id: REMOTE-TINY-ABCD-03
    icon: "mdi:remote"
    device_class: "button"
    event_types:
      - "A"
      - "B"  
      - "C"  
      - "D"
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2024.12.2
INFO Reading configuration /config/rf-bridge.yaml...
ERROR Unable to load component event:
Traceback (most recent call last):
  File "/esphome/esphome/loader.py", line 177, in _lookup_module
    module = importlib.import_module(f"esphome.components.{domain}")
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1206, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1178, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1149, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/esphome/esphome/components/event/__init__.py", line 49, in <module>
    cv.OnlyWith(CONF_MQTT_ID, "mqtt"): cv.declare_id(mqtt.MQTTEventComponent),
                                                     ^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: module 'esphome.components.mqtt' has no attribute 'MQTTEventComponent'
Failed config

event: [source /config/rf-bridge.yaml:76]
  
  Component not found: event.
  - platform: template
    name: REMOTE-WHITE-ABCD-01
    id: REMOTE-WHITE-ABCD-01
    icon: mdi:remote
    device_class: button
    event_types: 
      - A
      - B
      - C
      - D
  - platform: template
    name: REMOTE-TINY-ABCD-03
```

koying, 2025-01-13:
Same issue with a wizard config, btw

```yaml
esphome:
  name: test-event
  friendly_name: test event

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "foobar"

ota:
  - platform: esphome
    password: "barfoo"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Test-Event Fallback Hotspot"
    password: "vUQK15QFpMpn"

captive_portal:

event:
  - platform: template
    name: REMOTE-WHITE-ABCD-01
    id: REMOTE-WHITE-ABCD-01
    icon: "mdi:remote"
    device_class: "button"
    event_types:
      - "A"
      - "B"  
      - "C"  
      - "D"  
  - platform: template
    name: REMOTE-TINY-ABCD-03
    id: REMOTE-TINY-ABCD-03
    icon: "mdi:remote"
    device_class: "button"
    event_types:
      - "A"
      - "B"  
      - "C"  
      - "D"  
```

ssieb, 2025-01-13:
You have other issues in that first config that won't compile.  But after fixing that, I don't have any issues with either of those configs.  So there's something else in your environment.  Do you have a `custom_components` directory or an external component configured in that package?

koying, 2025-01-14:
Hi, thanks for trying it out. Good to know it's only me.

No, that package has just common boilerplate code.
I delete cache and .esphome to no avail.

Looks like esphome itself download some external components with a "mqtt.py", though

```shell
❯ sudo find . -name "mqtt.py"
[sudo] password for cbro: 
./config/.esphome/external_components/0b8510f9/esphome/mqtt.py
./config/.esphome/external_components/3a17a31a/esphome/mqtt.py
./config/.esphome/external_components/3a17a31a/esphome/dashboard/status/mqtt.py
```

Are you also using the vanilla docker image?

ssieb, 2025-01-14:
So you've picked up an external component somewhere.  Delete that directory, then watch when you compile to see if it downloads anything.

I use a direct pip install from my local fork of the git repo.

koying, 2025-01-14:
Yeah. Created a clean instance, and that works. I will investigate and post here the solution, for reference

closed 2025-01-14 by koying