esphome/issues #6673: Cannot use PID Climate pid sensor

soczkers, 2025-01-13:
### The problem

When I'm adding pid sensor to monitor values of resulted value (RESULT, but also tried others and get the same error) I'm getting error.
I'm using this documentatation:
https://esphome.io/components/climate/pid.html#pid-sensor
my full config:
```
esphome:
  name: esphome-web-9bf6c4
  friendly_name: ESPHome Web 9bf6c4
  min_version: 2024.11.0
  name_add_mac_suffix: false

esp32:
  board: esp32dev
  framework:
    type: esp-idf

# Enable logging
logger:

# Enable Home Assistant API
api:

# Allow Over-The-Air updates
ota:
- platform: esphome

wifi:
  ssid: "xx"
  password: "xx"

switch:
  - platform: gpio
    pin: GPIO14
    id: heater
    name: "Grzanie Małego Gekona"

output:
  - platform: slow_pwm
    id: my_slow_pwm
    period: 10s
    turn_on_action:
      - switch.turn_on: heater
    turn_off_action:
      - switch.turn_off: heater

sensor:
  - platform: homeassistant
    name: "gekon_temp"
    id: gekon
    accuracy_decimals: 2
    entity_id: sensor.38_1f_8d_fb_ea_21_ea21_temperature
  - platform: template 
    name: "temperature_sensor_template" 
    id: temperature_sensor_template 
    lambda: |- 
      return id(gekon).state; 
    update_interval: 1s
    accuracy_decimals: 2
  - platform: pid
    name: "PID Climate Result"
    type: RESULT

climate:
  - platform: pid
    name: "PID Climate Controller"
    sensor: temperature_sensor_template
    default_target_temperature: 24°C
    heat_output: my_slow_pwm
    control_parameters:
      kp: 1
      ki: 0
      kd: 0
      

```

Without adding that sensor there is no error. error output:
```INFO ESPHome 2024.12.2
INFO Reading configuration /config/esphome/esphome-web-9bf6c4.yaml...
INFO Generating C++ source...
INFO Updating https://github.com/espressif/esp-protocols.git@mdns-v1.3.2
INFO Compiling app...
Processing esphome-web-9bf6c4 (board: esp32dev; framework: espidf; platform: https://github.com/pioarduino/platform-espressif32.git#51.03.07)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.50105.0 (5.1.5) 
 - tool-cmake @ 3.21.3 
 - tool-esptoolpy @ 4.8.1 
 - tool-mklittlefs @ 3.2.0 
 - tool-ninja @ 1.7.1 
 - tool-riscv32-esp-elf-gdb @ 12.1.0+20221002 
 - tool-xtensa-esp-elf-gdb @ 12.1.0+20221002 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 12.2.0+20230208
Reading CMake configuration...
No dependencies
Compiling .pioenvs/esphome-web-9bf6c4/src/main.cpp.o
Linking .pioenvs/esphome-web-9bf6c4/firmware.elf
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld: .pioenvs/esphome-web-9bf6c4/src/main.cpp.o:(.literal._Z5setupv+0x108): undefined reference to `_ZTVN7esphome3pid16PIDClimateSensorE'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld: .pioenvs/esphome-web-9bf6c4/src/main.cpp.o:(.literal._Z5setupv+0x10c): undefined reference to `_ZTVN7esphome3pid16PIDClimateSensorE'
collect2: error: ld returned 1 exit status
*** [.pioenvs/esphome-web-9bf6c4/firmware.elf] Error 1
========================== [FAILED] Took 8.26 seconds ==========================```

### Which version of ESPHome has the issue?

min_version: 2024.11.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.1.2

### What platform are you using?

ESP32-IDF

### Board

lolin lite

### YAML Config

```yaml
esphome:
  name: esphome-web-9bf6c4
  friendly_name: ESPHome Web 9bf6c4
  min_version: 2024.11.0
  name_add_mac_suffix: false

esp32:
  board: esp32dev
  framework:
    type: esp-idf

# Enable logging
logger:

# Enable Home Assistant API
api:

# Allow Over-The-Air updates
ota:
- platform: esphome

wifi:
  ssid: "xx"
  password: "xx"

switch:
  - platform: gpio
    pin: GPIO14
    id: heater
    name: "Grzanie Małego Gekona"

output:
  - platform: slow_pwm
    id: my_slow_pwm
    period: 10s
    turn_on_action:
      - switch.turn_on: heater
    turn_off_action:
      - switch.turn_off: heater

sensor:
  - platform: homeassistant
    name: "gekon_temp"
    id: gekon
    accuracy_decimals: 2
    entity_id: sensor.38_1f_8d_fb_ea_21_ea21_temperature
  - platform: template 
    name: "temperature_sensor_template" 
    id: temperature_sensor_template 
    lambda: |- 
      return id(gekon).state; 
    update_interval: 1s
    accuracy_decimals: 2
  - platform: pid
    name: "PID Climate Result"
    type: RESULT

climate:
  - platform: pid
    name: "PID Climate Controller"
    sensor: temperature_sensor_template
    default_target_temperature: 24°C
    heat_output: my_slow_pwm
    control_parameters:
      kp: 1
      ki: 0
      kd: 0
```

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration /config/esphome/esphome-web-9bf6c4.yaml...
INFO Generating C++ source...
INFO Updating https://github.com/espressif/esp-protocols.git@mdns-v1.3.2
INFO Compiling app...
Processing esphome-web-9bf6c4 (board: esp32dev; framework: espidf; platform: https://github.com/pioarduino/platform-espressif32.git#51.03.07)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.50105.0 (5.1.5) 
 - tool-cmake @ 3.21.3 
 - tool-esptoolpy @ 4.8.1 
 - tool-mklittlefs @ 3.2.0 
 - tool-ninja @ 1.7.1 
 - tool-riscv32-esp-elf-gdb @ 12.1.0+20221002 
 - tool-xtensa-esp-elf-gdb @ 12.1.0+20221002 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 12.2.0+20230208
Reading CMake configuration...
No dependencies
Compiling .pioenvs/esphome-web-9bf6c4/src/main.cpp.o
Linking .pioenvs/esphome-web-9bf6c4/firmware.elf
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld: .pioenvs/esphome-web-9bf6c4/src/main.cpp.o:(.literal._Z5setupv+0x108): undefined reference to `_ZTVN7esphome3pid16PIDClimateSensorE'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld: .pioenvs/esphome-web-9bf6c4/src/main.cpp.o:(.literal._Z5setupv+0x10c): undefined reference to `_ZTVN7esphome3pid16PIDClimateSensorE'
collect2: error: ld returned 1 exit status
*** [.pioenvs/esphome-web-9bf6c4/firmware.elf] Error 1
========================== [FAILED] Took 8.26 seconds ==========================
```

ssieb, 2025-01-13:
You're using esp-idf.  Try doing a clean build.

soczkers, 2025-01-13:
works @ssieb ! thank you