esphome/issues #6675: SCD4x Temp offset doesnt apply minus figures correctly.

nmcrae85, 2025-01-13:
### The problem

If I apply any minus offsets to this sensor I get very odd values returned. If I do the same with positive values its ok.

So for example if I do this Temperature offset: -0.10 °C I will get reports of Temp and humidity like this.

[14:24:23][D][sensor:093]: 'SCD4x Temperature': Sending state -45.00000 °C with 1 decimals of accuracy
[14:24:23][D][sensor:093]: 'SCD4x Humidity': Sending state 99.99847 % with 0 decimals of accuracy

### Which version of ESPHome has the issue?

2024.12.2

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32-IDF

### Board

ESP32-S3-WROOM-2-N16R8V 

### YAML Config

```yaml
sensor:
  - platform: scd4x
    id: my_scd41
    co2:
      name: "CO2 Concentration"
      id: co2_concentration
    temperature:
      name: "SCD4x Temperature"
      id: scd4x_temperature
      accuracy_decimals: 1
    humidity:
      name: "SCD4x Humidity"
      id: scd4x_humidity
      accuracy_decimals: 0
    update_interval: 60s
    temperature_offset: -0.10
```

wifispray, 2025-01-15:
Anything on this one?

bloneu, 2025-02-14:
The SCD4x datasheet states:

"Recommended temperature offset values are between 0 °C and 20 °C".

Important!
By default, the temperature offset is set to 4 °C.
This means setting a value of 3 °C will increase the temperature output of the sensor relative to the default.
Try:

`- platform: scd4x
    id: sensorid
    temperature_offset: 9.5
    co2:
      name: "Co2"
    temperature:
      name: "Temperature"`

closed 2025-08-19 by github-actions[bot]