esphome/issues #6676: Popping sound using UDA1334 I2S

elik745i, 2025-01-13:
### The problem

Hi gurus, please help me to sort out popping sound prior to play anything on my ESP media player.
My project video presentation is here: https://youtu.be/mt_Qr-lGUJ4?si=XJ88u4Xt6REdcFA8
The issue and solution is described here: https://learn.adafruit.com/adafruit-i2s-stereo-decoder-uda1334a/pi-i2s-tweaks

However I think setting constant bitrate not supported for media player instance? (get compilling errors) 
My sketch is as below:

### Which version of ESPHome has the issue?

2024.12.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.1.2

### What platform are you using?

ESP32

### Board

Lolin32 mini

### Component causing the issue

i2s_audio

### YAML Config

```yaml
captive_portal:

web_server:
  port: 80
  auth:
    username: admin
    password: !secret web_password

# Define the LED connected to GPIO22
status_led:
  pin:
    number: GPIO22
    inverted: false

i2s_audio:
  - id: i2s_out
    i2s_lrclk_pin: GPIO26
    i2s_bclk_pin: GPIO27

  - id: i2s_in
    i2s_lrclk_pin: GPIO32  # ws
    i2s_bclk_pin: GPIO33  # sck

# LR to be grounded

media_player:
  - platform: i2s_audio
    name: ESPHome I2S Media Player
    dac_type: external
    i2s_audio_id: i2s_out
    i2s_dout_pin: GPIO25
    mode: stereo
    id: my_i2s_medial_player
    on_state:
      - lambda: |-
          auto state = id(my_i2s_medial_player).state;
          if (state == media_player::MEDIA_PLAYER_STATE_PLAYING) {
            ESP_LOGD("i2s_audio", "Playback started.");
          } else if (state == media_player::MEDIA_PLAYER_STATE_IDLE) {
            ESP_LOGD("i2s_audio", "Playback stopped.");
          }

# Example configuration entry
microphone:
  - platform: i2s_audio
    i2s_audio_id: i2s_in
    i2s_din_pin: GPIO14  # sd
    adc_type: external
    id: mic_i2s
    pdm: false

voice_assistant:
  microphone: mic_i2s

# Battery voltage monitoring using ADC
sensor:
  - platform: adc
    pin: GPIO36
    name: "Battery Voltage"
    id: battery_voltage
    update_interval: 60s
    attenuation: 12db
    filters:
      - multiply: 3.866  # Calibration factor for 100kΩ and 33kΩ resistors
      # Placeholder filter; will be dynamically updated
      - lambda: |-
          ESP_LOGD("adc", "Raw ADC Voltage: %.3f", x);
          return x;
  - platform: wifi_signal
    name: "WiFi Signal Strength"
    update_interval: 60s  # Update every 60 seconds
    
text_sensor:
  - platform: homeassistant
    name: "Measured Voltage"
    id: measured_voltage
    entity_id: input_text.measured_voltage

# Display voltage in logs and update calibration factor
interval:
  - interval: 60s
    then:
      - lambda: |-
          float measured = atof(id(measured_voltage).state.c_str());
          float esp_voltage = id(battery_voltage).state;
          if (esp_voltage > 0 && measured > 0) {
            float calibration_factor = measured / esp_voltage;
            ESP_LOGD("calibration", "New Calibration Factor: %.3f", calibration_factor);
            id(battery_voltage).set_filters({
              new esphome::sensor::MultiplyFilter(calibration_factor)
            });
          }

switch:
  - platform: restart
    name: "Reboot ESP Device"
```

majuss, 2025-01-13:
This is most likely created by insuffiecient EMI shielding. The UDA has a headphone amp, and amps will amplify, even electrical noise is amplified. Keep the i2s cable runs as short as possible dont leave any other cables above or below the UDA. Take EMI shielding tape and shield as much as possible. My crackling noises came from LEDs. As soon as I seperated the LEDs from the rest with copper tape everything was good.

elik745i, 2025-01-13:
> This is most likely created by insuffiecient EMI shielding. The UDA has a headphone amp, and amps will amplify, even electrical noise is amplified. Keep the i2s cable runs as short as possible dont [...]

Thing is, when internet radio or TTS is played, there is no popping sound, so that is definatelly not a setup issue but signal sent from ESPHome or Home Assistant...
My wires are as short as possible:
[image]

elik745i, 2025-01-13:
> This is most likely created by insuffiecient EMI shielding. The UDA has a headphone amp, and amps will amplify, even electrical noise is amplified. Keep the i2s cable runs as short as possible dont [...]

unfortunatelly, that had absolutelly no effect (( Even when module is fully shielded, I can still hear klicks when changing songs right before it starts playing.
[image]