esphome/issues #6677: Device partially stops responding after "HTTP Request failed"

davidstrobl, 2025-01-13:
### The problem

Hi, 
I am using a ESP device for monitoring. I retrieve the data via the API interface using the http_request component. Sometimes it happens that the device partially stops responding - the display gets stuck, the sensors stops updating and the logs stop being loaded into the HA ESPhome interface - see. below. 
Ping to device works. 
The web page of the sensors responds, but loads the header with information about the running time only (the time continues on). 
The device cannot be updated via OTA. 
The only thing that helps is a reboot.

The problem appears randomly, propably depending on the unspecified responses of monitored device, which is difficult to simulate. 
But the identical problem occurs and appears almost immediately when I set the timeout of request to a low value. The following logs describe such a situation.

```
[18:11:59][D][esp-idf:000]: W (285574) HTTP_CLIENT: Connection timed out before data was ready!

[18:11:59][E][http_request.idf:172]: HTTP Request failed; URL: http://10.27.27.200/; Code: -1
[18:11:59][E][component:164]: Component http_request set Error flag: unspecified
```
AND NOW IS OVER
After some time the following messages are added:

```
WARNING guition-esp32-4848s040 @ 10.27.28.30: Connection error occurred: Ping response not received after 90.0 seconds
INFO Processing unexpected disconnect from ESPHome API for guition-esp32-4848s040 @ 10.27.28.30
WARNING Disconnected from API
INFO Successfully connected to guition-esp32-4848s040 @ 10.27.28.30 in 0.133s
WARNING Can't connect to ESPHome API for guition-esp32-4848s040 @ 10.27.28.30: Timeout waiting for HelloResponse, ConnectResponse after 30.0s (TimeoutAPIError)
INFO Trying to connect to guition-esp32-4848s040 @ 10.27.28.30 in the background
WARNING guition-esp32-4848s040 @ 10.27.28.30: Connection error occurred: guition-esp32-4848s040 @ 10.27.28.30: Connection lost
INFO Successfully connected to guition-esp32-4848s040 @ 10.27.28.30 in 0.121s
WARNING guition-esp32-4848s040 @ 10.27.28.30: Connection error occurred: guition-esp32-4848s040 @ 10.27.28.30: Connection lost
INFO Successfully connected to guition-esp32-4848s040 @ 10.27.28.30 in 0.155s
WARNING guition-esp32-4848s040 @ 10.27.28.30: Connection error occurred: guition-esp32-4848s040 @ 10.27.28.30: Connection lost
INFO Successfully connected to guition-esp32-4848s040 @ 10.27.28.30 in 0.250s
WARNING guition-esp32-4848s040 @ 10.27.28.30: Connection error occurred: guition-esp32-4848s040 @ 10.27.28.30: Connection lost
```

Finally, I managed to catch the error message with a long enough timeout (8s):

```
21:15:45 [D] [esp-idf:000] [0;31mE (2416453) transport_base: poll_read select error 104, errno = Connection reset by peer, fd = 60 
21:15:45 [E] [http_request.idf:172] HTTP Request failed; URL: http://10.27.27.200/; Code: -1 
21:15:45 [E] [component:164] Component http_request set Error flag: unspecified
```

It looks like a problem in http_request when the request fails.

Thx for help.

### Which version of ESPHome has the issue?

2024.12.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.1.0

### What platform are you using?

ESP32-IDF

### Board

Guition ESP32-S3-4848S040

### Component causing the issue
[... 771 lines cut ...]
[19:01:31][I][solaxdotaz:538]: Request succeeded!
[19:01:31][D][text_sensor:064]: 'sporeba SUM': Sending state '375'
[19:01:31][D][text_sensor:064]: 'FVE vyroba': Sending state '0'
[19:01:31][D][text_sensor:064]: 'PV1': Sending state '0W'
[19:01:31][D][text_sensor:064]: 'PV2': Sending state '0W'
[19:01:31][D][text_sensor:064]: 'Stav baterie': Sending state '48%'
[19:01:31][D][sensor:093]: 'Feed in': Sending state 0.00000  with 1 decimals of accuracy
[19:01:31][D][sensor:093]: 'solax consumption': Sending state 375.00000  with 1 decimals of accuracy
[19:01:31][D][sensor:093]: 'battery power': Sending state -377.00000  with 1 decimals of accuracy
[19:01:31][D][text_sensor:064]: 'Grid in today': Sending state '67W'
[19:01:31][D][text_sensor:064]: 'dataSOLAX-pocet': Sending state '300'
[19:01:31][I][solaxdotaz:577]: zacatek nastavovani sensoru
[19:01:31][I][solaxdotaz:591]: konec nastavovani sensoru
[19:01:31][W][component:237]: Component interval took a long time for an operation (1653 ms).
[19:01:31][W][component:238]: Components should block for at most 30 ms.
[19:01:32][W][component:237]: Component lvgl took a long time for an operation (122 ms).
[19:01:32][W][component:238]: Components should block for at most 30 ms.
[19:01:32][I][safe_mode:041]: Boot seems successful; resetting boot loop counter
[19:01:32][D][esp32.preferences:114]: Saving 1 preferences to flash...
[19:01:32][D][esp32.preferences:142]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[19:01:49][D][esp-idf:000]: W (77447) HTTP_CLIENT: Connection timed out before data was ready!

[19:01:49][E][http_request.idf:172]: HTTP Request failed; URL: http://10.27.27.200/; Code: -1
[19:01:49][E][component:164]: Component http_request set Error flag: unspecified
```

Mikefila, 2025-01-22:
I have this same problem, did you make any headway with it?

davidstrobl, 2025-01-22:
No

Mikefila, 2025-01-23:
I think it's piper, changing the tts to google translate seems to fix it.

davidstrobl, 2025-01-23:
In my opinion, this is a bug, because problem that cannot be influenced shouldn't end up freezing the device. Network outage, temporary unavailability of the other party, etc.

davidstrobl, 2025-05-18:
Problem still persist with latest versions:

`10:09:54
[D]
[esp-idf:000]
[0;31mE (1700181) transport_base: poll_read select error 104, errno = Connection reset by peer, fd = 59`
`10:09:54
[E]
[http_request.idf:174]
HTTP Request failed; URL: http://10.27.27.200/; Code: -1`
`10:09:54
[E]
[component:164]
Component http_request set Error flag: unspecified`

nzottmann, 2025-09-29:
I experience the same problem.
I can reproduce this with my own local http server. If I add a delay in the route before returning the response, esphome gets stuck if the delay is longer than ~4s.

Changing the http timeout does not change this. There is no watchdog reset, it just stops sending out logs and sensor data and the lvgl interface freezes. If the delay is <4s, the interface freezes only until the response is received.

Setting `capture_response` to `False` resolves the issue, but does not help here as I need the response.