esphome/issues #6679: binary_sensor publish_state(false) not working

1iverea9er, 2025-01-14:
### The problem

This code not working if initial state of `binary_sensor` is `unknown`
`id(my_binary_sensor).publish_state(false);`

### Which version of ESPHome has the issue?

2024.12.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.1.2

### What platform are you using?

ESP8266

### Board

nodemcu

### Component causing the issue

binary_sensor

### YAML Config

```yaml
esphome:
  on_boot:
      priority: 800
      then:
          - delay: 1min
          - lambda: |-
                      id(warning).publish_state(false);
```

### Additional information

But as the same time `on` state publish worked...
`id(my_binary_sensor).publish_state(true);`

ssieb, 2025-01-14:
What kind of binary sensor?  If it's not a template, you shouldn't be publishing to it anyway.

1iverea9er, 2025-01-14:
Sorry. `platform: gpio`
I update yaml config in first post

ssieb, 2025-01-15:
You really shouldn't be trying to do that.  And it should always have a valid state.  Where do you see it as "unknown"?

ssieb, 2025-01-15:
Maybe try adding `publish_initial_state: true`

1iverea9er, 2025-01-15:
Its

> You really shouldn't be trying to do that. And it should always have a valid state. Where do you see it as "unknown"?

I see "unknown" state in HA.

Why then can I post the "on" state if this is the wrong path? It seems to me that the problem is not the choice of method, but that something is broken.

ESP logs:

[image]

Here you can see that no actions are performed in ESPHome itself after the publication of "false"

ssieb, 2025-01-16:
If you removed the filters, you might get something.  The binary sensor is updating every loop, so every 16ms.  Your published value will be immediately replaced with the real state and your filter will hide that change.

1iverea9er, 2025-01-17:
Yes, this is really happening due to the presence of filters. But then another question arises: Why should the intentional publication of the value depend on the filter?

ssieb, 2025-01-17:
You're missing my point.  You should not be publishing to a non-template binary sensor.  That binary sensor is updating itself every 16ms, so your filter is going to basically ignore the little blip that you published.  I don't know why the initial value isn't getting to HA.  That's the only possible issue here.

closed 2025-08-17 by github-actions[bot]