esphome/issues #6681: ESPHome does not decode SML telegrams coming from UART

riker187, 2025-01-15:
### The problem

I have a ESP (D1 Mini) receiving SML telegramms from a EMH smart energy meter. The messages are logged in hex format on the uart log. The expected behaviour would be that they will be passed on to the SML integration and then decoded. This does not happen.

### Which version of ESPHome has the issue?

2024.12.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.1.2

### What platform are you using?

ESP8266

### Board

D1 Mini

### Component causing the issue

sml

### YAML Config

```yaml
esphome:
  name: sml1
  friendly_name: SML1

esp8266:
  board: d1_mini

# Enable logging
logger:
  level: DEBUG
    # for initial setup: VERY_VERBOSE
  logs:
    sml: DEBUG
    text_sensor: DEBUG

# Enable Home Assistant API
api:
  encryption:
    key: "xxxx"

ota:
  - platform: esphome
    password: "xxxx"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

captive_portal:

    
uart:
  id: uart_bus
  rx_pin: 
    number: D1
  baud_rate: 9600
  data_bits: 8
  parity: NONE
  stop_bits: 1
  debug:

sml:
  id: mysml
  uart_id: uart_bus
  on_data:
    - logger.log: "telegram received"

text_sensor:
  - platform: sml
    name: "PV"
    #sml_id: mysml
    obis_code: "129-129:199.130.3"
    format: text

sensor:
  - platform: sml
    name: "SM_Total energy bought"
    #sml_id: mysml
    obis_code: "1-0:1.8.0"
    unit_of_measurement: kWh
    accuracy_decimals: 4
    device_class: energy
    state_class: total_increasing
    filters:
      - multiply: 0.0001
  - platform: sml
    name: "SM_Total energy bought-1.8.2"
    #sml_id: mysml
    obis_code: "1-0:1.8.2"
    unit_of_measurement: kWh
    accuracy_decimals: 4
    device_class: energy
    state_class: total_increasing
    filters:
      - multiply: 0.0001
  - platform: sml
    name: "SM_Total energy sold"
    #sml_id: mysml
    obis_code: "1-0:2.8.0"
    unit_of_measurement: kWh
    accuracy_decimals: 4
    device_class: energy
    state_class: total_increasing
    filters:
      - multiply: 0.0001
  - platform: sml
    name: "Current load"
    #sml_id: mysml
    obis_code: "1-0:16.7.0"
    unit_of_measurement: W
    accuracy_decimals: 0
    device_class: energy
    filters:
      - multiply: 0.1
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2024.12.2
INFO Reading configuration /config/esphome/sml1.yaml...
INFO Starting log output from 192.168.178.102 using esphome API
INFO Successfully connected to sml1 @ 192.168.178.102 in 0.003s
INFO Successful handshake with sml1 @ 192.168.178.102 in 2.522s
[12:17:03][I][app:100]: ESPHome version 2024.12.2 compiled on Jan 15 2025, 11:58:56
[12:17:03][C][wifi:600]: WiFi:
[12:17:03][C][wifi:428]:   Local MAC: 4C:EB:D6:ED:AF:82
[12:17:03][C][wifi:433]:   SSID: [redacted]
[12:17:03][C][wifi:436]:   IP Address: 192.168.178.102
[12:17:03][C][wifi:439]:   BSSID: [redacted]
[12:17:03][C][wifi:441]:   Hostname: 'sml1'
[12:17:03][C][wifi:443]:   Signal strength: -46 dB ▂▄▆█
[12:17:03][C][wifi:447]:   Channel: 5
[12:17:03][C][wifi:448]:   Subnet: 255.255.255.0
[12:17:03][C][wifi:449]:   Gateway: 192.168.178.1
[12:17:03][C][wifi:450]:   DNS1: 192.168.178.1
[12:17:03][C][wifi:451]:   DNS2: 8.8.8.8
[12:17:03][C][logger:185]: Logger:
[12:17:03][C][logger:186]:   Level: DEBUG
[12:17:03][C][logger:188]:   Log Baud Rate: 115200
[12:17:03][C][logger:189]:   Hardware UART: UART0
[12:17:03][C][logger:193]:   Level for 'sml': DEBUG
[12:17:03][C][logger:193]:   Level for 'text_sensor': DEBUG
[12:17:03][C][uart.arduino_esp8266:118]: UART Bus:
[12:17:03][C][uart.arduino_esp8266:120]:   RX Pin: GPIO5
[12:17:03][C][uart.arduino_esp8266:122]:   RX Buffer Size: 256
[12:17:03][C][uart.arduino_esp8266:124]:   Baud Rate: 9600 baud
[12:17:03][C][uart.arduino_esp8266:125]:   Data Bits: 8
[12:17:03][C][uart.arduino_esp8266:126]:   Parity: NONE
[12:17:03][C][uart.arduino_esp8266:127]:   Stop bits: 1
[12:17:03][C][uart.arduino_esp8266:131]:   Using software serial
[12:17:03][C][sml:104]: SML:
[12:17:03][C][sml_text_sensor:046]: SML 'PV'
[12:17:03][C][sml_text_sensor:050]:   OBIS Code: 129-129:199.130.3
[12:17:03][C][sml_sensor:033]: SML 'SM_Total energy bought'
[12:17:03][C][sml_sensor:033]:   Device Class: 'energy'
[12:17:03][C][sml_sensor:033]:   State Class: 'total_increasing'
[12:17:03][C][sml_sensor:033]:   Unit of Measurement: 'kWh'
[12:17:03][C][sml_sensor:033]:   Accuracy Decimals: 4
[12:17:03][C][sml_sensor:037]:   OBIS Code: 1-0:1.8.0
[12:17:04][C][sml_sensor:033]: SML 'SM_Total energy bought-1.8.2'
[12:17:04][C][sml_sensor:033]:   Device Class: 'energy'
[12:17:04][C][sml_sensor:033]:   State Class: 'total_increasing'
[12:17:04][C][sml_sensor:033]:   Unit of Measurement: 'kWh'
[12:17:04][C][sml_sensor:033]:   Accuracy Decimals: 4
[12:17:04][C][sml_sensor:037]:   OBIS Code: 1-0:1.8.2
[12:17:04][C][sml_sensor:033]: SML 'SM_Total energy sold'
[12:17:04][C][sml_sensor:033]:   Device Class: 'energy'
[12:17:04][C][sml_sensor:033]:   State Class: 'total_increasing'
[12:17:04][C][sml_sensor:033]:   Unit of Measurement: 'kWh'
[12:17:04][C][sml_sensor:033]:   Accuracy Decimals: 4
[12:17:04][C][sml_sensor:037]:   OBIS Code: 1-0:2.8.0
[12:17:04][C][sml_sensor:033]: SML 'Current load'
[12:17:04][C][sml_sensor:033]:   Device Class: 'energy'
[12:17:06][D][uart_debug:114]: <<< 1F:E0:FC:F0:FF:1F:F0:3E:F2:78:E0:D8:00:80:00:04:C0:00:08:20:F0:FE:10:70:FC:04:80:FC:00:7E:88:FE:F8:16:FF:E0:00:FE:E0:FF:FE:F8:FE:FF:F8:F8:27:7F:7C:80:FF:FE:FA:F0:9E:F8:38:DC:0C:20:C7:FC:FC:F8:FF:FF:FB:FF:DF:FA:FE:E0:3B:FE:FE:FE:FF:F8
[12:17:07][D][uart_debug:114]: <<< 3E:F0:F0:F0:FF:FC:F8:00:7E:3E:C0:F0:F0:1E:78:00:F0:30:00:80:40:E0:30:FE:3C:C0:FE:80:80:18:00:FF:E0:78:7E:40:F2:FC:00:00:01:80:80:0C:30:80:E0:F0:7F:FC:F8:FC:FE:FF:86:F8:F8:FC:FE:78:F8:E0:20:E0:1E:FC:FE:FF:FF:FF:FF:7E:C0:1C:FF:FF:FF:FF:FF
[12:17:09][D][uart_debug:114]: <<< FD:F4:E0:F0:FE:F8:CF:F8:C0:04:00:FB:FC:E3:03:0E:06:10:38:3E:84:FC:FE:04:B8:FF:38:1E:F0:FE:78:0C:CE:C0:C2:40:0F:38:00:10:0E:43:FE:00:EE:C0:FE:0C:02:84:FE:FC:FE:00:E0:FE:FC:F8:FC:FC:FF:1E:BE:1F:00:C6:F2:FC:00:3F:E8:DE:CF:F0:FF:F8:C0:FE:FE
[12:17:10][D][uart_debug:114]: <<< 1F:F6:F0:F8:FF:C0:FE:F3:C0:00:00:80:78:F8:02:E0:40:00:C4:E0:F0:20:E0:E0:FE:FE:CC:00:F8:F0:E0:60:30:F8:F2:78:30:78:10:08:18:00:3F:30:02:80:00:F8:80:FC:7C:E0:E0:FC:FC:00:FE:FF:E4:FF:FE:3F:E0:FE:FE:FE:FE:FF:1F:F0:83:F8:7C:FE:FC:FE:FE:7F:FE:60:FE:FC:FE:FC:FF:FE:E0
[12:17:12][D][uart_debug:114]: <<< E4:F0:F8:F6:F8:FF:F8:1E:CE:00:00:00:E0:80:F8:88:31:48:C0:80:7E:80:E0:D8:00:8C:00:FE:FE:30:0F:CC:E0:C0:C6:00:F0:84:00:10:0F:E0:D0:F8:F0:3E:E0:0E:FC:B0:00:04:30:C0:F3:FE:60:FF:EE:FE:FC:C6:F0:3E:FF:9E:8E:9E:C0:03:7E:00:FC:FE:00:FE:F0:E6:FF:FC:FF:FE:FE:FF:FF:FF:FE:38:F8:FF:FF:FF
[12:17:14][D][uart_debug:114]: <<< F8:E0:F8:F0:F8:E6:90:FE:1E:C0:0C:A0:7C:FE:00:9E:0C:00:18:7C:F0:FE:C2:F8:FC:FE:FF:FF:EC:F0:FE:E3:F8:FE:7F:CF:FF:FE:30:FC:F8:00:E0:F8:E0:F0:26:C0:FC:FC:7E:FE:8F:F8:FC:FC:43:F8:F8:FE:FF:F8:FC:FF:7F:FE:FC:FF:FF:FF
[12:17:15][D][uart_debug:114]: <<< F8:E0:F0:F0:E0:00:0F:8C:F0:EC:00:0C:1C:63:3F:10:00:38:F2:FF:BE:E6:F0:FC:F8:EC:C6:FF:9E:E1:86:C0:F0:FE:DC:CE:F8:FE:F8:FF:C0:00:FC:FE:90:FF:7C:FC
[12:17:17][D][uart_debug:114]: <<< F8:E0:F0:F0:F8:EC:F0:1F:1F:FE:1C:40:02:5E:F0:18:00:60:8C:08:30:3C:7E:20:F0:00:3E:7C:FC:FC:BF:00:FB:00:FF:FE:00:00:00:00:3C:00:00:E0:B8:64:C0:FC:7E:00:38:FC:FE:FE:48:FE:FC:DC:FC:00:FC:00:00:E0:C0:3B:FE:7E:03:00:C0:CF:FF:FC:FF:FE:C0:FE:07:F8
[12:17:19][D][uart_debug:114]: <<< FD:E0:F0:F0:FE:E0:20:00:80:F4:1C:80:B8:F0:C6:E0:06:30:C0:00:18:F8:F0:F0:F0:F0:F8:F8:E0:18:C0:E2:F0:F3:FF:FC:7E:82:0C:00:F3:FC:FE:90:FE:F8:80:8C:FE:F0:FC:7F:FF:8F:E2:F0:1C:FE
[12:17:20][D][uart_debug:114]: <<< F0:F0:F0:FF:FC:F0:FE:FC:1F:CE:F2:38:E0:78:F8:D8:02:0E:38:00:F0:FE:60:F0:F8:FC:F8:FE:F8:3C:EC:98:C6:F0:FC:E6:C0:F8:F8:F0:FC:FC:42:FC:F8:F8:E0:F8:F8:FC:7F:7F:3C:FC:FC:B0:FF:FC:FF:F8:FE:FF:FC
```

riker187, 2025-01-15:
There was already the same issue reported: https://github.com/esphome/issues/issues/4584 But it was closed because the user switched to another solution

TheOfficialMrBlah, 2025-01-29:
Exactly the same problem for me with an efr digital meter, but works with Tasmota.

closed 2025-06-15 by github-actions[bot]