esphome/issues #6682: `http_request` component stucks on infinite loop when HTTP server does not send `Content-Length` header

miceno, 2025-01-15:
### The problem

I am using `http_request` to call a HTTP server that is not fully HTTP compliant 😞 . The server is not responding with a `Content-Length` header, so the code in the ESPhome `http_request` component goes into an infinite loop.

### Which version of ESPHome has the issue?

2024.12.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.1.2

### What platform are you using?

ESP8266

### Board

d1_mini

### Component causing the issue

http_request

### YAML Config

```yaml
substitutions:
  zte_update_interval: "1min"

esphome:
  name: esp-httprequest
  friendly_name: esp-httprequest
  platformio_options:
    build_flags: 
      - "-DDEBUG_ESP_HTTP_CLIENT -DDEBUG_ESP_PORT=Serial"

esp8266:
  board: d1_mini

# Enable logging
logger:
  level: DEBUG
  logs: 
    http_request: DEBUG

# Enable Home Assistant API
api:
  encryption:
    key: "REDACTED"

ota:
  - platform: esphome
    password: "REDACTED"

wifi:
  ssid: !secret lab_wifi_ssid
  password: !secret lab_wifi_password
      
http_request:
  verify_ssl: False

sensor:
  - platform: template
    id: battery_vol_percent
    name: "Battery value"
    update_interval: never
    
interval:
  - interval: ${zte_update_interval}
    then: 
      - http_request.get:
          url: "http://192.168.0.1/goform/goform_get_cmd_process?isTest=false&multi_data=1&cmd=battery_vol_percent"
          # max_response_buffer_size: 64
          headers: 
            Host: "192.168.0.1"
            Referer: "http://192.168.0.1/index.html"
          capture_response: true
          on_response:
            then:
              - if:
                  condition:
                      lambda: return response->status_code == 200;
                  then:
                      - lambda: |-
                          json::parse_json(body, [](JsonObject root) -> bool {
                              if(root["battery_vol_percent"]){
                                id(battery_vol_percent).publish_state(root["battery_vol_percent"]);
                              }else {
                                ESP_LOGW("zte", "Missing key '%s' in array", "battery_vol_percent");
                                return false;
                              }
                            return true;
                          });
                      - logger.log:
                          format: "HTTP OK: Response status: %d, message %s"
                          args: [ 'response->status_code', 'body.c_str()' ]
```

### Anything in the logs that might be useful for us?

```txt
I hereby attach a (redacted) debug output using curl to the HTTP server:

$ curl --trace - 'http://192.168.0.1/goform/goform_get_cmd_process?isTest=false&multi_data=1&cmd=battery_vol_percent'  -H 'Host: 192.168.0.1' -H 'Referer: http://192.168.0.1/index.html' --insecure 
== Info:   Trying 192.168.0.1:80...
== Info: Connected to 192.168.0.1 (192.168.0.1) port 80
=> Send header, 193 bytes (0xc1)
0000: 47 45 54 20 2f 67 6f 66 6f 72 6d 2f 67 6f 66 6f GET /goform/gofo
0010: 72 6d 5f 67 65 74 5f 63 6d 64 5f 70 72 6f 63 65 rm_get_cmd_proce
0020: 73 73 3f 69 73 54 65 73 74 3d 66 61 6c 73 65 26 ss?isTest=false&
0030: 6d 75 6c 74 69 5f 64 61 74 61 3d 31 26 63 6d 64 multi_data=1&cmd
0040: 3d 62 61 74 74 65 72 79 5f 76 6f 6c 5f 70 65 72 =battery_vol_per
0050: 63 65 6e 74 20 48 54 54 50 2f 31 2e 31 0d 0a 48 cent HTTP/1.1..H
0060: 6f 73 74 3a 20 31 39 32 2e 31 36 38 2e 30 2e 31 ost: 192.168.0.1
0070: 0d 0a 55 73 65 72 2d 41 67 65 6e 74 3a 20 63 75 ..User-Agent: cu
0080: 72 6c 2f 38 2e 37 2e 31 0d 0a 41 63 63 65 70 74 rl/8.7.1..Accept
0090: 3a 20 2a 2f 2a 0d 0a 52 65 66 65 72 65 72 3a 20 : */*..Referer: 
00a0: 68 74 74 70 3a 2f 2f 31 39 32 2e 31 36 38 2e 30 http://192.168.0
00b0: 2e 31 2f 69 6e 64 65 78 2e 68 74 6d 6c 0d 0a 0d .1/index.html...
00c0: 0a                                              .
== Info: Request completely sent off
== Info: HTTP 1.0, assume close after body
<= Recv header, 17 bytes (0x11)
0000: 48 54 54 50 2f 31 2e 30 20 32 30 30 20 4f 4b 0d HTTP/1.0 200 OK.
0010: 0a                                              .
<= Recv header, 22 bytes (0x16)
0000: 53 65 72 76 65 72 3a 20 47 6f 41 68 65 61 64 2d Server: GoAhead-
0010: 57 65 62 73 0d 0a                               Webs..
<= Recv header, 18 bytes (0x12)
0000: 50 72 61 67 6d 61 3a 20 6e 6f 2d 63 61 63 68 65 Pragma: no-cache
0010: 0d 0a                                           ..
<= Recv header, 25 bytes (0x19)
0000: 43 61 63 68 65 2d 63 6f 6e 74 72 6f 6c 3a 20 6e Cache-control: n
0010: 6f 2d 63 61 63 68 65 0d 0a                      o-cache..
<= Recv header, 25 bytes (0x19)
0000: 43 6f 6e 74 65 6e 74 2d 54 79 70 65 3a 20 74 65 Content-Type: te
0010: 78 74 2f 68 74 6d 6c 0d 0a                      xt/html..
<= Recv header, 2 bytes (0x2)
0000: 0d 0a                                           ..
<= Recv data, 28 bytes (0x1c)
0000: 7b 22 62 61 74 74 65 72 79 5f 76 6f 6c 5f 70 65 {"battery_vol_pe
0010: 72 63 65 6e 74 22 3a 22 31 38 22 7d             rcent":"18"}
{"battery_vol_percent":"18"}<= Recv data, 0 bytes (0x0)
== Info: Closing connection

I also attach the (redacted) debug output from the device when using the attached yaml.
```text
[14:03:38][C][http_request:013]: HTTP Request:
[14:03:39][C][http_request:014]:   Timeout: 4500ms
[14:03:39][C][http_request:015]:   User-Agent: ESPHome/2024.12.2 (https://esphome.io)
[14:03:39][C][http_request:016]:   Follow redirects: YES
[14:03:39][C][http_request:017]:   Redirect limit: 3
[14:04:26][W][http_request.arduino:065]: Using HTTP on Arduino version >= 3.1 is **very** slow. Consider setting framework version to 3.0.2 in your YAML, or use HTTPS
[14:04:26][HTTP-Client][begin] url: http://192.168.0.1/goform/goform_get_cmd_process?isTest=false&multi_data=1&cmd=battery_value
[HTTP-Client][begin] host: 192.168.0.1 port: 80 url: /goform/goform_get_cmd_process?isTest=false&multi_data=1&cmd=battery_value
[HTTP-Client][sendRequest] type: 'GET' redirCount: 0
[HTTP-Client] connected to 192.168.0.1:80
[HTTP-Client] sending request header
-----
GET /goform/goform_get_cmd_process?isTest=false&multi_data=1&cmd=battery_value HTTP/1.1
[14:04:26]Host: 192.168.0.1
[14:04:26]User-Agent: ESPHome/2024.12.2 (https://esphome.io)
[14:04:26]Accept-Encoding: identity;q=1,chunked;q=0.1,*;q=0
[14:04:26]Connection: keep-alive
[14:04:26]Referer: http://192.168.0.1/index.html
[14:04:26]Content-Length: 0
[14:04:26]
[14:04:26][HTTP-Client] getStreamPtr: not connected
[E][http_request.arduino:138]: Stream pointer vanished!
[14:04:26][HTTP-Client] getStreamPtr: not connected
[E][http_request.arduino:138]: Stream pointer vanished!
[14:04:26][HTTP-Client] getStreamPtr: not connected
[E][http_request.arduino:138]: Stream pointer vanished!
[14:04:26][HTTP-Client] getStreamPtr: not connected
[E][http_request.arduino:138]: Stream pointer vanished!
[14:04:27][HTTP-Client] getStreamPtr: not connected
[E][http_request.arduino:138]: Stream pointer vanished!

```

It shows that the HTTPClient object inside the container vanishes, since the `http_request` component is reading beyond the size of the available data on the response and the connection is closed, making the stream pointer to the HTTPClient invalid.
```

### Additional information

The root of the problem lies [here](https://github.com/esphome/esphome/blob/4b51ba3fa4061df52e6c8abeac3d074003073502/esphome/components/http_request/http_request.h#L205) in the implementation of the read loop: 
the code is assuming the `read` operation will always succeed, but it is not, it will not succeed when reading a stream with no content length.

A simple way to solve it is by adding a check on the result of `HTTPContainer::read`, like this:

```cpp
207 if (read < 0){
208  ESP_LOGD("http_request", "Exit read loop");
209  break;
210 }
```

miceno, 2025-01-15:
I can prepare a PR for it, but not quite sure if it is the way to do myself or one of the maintainers should first agree on the solution.

Anyway, I created a "fork" of the component [here](https://github.com/miceno/esphome-http_request/tree/fix/infinite-loop) and I used it on my HA, feel free to use it.

You can test it in your environment with this:

```yaml
...
external_components:
  - source:
      type: git
      url: https://github.com/miceno/esphome-http_request
      ref: fix/infinite-loop
    components: [ http_request ]
    refresh: 1s
...
```

davidstrobl, 2025-01-16:
Couldn't the mentioned problem be related to issue I reported?   
https://github.com/esphome/issues/issues/6677

miceno, 2025-01-16:
I think it is not related, since from an initial review your report is for a connection issue, while the one I reported is about parsing the response of a request.