esphome/issues #6686: ESPHome rtc based lambda functions

adizanni, 2025-01-16:
### The problem

In reference to an issue that was reported and closed for being stale (https://github.com/esphome/issues/issues/2479).
The inclusion of lambda functions with rtc_io.h (attached), triggers a compilation error (attached).
This is preventing the proper implementation of a common use case: ESP32 CAM powered by batteries, used to go in deep sleep mode, wake up, take a snapshot and go back to sleep; by design ESP32 CAM in deep sleep mode turns on the flash led (with half brightness), this consumes considerable amount of battery charge and defies the purpose of the deep sleep mode. It can be considered as a ESP32 CAM blueprint issue, but a fix in ESPHome would be greatly appreciated. The snippet of Lambda code is considered as a working solution.

### Which version of ESPHome has the issue?

2024.12.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.12.1

### What platform are you using?

ESP32

### Board

ESP32 CAM (AI Thinker)

### Component causing the issue

lambda

### YAML Config

```yaml
on_shutdown:
    then:
      - lambda: |-
          #include <driver/gpio.h>
          #include <driver/rtc_io.h>
          pinMode(4, OUTPUT);
          digitalWrite(4, LOW);
          rtc_gpio_hold_en(GPIO_NUM_4);
          gpio_deep_sleep_hold_en();
          esp_deep_sleep_start();
```

### Anything in the logs that might be useful for us?

```txt
In file included from /root/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/driver/driver/rtc_io.h:22:0,
                 from /config/esphome/gas-meter-camera.yaml:9:
/root/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/soc/soc/rtc_periph.h: In lambda function:
/root/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/soc/soc/rtc_periph.h:23:8: error: expected unqualified-id before string constant
 extern "C"
        ^
/config/esphome/gas-meter-camera.yaml:118:1: error: expected '}' at end of input
/config/esphome/gas-meter-camera.yaml: In function 'void setup()':
/config/esphome/gas-meter-camera.yaml:118:1: error: expected ')' at end of input
/config/esphome/gas-meter-camera.yaml:118:1: error: no matching function for call to 'esphome::LambdaAction<>::LambdaAction()'
In file included from src/esphome.h:72:0,
                 from src/main.cpp:3:
src/esphome/core/base_automation.h:126:12: note: candidate: esphome::LambdaAction<Ts>::LambdaAction(std::function<void(Ts ...)>&&) [with Ts = {}]
   explicit LambdaAction(std::function<void(Ts...)> &&f) : f_(std::move(f)) {}
            ^
src/esphome/core/base_automation.h:126:12: note:   candidate expects 1 argument, 0 provided
src/esphome/core/base_automation.h:124:32: note: candidate: esphome::LambdaAction<>::LambdaAction(const esphome::LambdaAction<>&)
 template<typename... Ts> class LambdaAction : public Action<Ts...> {
                                ^
src/esphome/core/base_automation.h:124:32: note:   candidate expects 1 argument, 0 provided
src/esphome/core/base_automation.h:124:32: note: candidate: esphome::LambdaAction<>::LambdaAction(esphome::LambdaAction<>&&)
src/esphome/core/base_automation.h:124:32: note:   candidate expects 1 argument, 0 provided
/config/esphome/gas-meter-camera.yaml:118:1: error: expected '}' at end of input
```

ssieb, 2025-01-17:
I'm not sure what you're expecting here.  You can't use an include like that in a lambda function.  If you need to do something like this, use an included `.h` file that includes the header and possibly even includes the method you want to call from a lambda.

adizanni, 2025-01-17:
Thanks @ssieb for the fast reply, I was hoping to receive a workaround like the one you mentioned and/or a more elegant way to achieve what I was trying to do. In parallel I'm trying to build the same use case directly in esp-idf (of course not the same effort and not good for my "simple" home assistant architecture). 

I will try first your workaroung

closed 2025-06-01 by github-actions[bot]