esphome/issues #6689: Clamp filter returning infinity

morpheby, 2025-01-17:
### The problem

I’m using a filter chain of clamp, median, and then exponential moving average. The original value depends on HA input value, and during startup it can be a little late. This is enough for it to be 0 during at least one update cycle and, since it is a divisor in a formula, I get infinity as raw value. 

The exponential filter chokes on infinity for the eternity, so I intended to use clamp to prevent this and replace +inf/-inf with a value. Turns out, this doesn’t happen, as the clamp filter [here](https://github.com/esphome/esphome/blob/9e40d4cf45269e052bd62a78a5619ad8dc863594/esphome/components/sensor/filter.cpp#L469) checks if the value is finite, and if not, returns the value itself. This doesn’t look correct, as by design clamp filter ought to properly consider +inf/-inf to be higher or lower than max/min. Instead, it should only ignore NaN, as it can’t be compared to anything. 

### Which version of ESPHome has the issue?

2024.9.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.1.2

### What platform are you using?

ESP32

### Board

ESP32-S2 DevKit

### Component causing the issue

sensor/filter/clamp

### YAML Config

```yaml

```

### Anything in the logs that might be useful for us?

```txt

```

smashedr, 2025-01-17:
I recently switch from custom lambda to clamp. You can probably get the desired results with a lambda filter as a workaround.

morpheby, 2025-01-21:
@smashedr That’s what I did, thanks for the advice. Nonetheless, I still believe it to be a bug. This is doesn’t seem like the expected behavior of the clamp filter.

ssieb, 2025-01-21:
Your link points to the `round` filter, not clamp.  I agree that it's a bug.  It should just be checking for `nan`.
However, I also think you have an issue in whatever is feeding this, since a divide by 0 should be crashing the device, not returning `inf`.  Don't do that.

morpheby, 2025-01-21:
@ssieb Fixed link. 

Float div by zero is not an error (IEEE 754) and definitely should not be crashing the device :)

Not sure if I can do anything about it being temporarily zero though. It is an input_number from HA, somehow for a short time after ESP boots it is zero, and using clamp seems much simpler here, than fixing the zero issue.

closed 2025-06-05 by github-actions[bot]