esphome/issues #6690: Binary input timing conversions

maslyankov, 2025-01-17:
### The problem

When using timing with Xs it does not work:

```
binary_sensor:
  - platform: gpio
    pin: 
      number: GPIO34
      inverted: true
    name: "${friendly_name} Factory Button"
    on_multi_click:
      - timing:
          - ON for at least 3s
          - OFF for at least 0.5s
        then:
          - lambda: |-
              ESP_LOGI("button", "[Factory Button] Hold Detected! Doing Factory Reset in 5 seconds...");
          - uart.write: "[Factory Button] Hold Detected! Doing Factory Reset in 5 seconds...\n"  # Reboot message to be written to UART
          - delay: 5s
          - button.press: factory_reset_button
```

When using timing with ms equivalent it works:

```
binary_sensor:
  - platform: gpio
    pin: 
      number: GPIO34
      inverted: true
    name: "${friendly_name} Factory Button"
    on_multi_click:
      - timing:
          - ON for at least 3000ms
          - OFF for at least 0.5s
        then:
          - lambda: |-
              ESP_LOGI("button", "[Factory Button] Hold Detected! Doing Factory Reset in 5 seconds...");
          - uart.write: "[Factory Button] Hold Detected! Doing Factory Reset in 5 seconds...\n"  # Reboot message to be written to UART
          - delay: 5s
          - button.press: factory_reset_button
```

### Which version of ESPHome has the issue?

2024.12.4

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

### YAML Config

```yaml

```

### Anything in the logs that might be useful for us?

```txt

```

ssieb, 2025-01-17:
What do you see in the logs?  Set the logger to verbose to see multi-click progress messages.

smashedr, 2025-01-17:
I saw this out of the corner of my eye, and it made me curious to test my buttons, at least since 12+ (which has been somewhat buggy) and they all work with no issues. I have a few different buttons all with timings, here are a couple examples, that I just tested.

2024.11.3:
```yaml
binary_sensor:
  # https://esphome.io/#binary-sensor-components
  - platform: gpio
    name: "Touch 1 - ${box_num}"
    id: touch_1
    icon: mdi:gesture-tap-box
    pin:
      number: 23
      mode: INPUT_PULLUP
    on_multi_click:
      - timing:
          - ON for at most 0.9s
        then:
          - logger.log: "Single Click - Start Voice"
          - script.execute: va_start
      - timing:
          - ON for at least 1s
        then:
          - logger.log: "1+ Second Touch - Toggle Wake"
          - rtttl.play: "s:d=4,o=6,b=100:16d"
          - switch.toggle: micro_wake_toggle
```

2024.12.2:
```yaml
platform: gpio
name: "Physical Button ${box_num}"
icon: mdi:radiobox-marked
entity_category: diagnostic
disabled_by_default: true
pin:
  number: ${pin}
  mode: INPUT_PULLUP
filters:
  - invert:
on_multi_click:
  - timing:
      - ON for at least 3s
    then:
      - logger.log: "3 Second Reset"
      - button.press: btn_restart
  - timing:
      - ON for at most 2.9s
      - OFF for at least 0.1s
    then:
      - logger.log: "Single Click"
      - light.toggle: main_led

```

maslyankov, 2025-01-17:
> What do you see in the logs? Set the logger to verbose to see multi-click progress messages.

```

[23:09:19][V][binary_sensor.automation:026]: Multi Click: Starting multi click action!
[23:09:19][V][binary_sensor.automation:025]: START min=3000 max=4294967294
[23:09:19][V][binary_sensor.automation:026]: Multi Click: Starting multi click action!
[23:09:19][V][binary_sensor.automation:025]: START min=0 max=1000
[23:09:19][V][binary_sensor.automation:026]: Multi Click: Starting multi click action!
[23:09:19][V][binary_sensor.automation:096]: Multi Click: You waited too long to RELEASE.
[23:09:19][V][binary_sensor.automation:072]: Multi Click: Invalid length of press, starting cooldown of 1000 ms...
[23:09:19][V][binary_sensor.automation:096]: Multi Click: You waited too long to RELEASE.
[23:09:19][V][binary_sensor.automation:072]: Multi Click: Invalid length of press, starting cooldown of 1000 ms...
[23:09:20][V][binary_sensor.automation:096]: Multi Click: You waited too long to RELEASE.
[23:09:20][V][binary_sensor.automation:072]: Multi Click: Invalid length of press, starting cooldown of 1000 ms...
[23:09:20][I][safe_mode:041]: Boot seems successful; resetting boot loop counter
[23:09:20][D][esp32.preferences:114]: Saving 1 preferences to flash...
[23:09:20][V][esp32.preferences:126]: sync: key: 233825507, len: 4
[23:09:20][D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[23:09:20][V][binary_sensor.automation:075]: Multi Click: Cooldown ended, matching is now enabled again.
[23:09:20][V][binary_sensor.automation:075]: Multi Click: Cooldown ended, matching is now enabled again.
[23:09:21][V][binary_sensor.automation:075]: Multi Click: Cooldown ended, matching is now enabled again.
[23:09:21][D][binary_sensor:036]: 'DTWonder Relay Hub 8CH Factory Button': Sending state OFF
[23:09:21][V][binary_sensor.automation:035]: Multi Click: action not started because first level does not match!
[23:09:21][V][binary_sensor.automation:035]: Multi Click: action not started because first level does not match!
[23:09:21][V][binary_sensor.automation:072]: Multi Click: Invalid length of press, starting cooldown of 1000 ms...
[23:09:21][V][binary_sensor.automation:035]: Multi Click: action not started because first level does not match!
[23:09:22][V][binary_sensor.automation:075]: Multi Click: Cooldown ended, matching is now enabled again.
```

ssieb, 2025-01-17:
From those logs, you didn't hold the button long enough.  It has the right timings.

ssieb, 2025-01-17:
That other issue is very different from this.

maslyankov, 2025-01-17:
Okay. You are right. Holding it for 4-5 secs seems to work. Apparently there is some delay with the couter in the device or in my head... 😄

ssieb, 2025-01-17:
The timestamp in the log is from the computer you're watching the logs from, so the device seems to be correct.

closed 2025-01-17 by ssieb