esphome/issues #6691: Warning with ili9xxx and ESPHome 2024.12.4

andreas-bulling, 2025-01-17:
### The problem

I see the following warning in the logs, quickly repeating (roughly every second), when using the ili9xxx display component (esp32-dev, arduino):

```
[21:18:46][W][component:238]: Components should block for at most 30 ms.
[21:18:47][W][component:237]: Component display took a long time for an operation (91 ms).
```

This is the corresponding part of the configuration:

```
display:
  - platform: ili9xxx
    model: ILI9342
    spi_id: lcd
    cs_pin:
      number: GPIO15
      ignore_strapping_warning: true
    dc_pin:
      number: GPIO02
      ignore_strapping_warning: true
    rotation: 180
    color_order: RGB
    invert_colors: false
    lambda: |-
      int hs = it.get_width() / 2; // Horizontal Spacing = text data horizontal center point
      int hq = it.get_width() / 4; // text data horizontal center for two vertical lines
      int vs = it.get_height() / 8; // Vertical Center = text data vertical center point = how many lines
      int line_gap = 21; // distance of line from center of data text
      it.rectangle(0,  0, it.get_width(), it.get_height(), id(ha_blue));
      it.rectangle(0, 20, it.get_width(), it.get_height(), id(ha_blue)); 
     
      it.strftime(5, 5, id(arial_12), id(my_white), TextAlign::TOP_LEFT, "%H:%M:%S", id(homeassistant_time).now());
      it.print(hs, 5, id(arial_12), id(ha_blue), TextAlign::TOP_CENTER, "Solar"); //print title  
      
      if (id(system_status).state) {
        it.print(it.get_width()-5, 5, id(arial_12), id(my_green), TextAlign::TOP_RIGHT, "Online");
      } else {
        it.print(it.get_width()-5, 5, id(arial_12), id(my_red), TextAlign::TOP_RIGHT, "Offline");
      }
      
      it.print(hq, vs * 1.5, id(arial_12), id(my_white), TextAlign::CENTER, "Feed-in Total:");
      it.printf(3*hq, vs * 1.5, id(arial_12), id(my_white), TextAlign::CENTER, "%.1f kWh", id(energy_feedin_total).state);
      it.print(hq, vs * 2.5, id(arial_12), id(my_white), TextAlign::CENTER, "Feed-in Today:");
      it.printf(3*hq, vs * 2.5, id(arial_12), id(my_white), TextAlign::CENTER, "%.1f kWh", id(energy_feedin).state);
      it.print(hq, vs * 3.5, id(arial_12), id(my_white), TextAlign::CENTER, "Yield Today / Total:");
      it.printf(3*hq, vs * 3.5, id(arial_12), id(my_white), TextAlign::CENTER, "%.1f kWh / %.1f kWh", id(energy_today).state, id(energy_total).state);

      it.line(0, vs * 4.25, it.get_width(), vs * 4.25, id(ha_blue));

```

### Which version of ESPHome has the issue?

2024.12.4

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Component causing the issue

ili9xxx

### YAML Config

```yaml

```

### Anything in the logs that might be useful for us?

```txt

```

randybb, 2025-01-17:
Duplicate of https://github.com/esphome/issues/issues/4717

closed 2025-01-17 by randybb

andreas-bulling, 2025-01-17:
Well - but the fix (setting the I2C frequency to 400kHz) doesn't work for me. The display is not using I2C...