esphome/issues #6694: esp-idf framework not compiling - error in esphome i think

yen-yum-yum, 2025-01-19:
### The problem

While complileng yaml to create bin - ( for future use with web tool) i get foll error src/esphome/components/uart/uart_component_esp_idf.cpp: In member function 'virtual bool esphome::uart::IDFUARTComponent::peek_byte(uint8_t*)':
src/esphome/components/uart/uart_component_esp_idf.cpp:153:62: error: 'portTICK_RATE_MS' was not declared in this scope; did you mean 'portTICK_PERIOD_MS'?
  153 |     int len = uart_read_bytes(this->uart_num_, data, 1, 20 / portTICK_RATE_MS);
      |                                                              ^~~~~~~~~~~~~~~~
      |                                                              portTICK_PERIOD_MS
src/esphome/components/uart/uart_component_esp_idf.cpp: In member function 'virtual bool esphome::uart::IDFUARTComponent::read_array(uint8_t*, size_t)':
src/esphome/components/uart/uart_component_esp_idf.cpp:177:65: error: 'portTICK_RATE_MS' was not declared in this scope; did you mean 'portTICK_PERIOD_MS'?
  177 |     uart_read_bytes(this->uart_num_, data, length_to_read, 20 / portTICK_RATE_MS);
      |                                                                 ^~~~~~~~~~~~~~~~
      |                                                                 portTICK_PERIOD_MS
Compiling .pioenvs/living-mini-sensor/src/esphome/components/wifi/wifi_component_esp8266.cpp.o
*** [.pioenvs/living-mini-sensor/src/esphome/components/uart/uart_component_esp_idf.cpp.o] Error 1     - understand its some depreciated function - sorry i cant help more - i dont have tech skills 

### Which version of ESPHome has the issue?

ESPHome 2024.12.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

 Core 2025.1.2 Supervisor 2024.12.3 Operating System 14.1 Frontend 20250109.0

### What platform are you using?

ESP32

### Board

Elcrow Esp32 -C3 using esp32 c3fn4 /fh4

### Component causing the issue

i dont know whats meant by it - board is meant to run a Ld2410c sensor and double as blutooth proxy

### YAML Config

```yaml
#This is a custom firmware for Human Presence Sensor 1U
# https://screek.io/1u

esphome:
  name: living-mini-sensor
  comment: Screek Human Presence Sensor 1U
  friendly_name: living_mini_sensor
  name_add_mac_suffix: True
  on_boot:
    priority: 900
    then:
       lambda: |-
        id(cpu_speed) = ESP.getCpuFreqMHz() ;
  project: 
    name: Screek.Human_Presence_Sensor
    version: 1U
  min_version: 2023.6.1

globals:
   - id: cpu_speed
     type: int
     restore_value: no
     initial_value: '0'

external_components:
  # - source: github://screek-workshop/custom_components_support_for_screek_1u
  - source:
      type: git
[... 251 lines cut ...]
Reading CMake configuration...
Dependency Graph
|-- noise-c @ 0.1.6
Compiling .pioenvs/living-mini-sensor/src/esphome/components/uart/uart_component_esp_idf.cpp.o
Compiling .pioenvs/living-mini-sensor/src/esphome/components/web_server_idf/web_server_idf.cpp.o
Compiling .pioenvs/living-mini-sensor/src/esphome/components/wifi/wifi_component.cpp.o
Compiling .pioenvs/living-mini-sensor/src/esphome/components/wifi/wifi_component_esp32_arduino.cpp.o
src/esphome/components/uart/uart_component_esp_idf.cpp: In member function 'virtual bool esphome::uart::IDFUARTComponent::peek_byte(uint8_t*)':
src/esphome/components/uart/uart_component_esp_idf.cpp:153:62: error: 'portTICK_RATE_MS' was not declared in this scope; did you mean 'portTICK_PERIOD_MS'?
  153 |     int len = uart_read_bytes(this->uart_num_, data, 1, 20 / portTICK_RATE_MS);
      |                                                              ^~~~~~~~~~~~~~~~
      |                                                              portTICK_PERIOD_MS
src/esphome/components/uart/uart_component_esp_idf.cpp: In member function 'virtual bool esphome::uart::IDFUARTComponent::read_array(uint8_t*, size_t)':
src/esphome/components/uart/uart_component_esp_idf.cpp:177:65: error: 'portTICK_RATE_MS' was not declared in this scope; did you mean 'portTICK_PERIOD_MS'?
  177 |     uart_read_bytes(this->uart_num_, data, length_to_read, 20 / portTICK_RATE_MS);
      |                                                                 ^~~~~~~~~~~~~~~~
      |                                                                 portTICK_PERIOD_MS
Compiling .pioenvs/living-mini-sensor/src/esphome/components/wifi/wifi_component_esp8266.cpp.o
*** [.pioenvs/living-mini-sensor/src/esphome/components/uart/uart_component_esp_idf.cpp.o] Error 1
========================= [FAILED] Took 37.86 seconds =========================
```

### Additional information

it compliles without any problem if i change framework to arduino from esp-idf  which was what was reccomended by esphome docs for blutooth .proxy.    device is also  working fine with arduino framework .. sorry cant be of more help -

ssieb, 2025-01-19:
That file comes from the external component you're using.  You need to talk to them about the issue.  I'm curious what changes they needed to make to the uart component, but their code is 2 years old, so very out of date with esphome.

closed 2025-01-19 by ssieb

yen-yum-yum, 2025-01-19:
Thank you so much for your reply and clarification... I will take this
up... Apologies for wasting your time...
Thanks so much.. And thank you for the wonderful work you all do on
Esphome..