esphome/issues #6700: Modbus Binary Sensor

chuyskywalker, 2025-01-20:
### The problem

Based on [the binary sensor modbus docs](https://esphome.io/components/binary_sensor/modbus_controller) I should be able to read several registers to cobble together a 64bit int so that I can bitmask against that for a big long list of faults that my solark inverter reports

```yaml
  - platform: modbus_controller
    modbus_controller_id: modbus_client_sa
    name: "Alarm: F08 GFDI_Relay_Failure"
    address: 103
    register_type: holding
    register_count: 4
    bitmask: 0x80  # position 7
```

As an example. However, this causes compile error 

```
[register_count] is an invalid option for [binary_sensor.modbus_controller]. Did you mean [register_type]?
```

In fact, as best I can tell, there's no way to ask for more than a single register when attempting to extract bitmask values from a larger int.

### Which version of ESPHome has the issue?

2024.12.4

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

n.a

### What platform are you using?

ESP32

### Board

n/a

### Component causing the issue

modbus

### YAML Config

```yaml
esphome:
  name: demo

esp32:
  board: esp32dev
  framework:
    type: arduino
uart:
  - id: uart_modbus_client_sa
    rx_pin: 13
    tx_pin: 16
    baud_rate: 9600
    data_bits: 8
    parity: NONE
    stop_bits: 1

modbus:
  - id: modbus_sa
    uart_id: uart_modbus_client_sa

modbus_controller:
- id: modbus_client_sa
  modbus_id: modbus_sa
  address: 0x1

binary_sensor:
  - platform: modbus_controller
    modbus_controller_id: modbus_client_sa
    name: "Alarm: F08 GFDI_Relay_Failure"
    address: 103
    register_type: holding
    register_count: 4
    bitmask: 0x80  # position 7
```

### Anything in the logs that might be useful for us?

```txt

```

chuyskywalker, 2025-01-22:
Tangentially related to #5985

Hadatko, 2025-06-24:
Same for switch and number